Koenkk/zigbee2mqtt #18139: IKEA E2112 Vindstyrka air quality and humidity sensor no longer updating

ThatTallGuy21, 2023-06-28:
### What happened?

My E2112 Vindstyrka has been connected for maybe 1-2 months and I've noticed that sometimes the data isn't updating despite LQI reading 255. It always works initially for some amount of time and frequently refreshes the values, and then on days like today where the AQI is horrendous outside (due to wildfire smoke) I notice it hasn't updated. No idea when or why it stops. For example, the unit itself reads PM2.5 = 76 while the reading in Z2M/Developer Tools says it's only 6. Same applies for the TVOC reading as well.

### What did you expect to happen?

The PM2.5 and TVOC readings to continuously update. PM2.5 hasn't been as low as the reading states since last night/early morning, which is several hours ago.

### How to reproduce it (minimal and precise)

Not sure. 

I remove/readd it to Z2M and then it randomly stops reporting updated information over time. This appears to take weeks (maybe?) and both times its been on days when AQ is really bad outside and way higher than normal readings from the unit. 

Just a guess, but perhaps it gets hung up if readings are changing frequently in a short period of time? 

I tried unplugging/replugging in the unit and it didn't change anything.

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

ThatTallGuy21, 2023-07-29:
This is still an issue.

ThatTallGuy21, 2023-08-10:
Thank you! I just repaired my device and configured mine to align with yours. Hopefully that does the trick. (crosses fingers)

sjorge, 2023-08-18:
Something I noticed, after installing the available OTA update, it seems like the device crashes and restarts every 10-20minutes. Perhaps this is related to people no longer getting updates.

MattWestb, 2023-08-20:
@sjorge I have my connected to Dirigera and have not seen any problems with connections only sometime the Homekit controller integration in HA is stop reporting some of the senors (have seen it 3 times) but is starting working if restarting Homekit controller in HA so i think my problem is not Zigbee related.

ThatTallGuy21, 2023-08-31:
3 weeks later and my Vindstyrkas still running without hanging. 

> Something I noticed, after installing the available OTA update, it seems like the device crashes and restarts every 10-20minutes. Perhaps this is related to people no longer getting updates.

I'm on firmware version 1.0.11 and so far it's been alright with the corrections suggested above. Are you still having issues with yours? What firmware version are you on?

sjorge, 2023-08-31:
I swapped mine for a new unit that came with the updated fw out of the box.

It working fine now.

MattWestb, 2023-09-12:
My Vindstyrka is connected to Dirigera and is normally working well.
After last HA update the Air quality and PM2.5 was frozen and must restarting the Homekit client integration. Now some day after its the same AirQ and PM2.5 is frozen and must restart the HK.
Humidity and Temp is always reporting OK.

Im little confused if its one bug in HomeKit Client that is doing it then if it  was one Zigbee bug in the device the restart of HK shall not fixing it.

Only for information and it  can being  complete out  of scope but not sure of it.

nathan-gs, 2023-10-13:
I got a similar issue; the pm2.5 sensor stopped updating (kept on 0) while all other sensors keep updating (voc/humidity & temp). I'm running the latest OTA update.

MattWestb, 2023-10-14:
I have mine connected to Dirigera but and HA is keeping doing the same so im not sure if its device that do it or the host system (homekit integration).

@sjorge How is your device working ?

sjorge, 2023-10-14:
I stopped using it, it would still randomly stop reporting and the lack of precision in temperature was annoying.
It's in my box of random devices I keep around for testing if needed but won't use... which is quite big at this point.

MattWestb, 2023-10-14:
I agree with you the temperature sensor is 0.5°C and IKEA is only reporting it 1.0°C precision but the Humidity and PM2.5 is looking very OK.

closed 2024-05-12 by github-actions[bot]

MathijsG, 2026-08-16:
My Vindstyrka on firmware 1.0.11 doesn't seem to update any value at all. Only when I manually refresh each entity in Z2M (the refresh arrow icon) on the Exposes page, the value gets updated.

andrei-lazarov, 2026-08-16:
@MathijsG press Interview and Reconfigure. My units work

MathijsG, 2026-08-16:
> [@MathijsG](https://github.com/MathijsG) press Interview and Reconfigure. My units work

Unfortunately it doesn't seem to respect the set intervals. Also tried the posted configs here. For now I have an automation that does a get on the mqtt topic for the sensor. With this I get very responsive sensor values.

For people that are interested as well, this is the automation:

```
alias: VINDSTYRKA uitlezen
description: Lees de VINDSTYRKA iedere minuut actief uit via Zigbee2MQTT
triggers:
  - trigger: time_pattern
    minutes: "/1"

conditions: []

actions:
  - action: mqtt.publish
    data:
      topic: zigbee2mqtt/Luchtkwaliteitssensor/get
      payload: >-
        {
          "temperature": "",
          "humidity": "",
          "pm25": "",
          "voc_index": ""
        }

mode: single

ThatTallGuy21, 2026-08-16:
Back in 2024 or early 2025 I updated the firmware on my conbee II coordinator and since then it's been rock solid. So for me, this issue is stale.