Koenkk/zigbee2mqtt #18141: Socket TuYa BSD29

mati1988r, 2023-06-29:
### What happened?

I have problem with this socket,  how reset  total energy ? comand in console not working

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

20221226

### Adapter

sonoff dongle p

fips21, 2023-08-01:
Any progress on this? I would really apreciate a possibility to reset energy meter for this device.

fips21, 2023-08-01:
Sending raw message in dev console with endpoint: 1, cluster: 0x00, command: 0 and  payload: {} did not help.

closed 2023-09-21 by github-actions[bot]

silvertoes93, 2023-12-29:
+1

dev console with endpoint: 1, cluster: 0x00, command: 0 and payload: {} did not help me either.

redpixy, 2024-01-09:
resetting the energy from the GUI doesn't seems to be possible currently. 

I managed to reset the energy by doing a hard reset on the switch. 
1. It involves removing the smart switch from the zigbee2mqtt network.
2. power off the smart switch from the mains.
3. turn on the smart switch and press and hold the pairing button for around 6secs and you will see the led flashing

If done correctly the energy should display 0

redpixy, 2024-01-09:
> Unfortunately that did not work for me.

did you completely factory reset the smart switch? TBH, it took me a couple reset it.

fips21, 2024-01-09:
I think so. As you recommended, I first deleted it from z2m, disconnected it from mains, put in again and hold the button for 6s. But something changed. After joining it had only 4,14kWh. Before it was 4,35kWh. So I have to do this 20 times to get it to 0kWh? :-)

redpixy, 2024-01-12:
now that's strange. mine resetted to 0 from 32kwh after the factory reset was done.

Hopefully someone figures out how to reset the kwh from the UI side.. its quite a pain to factory reset the plug physically.

NODeeJay, 2024-02-06:
> Hopefully someone figures out how to reset the kwh from the UI side.. its quite a pain to factory reset the plug physically.

I tried your suggestion but it seems to just activate pairing mode. I do not have any changes shown afterwards, also no reductions like fips21. I also paired it with a Tuya hub and wiped data on removal which did not help either.

redpixy, 2024-02-07:
> I tried your suggestion but it seems to just activate pairing mode. I do not have any changes shown afterwards, also no reductions like fips21. I also paired it with a Tuya hub and wiped data on [...]

you should turn off smartplug complete, aka disconnect from the power for at least a min.
Afterwhich turn on the plug, press and hold Button for over 6 seconds.
You should see the Smart Plug light will start blinking fast.

Just note this is the painful step, as you mention it sometimes enters the pairing mode which just means you gotta repeat the steps again...

fips21, 2024-02-07:
It has a red light for relay state and a blue light which is blinking in pairing mode. Which one should start blinking fast?

redpixy, 2024-02-08:
I can't remember which colour light, but the light will flash quite rapidly. Its different from the pairing flashing frequency.

kidekoop2016, 2024-03-05:
Hi there¡
Same problem here. Any update? I cannot reset energy

redpixy, 2024-03-06:
> Hi there¡ Same problem here. Any update? I cannot reset energy

try doing a physical hard factory reset on the device. i work for me.

Reset via the gui don't work.

silvertoes93, 2024-11-13:
Have tried the hard reset, but it just goes into pairing mode as soon as I turn it on.