Koenkk/zigbee2mqtt #18143: Interview Failed

2proof4u, 2023-06-29:
### What happened?

I am experiencing some problems Phillip hue motion sensors. 

So far within the last 2 weeks I have experienced 2 times where different sensors has failed that has worked for more than a year.

It reports interview fail with a little symbol on the picture.
The first one I could only get to work again if I removed it completely from Zigbee2Mqtt and put the sensor in pairing mode.

The second one I have not done anything with yet in the hope that it is a bug that can be fixed in Zigbee2Mqtt.
I don’t know what to look for and it is a bit annoying if it has to be completely removed first.

I don’t know if it is related but I have also experienced that several hue sensors suddenly stop working because nothing is in reporting menu.
If I enter the data from a working sensor in the reporting menu it starts to work again.

Another thing I noticed but might not be related is that I get a lot of warnings in the supervisor from Zigbee2Mqtt
WARNING (MainThread) [supervisor.addons.options] Unknown option 'port' for Zigbee2MQTT (45df7312_zigbee2mqtt)

Is there something I can do or look for to stop these problems.

### What did you expect to happen?

It just continue to work

### How to reproduce it (minimal and precise)

I dont know

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

20221226

### Adapter

Sonoff dongle plus (p)

closed 2023-08-06 by github-actions[bot]

rjhilgefort, 2023-10-31:
I'm seeing the same thing. @2proof4u what was your solution?

2proof4u, 2023-10-31:
I don't have a solution but it has been working for a while now regarding the specific problem.

however my Zigbee network now is a bit slow sometimes.
It takes up to 5 sec in the last rooms I enter before it turn on if I walk around in the house.
Sometimes my hue sensors also hang as "Detected" and I have to go past it again to get it to update and send "Clear" 

I have close to 200 units on my network and think it might be to big and this is causing my new problems. I differently don't miss routers.

Please write back if you find a solution to the original problem so Ill know if it start to happen again here.

BlutorangeX, 2024-02-16:
I have the same issues. Sometimes my Hue Motion Sensor just disconnects and disappears from z2mqtt. If i try to repair, it just tells me "failed to interview". I restarted z2mqtt several times but nothing has changed yet. Is there any solution?

2proof4u, 2024-02-16:
sorry I dont have a solution