Koenkk/zigbee2mqtt #18147: Trust ZLED-RGB9 trying to ping wrong endpoint/cluster

YannickvanVugt, 2023-06-29:
### What happened?

For a long time I have been trying to convince some family member to make the step from his Trust ICS-2000 to Home Assistant, but after multiple tries and having issues with the ZLED-RGB9 lights, I bought some myself to dive deeper into the issue.

The light is always reported to be offline after set availability period as it does not respond to pings. Diving deeper into the issue it seems like it tries to ping a non existing cluster.

> Failed to ping 'Inloop Slaapkamer Yannick Trust 1' (attempt 1/2, Read 0x00158d00011ec27a/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 21992 - 1 - 171 - 0 - 1 after 10000ms))

It seems like it is looking for endpoint 1, which only consists of the touchlink clusters, where as genBasic and other clusters live in endpoint 2. After which the ping fails and it states that the device is offline.

Another strange issue, which I don't think can be solved quite easily, is that it in the same room with the coordinator, as well as 3 other routers, it reports a very low LQI. Always <70 and it is only situated 2.5 meters from the coordinator and less than 15cm from another router (lightbulb). It seems to always respond, so it is not a big issue.

[image]

[image]

### What did you expect to happen?

That the light would stay online and not get greyed out in the Home Assistant front-end. 😂

### How to reproduce it (minimal and precise)

Not much needed, just add the light and make sure you have the availability feature enabled.

[Other people reporting the same issue](https://github.com/Koenkk/zigbee2mqtt/issues/8027#issuecomment-1272400998)

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

20230507

### Adapter

Slaesh's CC2652RB stick

### Debug log

Warning 2023-06-29 22:38:25Failed to ping 'Inloop Slaapkamer Yannick Trust 1' (attempt 1/2, Read 0x00158d00011ec27a/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 21992 - 1 - 55 - 0 - 1 after 10000ms))

YannickvanVugt, 2023-08-01:
Issue still there.

Koenkk, 2023-08-01:
Can you check with 1.32.2?

YannickvanVugt, 2023-08-01:
> Can you check with 1.32.2?

Awesome, seems to be fixed indeed! Thanks for fixing the issue!

Koenkk, 2023-08-02:
Yes, from the release notes:

- [#18121](https://github.com/Koenkk/zigbee2mqtt/issues/18121) Fix ping for devices which have genBasic on non first endpoint (@Koenkk)

closed 2023-08-02 by Koenkk