Koenkk/zigbee2mqtt #18151: BSD29 OTA update error

terierul, 2023-06-30:
### What happened?

Device = BSD29
https://www.zigbee2mqtt.io/devices/BSD29.html

In OTA menu update there is an error related with missing OTA endpoint (see attached printscreen)
[image]

### Zigbee2MQTT version

1.31.2-1

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

JokeRsevilla, 2023-07-08:
hello! I have the same problem

Outsidewall, 2023-07-24:
Hello, Same issue here as well.

fips21, 2023-08-01:
Same problem here.

sereda-oleg, 2023-09-18:
I have the same problem

Koenkk, 2023-09-18:
Could you check if the OTA issue is fixed with: https://gist.github.com/Koenkk/064a88d7b995bdf170121e916454be91

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

JokeRsevilla, 2023-09-18:
> Could you check if the OTA issue is fixed with: https://gist.github.com/Koenkk/064a88d7b995bdf170121e916454be91
>
> * save this as file next to `configuration.yaml` as `ext_converter.js`
> * add it to [...]

[image]
[image]

RomchikL, 2023-09-19:
> Could you check if the OTA issue is fixed with: https://gist.github.com/Koenkk/064a88d7b995bdf170121e916454be91

I have the same result as @JokeRsevilla. But I did not reconnect the switch, only restart z2m with this external converter.

Koenkk, 2023-09-19:
does this device support otas through the TuYa gateway?

RomchikL, 2023-09-19:
I don't have TuYa gateway, can't check this, maybe @JokeRsevilla or others will answer

sereda-oleg, 2023-09-20:
> Could you check if the OTA issue is fixed with: https://gist.github.com/Koenkk/064a88d7b995bdf170121e916454be91
>
> * save this as file next to `configuration.yaml` as `ext_converter.js`
> * add it to [...]

Failed to check if update available for 'xxxxxxxxxx' (Device didn't respond to OTA request)

... reconnect switch, restart z2mqtt, coordinator Sonoff ZBDongle-P.

RomchikL, 2023-10-15:
@terierul, @Outsidewall @fips21 do you have a Tuya gateway? Could you check ota?

terierul, 2023-10-15:
> @terierul, @Outsidewall @fips21 do you have a Tuya gateway? Could you check ota?

sorry. I use a conbee 2 stick

migor7, 2023-10-24:
Same problem here.

RomchikL, 2023-10-24:
> Same problem here.

Do you have a Tuya gateway to check ota updates?

mati1988r, 2023-10-24:
> > Same problem here.
> 
> Do you have a Tuya gateway to check ota updates?

I have and I check tomorrow

mysatbox, 2023-11-16:
I have a tuya gateway and my plugs the already arribei with the latest firmware v.3.0.0

Wendigogo, 2023-11-22:
I got the same error here too (Home Assistant, Z2M, SONOFF dongle) with a [BSD29_1](https://www.zigbee2mqtt.io/devices/BSD29_1.html#tuya-bsd29_1) model.
Informations I could get are :
Firmware date : 20210625 / firmware version : 500+TZSKT31BS102
And on "State" page I got : 
```
"child_lock": null,
"indicator_mode": null,
"power_outage_memory": null,
"update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
"update_available": null
```
These "null" states are suspicious.

daimonkor, 2023-11-25:
I didn't use external converter, but I got update error. Please fix error.

VerodinDev, 2023-11-28:
> I got the same error here too (Home Assistant, Z2M, SONOFF dongle) with a [BSD29_1](https://www.zigbee2mqtt.io/devices/BSD29_1.html#tuya-bsd29_1) model. Informations I could get are : Firmware date : 20210625 / firmware version : 500+TZSKT31BS102 And on "State" page I got :

Same setup here. I have 3 of these devices, all with the exact same problem. Zigbee2MQTT v1.33.2

`{
    "current": 2.73,
    "energy": 20.52,
    "linkquality": 60,
    "power": 638,
    "power_outage_memory": "on",
    "state": "ON",
    "voltage": 228,
    "child_lock": null,
    "indicator_mode": null,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null
}`

Blackduke77, 2024-01-21:
Hi I have around 10 of these just arrived and getting the same OTA errors, do we have a solution yet. this happens in main and dev branches

Blackduke77, 2024-01-22:
@Koenkk whats the way forward to get this resolved, do I need to get a Tuya gateway, I am happy to work with someone and do whats needed to get them working again in Z2M

Koenkk, 2024-01-22:
Someone has to check wether OTA updates work when using the original hub

MBRetrofit, 2024-02-04:
Any news about firmware update for 500+TZSKT31BS102
This from time to time need to be disconnected from main to work again. I don't know if is a HW or a SW issue.

alarik84, 2024-02-07:
Hi.
I don't think it's related to the gteway. I've got two Tuya plugs:
[image]
[image]
For first one, I was able to update, for the second, I constantly get 'device didn't respond to OTA request'.

z647230, 2024-02-09:
I got multiple of these plugs that all report as BSD29_1, some connected to ZBdongle E with zigbee2mqtt and some to a R7070 Tuya gateway with the Tuya App. 

- with zigbee2mqtt i get an error on ota
- with tuya app and r7070 i get no update available

Will start debugging but if anybody already has clue, please let me know...

HughODwyer90, 2024-02-14:
Same issue here. Commenting so I get updates.

BeeSci, 2024-02-14:
Same issue here. Subscribed to get updates.

BeeSci, 2024-02-17:
I switched to debug mode, forcefully removed the plug, rejoined it twice and tried to start the OTA which failed for the same reasons as mentioned above.
This is the result in log.txt. Hope, it helps.
`debug 2024-02-17 18:29:19: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"f8yju-1","value":true}'
info  2024-02-17 18:29:19: Zigbee: allowing new devices to join.
info  2024-02-17 18:29:20: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"f8yju-1"}'
debug 2024-02-17 18:29:30: Received Zigbee message from 'Steckdose_Dachkammer', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
info  2024-02-17 18:29:30: MQTT publish: topic 'zigbee2mqtt/Steckdose_Dachkammer', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2024-02-17T18:29:30+01:00","linkquality":10,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":232}'
warn  2024-02-17 18:29:32: Device '0x70b3d52b6002838e' left the network
info  2024-02-17 18:29:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b6002838e","ieee_address":"0x70b3d52b6002838e"},"type":"device_leave"}'
warn  2024-02-17 18:29:33: Device '0x70b3d52b6002838e' left the network
info  2024-02-17 18:29:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b6002838e","ieee_address":"0x70b3d52b6002838e"},"type":"device_leave"}'
info  2024-02-17 18:29:38: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b6002838e', payload '{"last_seen":"2024-02-17T18:29:38+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1}}'
info  2024-02-17 18:29:38: Device '0x70b3d52b6002838e' joined
info  2024-02-17 18:29:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b6002838e","ieee_address":"0x70b3d52b6002838e"},"type":"device_joined"}'
info  2024-02-17 18:29:38: Starting interview of '0x70b3d52b6002838e'
info  2024-02-17 18:29:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b6002838e","ieee_address":"0x70b3d52b6002838e","status":"started"},"type":"device_interview"}'
info  2024-02-17 18:29:38: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b6002838e', payload '{"last_seen":"2024-02-17T18:29:38+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1}}'
debug 2024-02-17 18:29:38: Device '0x70b3d52b6002838e' announced itself
info  2024-02-17 18:29:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b6002838e","ieee_address":"0x70b3d52b6002838e"},"type":"device_announce"}'
info  2024-02-17 18:29:38: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b6002838e', payload '{"last_seen":"2024-02-17T18:29:38+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1}}'
info  2024-02-17 18:29:39: Successfully interviewed '0x70b3d52b6002838e', device has successfully been paired
info  2024-02-17 18:29:39: Device '0x70b3d52b6002838e' is supported, identified as: TuYa Smart plug (with power monitoring by polling) (BSD29_1)
info  2024-02-17 18:29:39: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart plug (with power monitoring by polling)","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":7,"description":"Recover state after power outage","label":"Power outage memory","name":"power_outage_memory","property":"power_outage_memory","type":"enum","values":["on","off","restore"]},{"access":7,"description":"LED indicator mode","label":"Indicator mode","name":"indicator_mode","property":"indicator_mode","type":"enum","values":["off","off/on","on/off","on"]},{"access":1,"category":"diagnostic","description":"Instantaneous measured power","label":"Power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"category":"diagnostic","description":"Instantaneous measured electrical current","label":"Current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"category":"diagnostic","description":"Measured electrical potential value","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Sum of consumed energy","label":"Energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"features":[{"access":3,"description":"Enables/disables physical input on the device","label":"State","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"label":"Child lock","type":"lock"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"BSD29_1","options":[{"access":2,"description":"This device does not support reporting electric measurements so it is polled instead. The default poll interval is 60 seconds, set to -1 to disable.","label":"Measurement poll interval", [... cut ...]
debug 2024-02-17 18:31:19: Received Zigbee message from '0x70b3d52b6002838e', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":236}' from endpoint 1 with groupID 0
info  2024-02-17 18:31:19: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b6002838e', payload '{"child_lock":null,"current":0,"energy":0,"indicator_mode":null,"last_seen":"2024-02-17T18:31:19+01:00","linkquality":58,"power":0,"power_outage_memory":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":236}'
`

FSSDawid, 2024-03-04:
Hey Guys, 
Just to bump the problem I have the same problem with the device OTA, 
And on top of that it seems really unresponsive no idea why it reports LQI > 160 but then I can't communicate with it for 10 minutes then it's back again ... weird... 

[image]

`{
    "current": 0,
    "energy": 6.36,
    "indicator_mode": "off",
    "last_seen": "2024-03-04T10:18:14+01:00",
    "linkquality": 120,
    "power": 0,
    "power_outage_memory": "off",
    "state": "OFF",
    "voltage": 237,
    "child_lock": null,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null
}`

this is after trying ext converter

Arcaios, 2024-03-10:
It seems that this device doesn't supports OTA updates...

S474N, 2024-04-20:
Same problem here.

Kopetan4egX, 2024-05-06:
Same problem on TuYa BSD29_1

mrlordziemniak, 2024-07-01:
i have same problem with that socket, it is very laggy

tonbor, 2024-08-10:
any progress?

helljol, 2024-08-13:
> Someone has to check wether OTA updates work when using the original hub

Can someone with a Tuya gateway test if OTA is working for those plugs ?
I got 2 https://www.zigbee2mqtt.io/devices/LSPA9.html and do have the same issue.

Triskae, 2024-08-16:
Hello ! Did someone tried to re-pair it ? I have the same problem, delete it from Z2M, but cannot put it back into pairing mode ...

alarik84, 2024-08-16:
> Hello ! Did someone tried to re-pair it ? I have the same problem, delete it from Z2M, but cannot put it back into pairing mode ...

I checked with one reported as Elivco LSPA9, and it can be re-paired correctly on my configuration

z647230, 2024-08-16:
> Hello ! Did someone tried to re-pair it ? I have the same problem, delete it from Z2M, but cannot put it back into pairing mode ...

If it doesn’t repair, remove it from power, stop zha/z2m (whatever you are using) put plug back into power reset plug by keeping button for approx 20sec. Start zha/z2m and re-pair.

helljol, 2024-08-24:
> > Someone has to check wether OTA updates work when using the original hub
> 
> Can someone with a Tuya gateway test if OTA is working for those plugs ? I got 2 https://www.zigbee2mqtt.io/devices/LSPA9.html and do have the same issue.

replying to myself , looks like those Elvico plugs will never support OTA.
see: https://github.com/Koenkk/zigbee2mqtt/issues/21239#issuecomment-2199523305

b2un0, 2024-08-28:
> > Someone has to check wether OTA updates work when using the original hub
> 
> Can someone with a Tuya gateway test if OTA is working for those plugs ? I got 2 https://www.zigbee2mqtt.io/devices/LSPA9.html and do have the same issue.

already done here
https://github.com/Koenkk/zigbee2mqtt/issues/21239#issuecomment-2199523305

tonbor, 2024-08-28:
Stil wrong picture and polling with two devices Girier and Elvico with latest firmware. Not only ota!
[image]
[image]

Trashwarez, 2024-09-02:
External converter does not work anymore

alexsotoaguilera, 2024-10-13:
How can I change or force the model from BSD29_1 to TS011F_plug_1?

Once pairing with environment interferences, when plugged in it was added as TS011F_plug_1, but never happened again.

Trashwarez, 2024-10-13:
I posted an issue here, but did not receive any comment: https://github.com/Koenkk/zigbee2mqtt/issues/23818

davidclaverie, 2024-11-27:
I have same problem with my plus
Do you fond solution
Sincerely

tonbor, 2024-11-27:
no, with the lastest firmware still polling, wrong picture en no ota

tonbor, 2025-01-27:
Do not close it but make this device working correct with zigbee2mqtt

covers007, 2025-02-07:
Up please

tonbor, 2025-04-09:
do not stale but solve

Bjk8kds, 2025-10-19:
Yes

quensen, 2025-12-19:
Hello dear Bot,
Please don’t close it

Merry Xmas to everyone!

closed 2026-02-25 by github-actions[bot]