Koenkk/zigbee2mqtt #18153: ZTH02 small with cr2032 battery not well implemented

tonbor, 2023-06-30:
### What happened?

The ZTH02 SMALL cr2032 battery is not well implemented, battery state is wrong type

### What did you expect to happen?

To he well implented

### How to reproduce it (minimal and precise)

Just addcto zigbee2mqtt

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Debug log

[image]
[image]
[image]

tonbor, 2023-07-06:
Any news on this problem?

tonbor, 2023-07-23:
Is somebody going to look into this?

tonbor, 2023-08-18:
any solution?

francisp2, 2023-08-18:
Zigbee2MQTT shows the correct entities, Domoticz does not.  Question for Domoticz ?

closed 2023-09-26 by github-actions[bot]