Koenkk/zigbee2mqtt #18164: BAC-002-ALZB@FIX stop working after updates

Rick5o, 2023-07-01:
### What happened?

After update to 1.31.2-1 external converter stop working.
There is better implamantation of BAC-002-ALZB but it stop working after update. I don't know how to modify it to work with new zigbee2mqqt. I hope someone can modify it so it could be properly put to zigbee2mqqt 

error:

> /app/node_modules/zigbee-herdsman-converters/index.js:79
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const ota = require('zigbee-herdsman-converters/lib/ota');
const {repInterval, defaultBindGroup, OneJanuary2000} = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fz_moes_thermostat_custom = {
	cluster: 'manuSpecificTuya',
	type: ['commandGetData', 'commandSetDataResponse', 'commandDataResponse'],
	convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'moes_thermostat');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
		let temperature;
		/* See tuyaThermostat above for message structure comment */
		switch (dp) {
		case tuya.dataPoints.moesSchedule:
			return {
				program: [
					{weekdays_p1: value[0] + 'h:' + value[1] + 'm ' + value[2]/2 + '°C'},
					{weekdays_p2: value[3] + 'h:' + value[4] + 'm ' + value[5]/2 + '°C'},
					{weekdays_p3: value[6] + 'h:' + value[7] + 'm ' + value[8]/2 + '°C'},
					{weekdays_p4: value[9] + 'h:' + value[10] + 'm ' + value[11]/2 + '°C'},
					{saturday_p1: value[12] + 'h:' + value[13] + 'm ' + value[14]/2+ '°C'},
					{saturday_p2: value[15] + 'h:' + value[16] + 'm ' + value[17]/2 + '°C'},
					{saturday_p3: value[18] + 'h:' + value[19] + 'm ' + value[20]/2 + '°C'},
					{saturday_p4: value[21] + 'h:' + value[22] + 'm ' + value[23]/2 + '°C'},
					{sunday_p1: value[24] + 'h:' + value[25] + 'm ' + value[26]/2 + '°C'},
					{sunday_p2: value[27] + 'h:' + value[28] + 'm ' + value[29]/2 + '°C'},
					{sunday_p3: value[30] + 'h:' + value[31] + 'm ' + value[32]/2 + '°C'},
					{sunday_p4: value[33] + 'h:' + value[34] + 'm ' + value[35]/2 + '°C'},
				],
			};
		case 1: // system mode (OFF, ON)
			if(value == 0) {
				return {state: 'OFF'};
			} else if(value == 1) {
				return {state: 'ON'};
			} else {
				return {};
			}
		case 2: // system mode (cooling, heating, fan)
			if(value == 0) {
				return {system_mode: 'cool'};
			} else if(value == 1) {
				return {system_mode: 'heat'};
			} else if(value == 2) {
				return {system_mode: 'fan_only'};
			} else {
				return {};
			}
		case 28: // fan mode
			if(value == 0) {
				return {fan_mode: 'low'};
			} else if(value == 1) {
				return {fan_mode: 'medium'};
			} else if(value == 2) {
				return {fan_mode: 'high'};
			} else if(value == 3) {
				return {fan_mode: 'auto'};
			} else {
				return {};
			}
		case 16: // temperature setpoint for both heating and cooling
			return {occupied_heating_setpoint: value};
		case tuya.dataPoints.moesChildLock:
			return {child_lock: value ? 'LOCK' : 'UNLOCK'};
		case tuya.dataPoints.moesMaxTempLimit:
			return {max_temperature_limit: value};
		case tuya.dataPoints.moesMaxTemp:
			return {max_temperature: value};
		case tuya.dataPoints.moesDeadZoneTemp:
			return {deadzone_temperature: value};
		case tuya.dataPoints.moesLocalTemp:
			temperature = value & 1<<15 ? value - (1<<16) + 1 : value;
			return {local_temperature: parseFloat((temperature / 10).toFixed(1))};
		case tuya.dataPoints.moesTempCalibration:
			temperature = value;
			// for negative values produce complimentary hex (equivalent to negative values)
			if (temperature > 4000) temperature = temperature - 4096;
			return {local_temperature_calibration: temperature};
		case tuya.dataPoints.moesHold: // state is inverted, preset_mode is deprecated
			return {preset_mode: value ? 'program' : 'hold', preset: value ? 'program' : 'hold'};
		case tuya.dataPoints.moesScheduleEnable: // state is inverted, preset_mode is deprecated
			return {preset_mode: value ? 'hold' : 'program', preset: value ? 'hold' : 'program'};
		case tuya.dataPoints.moesValve:
			return {heat: value ? 'OFF' : 'ON'};
		case tuya.dataPoints.moesSensor:
			switch (value) {
			case 0:
				return {sensor: 'IN'};
			case 1:
				return {sensor: 'AL'};
			case 2:
				return {sensor: 'OU'};
			default:
				return {sensor: 'not_supported'};
			}
		default: // DataPoint 17 is unknown
			meta.logger.warn(`zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #${
				dp} with data ${JSON.stringify(msg.data)}`);
		}
	},
};

const tz_state = {
	key: ['state'],
	convertSet: async (entity, key, value, meta) => {
		if(value == "OFF") {
			await tuya.sendDataPointBool(entity, 1, false);
		} else if(value == "ON") {
			await tuya.sendDataPointBool(entity, 1, true);
		}
	},
};

const tz_system_mode = {
	key: ['system_mode'],
	convertSet: async (entity, key, value, meta) => {
		if(value == "cool") {
			await tuya.sendDataPointEnum(entity, 2, 0);
		} else if(value == "heat") {
			await tuya.sendDataPointEnum(entity, 2, 1);
		} else if(value == "fan_only") {
			await tuya.sendDataPointEnum(entity, 2, 2);
		}
	},
};

const tz_temperature_setpoint = {
	key: ['occupied_heating_setpoint'],
	convertSet: async (entity, key, value, meta) => {
		await tuya.sendDataPointValue(entity, 16, value);
	},
};

const tz_fan_mode = {
	key: ['fan_mode'],
	convertSet: async (entity, key, value, meta) => {
		if(value == "low") {
			await tuya.sendDataPointEnum(entity, 28, 0);
		} else if(value == "medium") {
			await tuya.sendDataPointEnum(entity, 28, 1);
		} else if(value == "high") {
			await tuya.sendDataPointEnum(entity, 28, 2);			
		} else if(value == "auto") {
			await tuya.sendDataPointEnum(entity, 28, 3);			
		}
	},
};

module.exports = [
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_dzuqwsyg'}],
        model: 'BAC-002-ALZB',
        vendor: 'HKGK',
        description: 'BAC series thermostat',
        fromZigbee: [fz_moes_thermostat_custom],
		onEvent: tuya.onEventSetLocalTime,
        toZigbee: [tz_state, tz_system_mode, tz_fan_mode, tz_temperature_setpoint, tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode,
            tz.hgkg_thermostat_standby, tz.moes_thermostat_sensor, tz.moes_thermostat_calibration,
            tz.moes_thermostat_deadzone_temperature, tz.moes_thermostat_max_temperature_limit],
			exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(),
			exposes.climate()
			.withSetpoint('occupied_heating_setpoint', 16, 35, 0.1, ea.STATE_SET)
			.withLocalTemperature(ea.STATE)
			.withLocalTemperatureCalibration(ea.STATE_SET)
			.withSystemMode(['cool', 'heat', 'fan_only'], ea.STATE_SET)
			.withFanMode(['low', 'medium', 'high', 'auto', 'off'], ea.STATE_SET)
			.withRunningState(['idle', 'heat', 'cool'], ea.STATE)
			.withPreset(['hold', 'program']),
			e.temperature_sensor_select(['IN', 'AL', 'OU']),
			
			exposes.composite('programming_mode', 'programming_mode', ea.STATE)
                .withDescription(
                    'Schedule MODE ⏱ - In this mode, the device executes a preset week programming temperature time and temperature.',
                )
                .withFeature(e.week())
                .withFeature(exposes.text('workdays_schedule', ea.STATE_SET))
                .withFeature(exposes.text('holidays_schedule', ea.STATE_SET)),
			
			exposes.switch()
				.withState('state', false, 'OFF/ON state'),
	],
    },
];`

### What did you expect to happen?

I except to get extended function of BAC-002

### How to reproduce it (minimal and precise)

run new zigbee2mqqt

### Zigbee2MQTT version

1.31.2-1

### Adapter firmware version

20221226

### Adapter

ZBDongle-P

Koenkk, 2023-07-02:
- In the top of the file add `const legacy = require('zigbee-herdsman-converters/lib/legacy');`
- Try changing all `tz.` to `legacy.tz.`

Rick5o, 2023-07-02:
config:
`const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const ota = require('zigbee-herdsman-converters/lib/ota');
const {repInterval, defaultBindGroup, OneJanuary2000} = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fz_moes_thermostat_custom = {
	cluster: 'manuSpecificTuya',
	type: ['commandGetData', 'commandSetDataResponse', 'commandDataResponse'],
	convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'moes_thermostat');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
		let temperature;
		/* See tuyaThermostat above for message structure comment */
		switch (dp) {
		case tuya.dataPoints.moesSchedule:
			return {
				program: [
					{weekdays_p1: value[0] + 'h:' + value[1] + 'm ' + value[2]/2 + '°C'},
					{weekdays_p2: value[3] + 'h:' + value[4] + 'm ' + value[5]/2 + '°C'},
					{weekdays_p3: value[6] + 'h:' + value[7] + 'm ' + value[8]/2 + '°C'},
					{weekdays_p4: value[9] + 'h:' + value[10] + 'm ' + value[11]/2 + '°C'},
					{saturday_p1: value[12] + 'h:' + value[13] + 'm ' + value[14]/2+ '°C'},
					{saturday_p2: value[15] + 'h:' + value[16] + 'm ' + value[17]/2 + '°C'},
					{saturday_p3: value[18] + 'h:' + value[19] + 'm ' + value[20]/2 + '°C'},
					{saturday_p4: value[21] + 'h:' + value[22] + 'm ' + value[23]/2 + '°C'},
					{sunday_p1: value[24] + 'h:' + value[25] + 'm ' + value[26]/2 + '°C'},
					{sunday_p2: value[27] + 'h:' + value[28] + 'm ' + value[29]/2 + '°C'},
					{sunday_p3: value[30] + 'h:' + value[31] + 'm ' + value[32]/2 + '°C'},
					{sunday_p4: value[33] + 'h:' + value[34] + 'm ' + value[35]/2 + '°C'},
				],
			};
		case 1: // system mode (OFF, ON)
			if(value == 0) {
				return {state: 'OFF'};
			} else if(value == 1) {
				return {state: 'ON'};
			} else {
				return {};
			}
		case 2: // system mode (cooling, heating, fan)
			if(value == 0) {
				return {system_mode: 'cool'};
			} else if(value == 1) {
				return {system_mode: 'heat'};
			} else if(value == 2) {
				return {system_mode: 'fan_only'};
			} else {
				return {};
			}
		case 28: // fan mode
			if(value == 0) {
				return {fan_mode: 'low'};
			} else if(value == 1) {
				return {fan_mode: 'medium'};
			} else if(value == 2) {
				return {fan_mode: 'high'};
			} else if(value == 3) {
				return {fan_mode: 'auto'};
			} else {
				return {};
[... 103 lines cut ...]
			.withLocalTemperature(ea.STATE)
			.withLocalTemperatureCalibration(ea.STATE_SET)
			.withSystemMode(['cool', 'heat', 'fan_only'], ea.STATE_SET)
			.withFanMode(['low', 'medium', 'high', 'auto', 'off'], ea.STATE_SET)
			.withRunningState(['idle', 'heat', 'cool'], ea.STATE)
			.withPreset(['hold', 'program']),
			e.temperature_sensor_select(['IN', 'AL', 'OU']),
			
			exposes.composite('programming_mode', 'programming_mode', ea.STATE)
                .withDescription(
                    'Schedule MODE ⏱ - In this mode, the device executes a preset week programming temperature time and temperature.',
                )
                .withFeature(e.week())
                .withFeature(exposes.text('workdays_schedule', ea.STATE_SET))
                .withFeature(exposes.text('holidays_schedule', ea.STATE_SET)),
			
			exposes.switch()
				.withState('state', false, 'OFF/ON state'),
	],
    },
];`

When turning on system (fan):  Exception while calling fromZigbee converter: tuya.firstDpValue is not a function}
When changing heat/cool/fan: Exception while calling fromZigbee converter: tuya.firstDpValue is not a function}
When changing temperature: Publish 'set' 'occupied_heating_setpoint' to '0x9035eafffe0c4655' failed: 'TypeError: tuya.sendDataPointValue is not a function'

Rick5o, 2023-07-05:
Default setting without external converters:
I can't switch heat/cooling/fan mode only/off.  There is only cooling/off and in my case cooling is actually heating.  
I can't switch off fan when cooling/heating is on.

snagornyy, 2023-07-23:
I have the same problem too (((
When turning on system (fan): Exception while calling fromZigbee converter: tuya.firstDpValue is not a function}
When changing heat/cool/fan: Exception while calling fromZigbee converter: tuya.firstDpValue is not a function}
When changing temperature: Publish 'set' 'occupied_heating_setpoint' to '0x9035eafffe0c4655' failed: 'TypeError: tuya.sendDataPointValue is not a function'

closed 2023-08-30 by github-actions[bot]

piposant, 2023-09-06:
Hello
Same issue on my side.

gciman, 2023-09-16:
Hello
Same issue on my side.

Kalltech, 2024-01-07:
Hi guys,
Same issue on my side.
Does anyone find a solution?
Thanks

seakrebel, 2024-02-25:
Same issue here

Rick5o, 2024-02-26:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =
{
    fingerprint: [{
            modelID: 'TS0601',
            manufacturerName: '_TZE200_dzuqwsyg',
        },
    ],
    model: 'BAC-002-ALZB',
    vendor: 'HKGK',
    description: 'BAC series thermostat',

    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEvent({timeStart: '2000'}),
    configure: tuya.configureMagicPacket,
    exposes: [
        e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the thermostat ON/OFF'),
        e.child_lock(),
        e.climate()
            .withLocalTemperature(ea.STATE)
            .withSetpoint('current_heating_setpoint', 10, 35, 1, ea.STATE_SET)
            .withSystemMode(['cool', 'heat', 'fan_only'], ea.STATE_SET)
            .withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
            .withLocalTemperatureCalibration(-5, 5, 0.5, ea.STATE_SET),
        e.min_temperature().withValueMin(5).withValueMax(15),
        e.max_temperature().withValueMin(15).withValueMax(35),
        e.binary('eco_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('ECO mode ON/OFF'),
        e.max_temperature_limit().withDescription('ECO Heating energy-saving temperature (default: 20 ºC)').withValueMin(15).withValueMax(30),
        e.min_temperature_limit().withDescription('ECO Cooling energy-saving temperature (default: 26 ºC)').withValueMin(15).withValueMax(30),
        e.deadzone_temperature().withValueMin(1).withValueMax(5).withValueStep(1),
        e.binary('valve', ea.STATE, 'CLOSED', 'OPEN').withDescription('3-Way Valve status'),
        e.binary('manual_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('Manual = ON or Schedule = OFF'),
        ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
    ],
    meta: {
        tuyaDatapoints:
        [
            [1, 'state', tuya.valueConverter.onOff],
            [2, 'system_mode', tuya.valueConverterBasic.lookup({'cool': tuya.enum(0), 'heat': tuya.enum(1), 'fan_only': tuya.enum(2)})],
            [4, 'eco_mode', tuya.valueConverter.onOff],
            [16, 'current_heating_setpoint', tuya.valueConverter.raw],
            [19, 'max_temperature', tuya.valueConverter.raw],
            [24, 'local_temperature', tuya.valueConverter.divideBy10],
            [26, 'min_temperature', tuya.valueConverter.raw],
            [27, 'local_temperature_calibration', tuya.valueConverter.localTemperatureCalibration],
            [28, 'fan_mode', tuya.valueConverterBasic.lookup(
                {'low': tuya.enum(0), 'medium': tuya.enum(1), 'high': tuya.enum(2), 'auto': tuya.enum(3)})],
            [36, 'valve', tuya.valueConverterBasic.lookup({'OPEN': 0, 'CLOSE': 1})],
            [40, 'child_lock', tuya.valueConverter.lockUnlock],
            [103, 'deadzone_temperature', tuya.valueConverter.divideBy10],
            [104, 'min_temperature_limit', tuya.valueConverter.raw],
            [105, 'max_temperature_limit', tuya.valueConverter.raw],
            [106, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [108, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [109, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [110, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [111, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
            [112, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
            [101, 'manual_mode', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;

`

seakrebel, 2024-02-26:
I think I have resolved it by replacing all occurrences of `tuya.` with `legacy.` and adding the import
```
const legacy = require('zigbee-herdsman-converters/lib/legacy');
```
Also switched to `dev` branch of zigbee2mqtt to ensure it works. So far it throws no errors and thermostat works as it used to.

Rick5o, 2024-02-28:
> I think I have resolved it by replacing all occurrences of `tuya.` with `legacy.` and adding the import
>
> ```
> const legacy = require('zigbee-herdsman-converters/lib/legacy');
> ```
>
> Also switched to [...]

Please share finished code :)