Koenkk/zigbee2mqtt #18167: Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '63271'} after update to 1.32.0

Jeepmg, 2023-07-01:
### What happened?

I got this error messages :+1: 

error 2023-07-01 13:59:01Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '63271'}
error 2023-07-01 14:00:00Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '22860'}
error 2023-07-01 14:01:03Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '10673'}
error 2023-07-01 14:06:01Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '22860'}
error 2023-07-01 14:06:21Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '63271'}
error 2023-07-01 14:07:32Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '10673'}
error 2023-07-01 14:11:46Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '22860'}
error 2023-07-01 14:13:39Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '63271'}
error 2023-07-01 14:15:12Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '10673'}
error 2023-07-01 14:19:57Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '22860'}
error 2023-07-01 14:21:38Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '63271'}
error 2023-07-01 14:21:54Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '10673'}
error 2023-07-01 14:25:42Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '22860'}
error 2023-07-01 14:29:00Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '10673'}
error 2023-07-01 14:30:12Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '63271'}

### What did you expect to happen?

nothing

### How to reproduce it (minimal and precise)

update to 1.32.0

### Zigbee2MQTT version

1.32.0

### Adapter firmware version

20220219

### Adapter

zStack3x0

### Debug log

error 2023-07-01 13:59:01Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '63271'}
error 2023-07-01 14:00:00Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '22860'}
error 2023-07-01 14:01:03Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '10673'}
error 2023-07-01 14:06:01Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '22860'}
error 2023-07-01 14:06:21Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '63271'}
error 2023-07-01 14:07:32Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '10673'}
error 2023-07-01 14:11:46Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '22860'}
error 2023-07-01 14:13:39Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '63271'}
error 2023-07-01 14:15:12Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '10673'}
error 2023-07-01 14:19:57Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '22860'}
error 2023-07-01 14:21:38Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '63271'}
error 2023-07-01 14:21:54Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '10673'}
error 2023-07-01 14:25:42Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '22860'}
error 2023-07-01 14:29:00Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '10673'}
error 2023-07-01 14:30:12Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '63271'}

oopiicaa, 2023-07-01:
Well, update to 1.32.0 (1.32.0-1 to be exact as it is HA add-on) completely broke down my Zigbee network - Z2M does start normally and everything looks normal (log messages from devices), it's just that MQTT (EMQX) shows all entities as unavailable. I reverted back to 1.31.2, where entities are okay again, but I have a weird problem now with buttons not detecting clicks (multiple Aqara WXKG01LM and IKEA Tradfri E1743 for now) ... Something wen't completely wrong with this update - at least for me.

eljotpl, 2023-07-01:
Hello,

I have the same issue like @oopiicaa. In theory everything works, but all devices are listed as unavaliable. My config - sonoff ZBDongle-E with multi-pan firmware, SiliconLabs Zigbee/OpenThread Multiprotocol Add-on.

Jeepmg, 2023-07-01:
Flash Coordinator to 20230507 : no change

Aqara SSM-U02 may be the cause of the problem

avincon, 2023-07-01:
I do have the same issue, I was able to fix some of the devices by repairing them, however I'm still not able to get an "IKEA E2007" to work. In my case the error message is "... Expected one of: 0, 1, got: '2' " and occurs every 30 seconds. The overall performance of the network is pretty slow.
I'm running a Sonoff Zigbee Dongle with the 20230507 firmware.

Koenkk, 2023-07-02:
Please provide the debug log.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

avincon, 2023-07-03:
Sorry for the late reply - at least for me the problems seem to have disappeared with a longer system uptime, right now everything is working as expected. I've enabled debug logging now, so if I'm getting error messages again I'll provide the debug log. Thank you very much for your work!

Jeepmg, 2023-07-05:
Sorry I was far from home this last days :

2023-07-05 14:13:02Received Zigbee message from 'Led Gauche', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,0,3,40,24,5,33,8,0,154,32,0,8,33,22,1,10,33,39,247,11,32,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
2023-07-05 14:13:02SSM-U02: Processed buffer into data {"3":24,"5":8,"8":278,"10":63271,"11":0,"100":0,"154":0}
2023-07-05 14:13:02Error: Expected one of: 1, 2, got: '63271' at getFromLookup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:512:37) at numericAttributes2Payload (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/xiaomi.ts:213:52) at Object.numericAttributes2Payload (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/xiaomi.ts:478:42) at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/converters/fromZigbee.js:3967:33) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:150:51) at EventEmitter.emit (node:events:525:35) at EventBus.emitDeviceMessage (/opt/zigbee2mqtt/lib/eventBus.ts:102:22) at Controller.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.ts:106:27) at Controller.emit (node:events:513:28) at Controller.selfAndDeviceEmit (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
2023-07-05 14:13:15Saving state to file /opt/zigbee2mqtt/data/state.json
2023-07-05 14:14:04Received Zigbee message from 'Guirlande', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
2023-07-05 14:14:40Received Zigbee message from 'Lampadaire', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":234}' from endpoint 1 with groupID 0
2023-07-05 14:14:40Received Zigbee message from 'Lampadaire', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,104]}' from endpoint 1 with groupID 0
2023-07-05 14:14:55Received Zigbee message from 'Chargeur', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
2023-07-05 14:15:01Received Zigbee message from 'Sapin', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
2023-07-05 14:15:09Received Zigbee message from 'Lampadaire', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
2023-07-05 14:15:53Received Zigbee message from 'Couloir', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,0,3,40,24,5,33,9,0,154,32,0,8,33,24,1,11,32,0,10,33,76,89,12,32,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
2023-07-05 14:15:53SSM-U02: Processed buffer into data {"3":24,"5":9,"8":280,"10":22860,"11":0,"12":1,"100":0,"154":0}
2023-07-05 14:15:53Error: Expected one of: 1, 2, got: '22860' at getFromLookup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:512:37) at numericAttributes2Payload (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/xiaomi.ts:213:52) at Object.numericAttributes2Payload (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/xiaomi.ts:478:42) at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/converters/fromZigbee.js:3967:33) at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:150:51) at EventEmitter.emit (node:events:525:35) at EventBus.emitDeviceMessage (/opt/zigbee2mqtt/lib/eventBus.ts:102:22) at Controller.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.ts:106:27) at Controller.emit (node:events:513:28) at Controller.selfAndDeviceEmit (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
2023-07-05 14:16:35Received Zigbee message from 'Thermo Chambre', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":4105,"imageType":2053,"manufacturerCode":4742}' from endpoint 1 with groupID 0
2023-07-05 14:16:35Device 'Thermo Chambre' requested OTA
2023-07-05 14:16:35Responded to OTA request of 'Thermo Chambre' with 'NO_IMAGE_AVAILABLE'
2023-07-05 14:16:47Received Zigbee message from 'Guirlande', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
2023-07-05 14:17:00Received Zigbee message from 'Lampadaire', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
2023-07-05 14:17:14Received Zigbee message from 'Chargeur', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":234}' from endpoint 1 with groupID 0
2023-07-05 14:17:14Received Zigbee message from 'Chargeur', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,116]}' from endpoint 1 with groupID 0
2023-07-05 14:17:14Received Zigbee message from 'Thermo Chambre', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2400}' from endpoint 1 with groupID 0
2023-07-05 14:17:31Received Zigbee message from 'Sapin', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":235}' from endpoint 1 with groupID 0
2023-07-05 14:17:31Received Zigbee message from 'Sapin', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,17]}' from endpoint 1 with groupID 0
[... cut ...]
2023-07-05 14:18:02Received Zigbee message from 'Guirlande', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":233}' from endpoint 1 with groupID 0
2023-07-05 14:18:02Received Zigbee message from 'Guirlande', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,27]}' from endpoint 1 with groupID 0
2023-07-05 14:18:15Saving state to file /opt/zigbee2mqtt/data/state.json

**Aqara SSM-U02 may be the cause of the problem**

Koenkk, 2023-07-05:
@Jeepmg that should be fixed in the dev branch  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-08-14 by github-actions[bot]