Koenkk/zigbee2mqtt #18168: set color_temp no longer works with 1.32.0

stickpin, 2023-07-01:
### What happened?

The Home Assistant sends a command to set the color temperature to 3200K to a group of bulbs.
Before release 1.32.0 it was working fine, now I am getting an error and some bulbs are set to "cold white" instead of "warm white".

Any idea what's changed in the latest version and how to solve it?

### Zigbee2MQTT version

1.32.0

### Adapter firmware version

20230507

### Adapter

Slaesh's CC2652RB stick

### Debug log

```
Debug 2023-07-01 14:46:54 Received MQTT message on 'zigbee2mqtt/Living Room Main and Table Lights ZGroup/set' with data '{"state":"ON","brightness":254,"color_temp":312}'
Debug 2023-07-01 14:46:54 Publishing 'set' 'brightness' to 'Living Room Main and Table Lights ZGroup'
Debug 2023-07-01 14:46:54 Publishing 'set' 'color_temp' to 'Living Room Main and Table Lights ZGroup'
Error 2023-07-01 14:46:55 Publish 'set' 'color_temp' to 'Living Room Main and Table Lights ZGroup' failed: 'TypeError: Cannot read properties of undefined (reading 'hasOwnProperty')'
Info 2023-07-01 14:46:55MQTT publish: topic 'zigbee2mqtt/Living Room Main and Table Lights ZGroup', payload '{"brightness":254,"color":{"h":36,"hue":36,"s":70,"saturation":70,"x":0.4234,"y":0.399},"color_mode":"color_temp","color_temp":312,"state":"ON"}'
```

ExMacro, 2023-07-01:
I just noticed this too since my automations change the light colour temp to warmer when the night comes. Now a group with different lights doesn't work anymore. A group with only similar lights still seems to work. Changing the colour temp of a group used to work very well.

oscrey, 2023-07-01:
Same error for me too
`error 2023-07-02 00:48:30Publish 'set' 'color_temp' to 'ilg' failed: 'TypeError: Cannot read properties of undefined (reading 'hasOwnProperty')'`
`info 2023-07-02 00:48:30MQTT publish: topic 'zigbee2mqtt_efr32/bridge/log', payload '{"message":"Publish 'set' 'color_temp' to 'ilg' failed: 'TypeError: Cannot read properties of undefined (reading 'hasOwnProperty')'","meta":{"friendly_name":"ilg"},"type":"zigbee_publish_error"}'`

savdagod, 2023-07-02:
Same thing happens to me, rolling back to 1.31 until this is fixed.

jsenecal, 2023-07-02:
I also noticed a similar issue with similar output than both @oscrey and @stickpin. Unfortunately, I cannot seem to be able to rollback my Hassio installation of the addon as downgrades are not supported...

Koenkk, 2023-07-02:
@stickpin can you provide more of the debug logging after this? This should contain the stack trace.

craigmdennis, 2023-07-02:
```
zigbee2MQTT:error 2023-07-02 11:02:43: Publish 'set' 'color_temp' to 'Hall Lights' failed: 'TypeError: Cannot read properties of undefined (reading 'hasOwnProperty')'
```

Also 

```
Zigbee2MQTT:error 2023-07-02 11:02:41: Cannot find '0x0c4314fffe328343/1' of group 'Office Bookshelf Lights'
```

As well as direct binding of zigbee devices seemed to have issues with on/off toggle. I'm unsure which would be part of the latest update.

ExMacro, 2023-07-02:
> zigbee2MQTT:error 2023-07-02 11:02:43: Publish 'set' 'color_temp' to 'Hall Lights' failed: 'TypeError: Cannot read properties of undefined (reading 'hasOwnProperty')'

I get this exactly same error with my setup for light groups.

stickpin, 2023-07-02:
@Koenkk I guess that's what you looking for:
```
Error 2023-07-02 11:20:52Publish 'set' 'color_temp' to 'Living Room Main and Table Lights ZGroup' failed: 'TypeError: Cannot read properties of undefined (reading 'hasOwnProperty')'
Debug 2023-07-02 11:20:52TypeError: Cannot read properties of undefined (reading 'hasOwnProperty') at Function.fromObject (/app/node_modules/zigbee-herdsman-converters/src/lib/color.ts:226:17) at Function.fromMireds (/app/node_modules/zigbee-herdsman-converters/src/lib/color.ts:239:24) at Object.syncColorState (/app/node_modules/zigbee-herdsman-converters/src/lib/color.ts:773:32) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.js:1033:33) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.js:1182:32) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:248:36)
```

Please let me know if you need more details.

JimmyGiant, 2023-07-03:
Got the same issue with basically the same error-code:
`Publish 'set' 'color_temp' to 'Buero' failed: 'TypeError: Cannot read properties of undefined (reading 'hasOwnProperty')'`

holstweb, 2023-07-03:
Similar problem here. Some lights respond as usual, some Hue lights give errors in a Zigbee Group.

`Zigbee2MQTT:error 2023-07-03 15:41:26: Publish 'set' 'color_temp' to 'group_eettafel' failed: 'TypeError: Cannot read properties of undefined (reading 'hasOwnProperty')'
Zigbee2MQTT:debug 2023-07-03 15:41:26: TypeError: Cannot read properties of undefined (reading 'hasOwnProperty')
    at Function.fromObject (/app/node_modules/zigbee-herdsman-converters/src/lib/color.ts:226:17)
    at Function.fromMireds (/app/node_modules/zigbee-herdsman-converters/src/lib/color.ts:239:24)
    at Object.syncColorState (/app/node_modules/zigbee-herdsman-converters/src/lib/color.ts:773:32)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.js:1033:33)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:248:36)
Zigbee2MQTT:info  2023-07-03 15:41:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'color_temp' to 'group_eettafel' failed: 'TypeError: Cannot read properties of undefined (reading 'hasOwnProperty')'","meta":{"friendly_name":"group_eettafel"},"type":"zigbee_publish_error"}'`

ahd71, 2023-07-03:
Same issue here. Preliminary analys is that it is problem for a group having mixed type of devices (probably with and without support for color temp) while another group with same type of devices indeed still can set color temp.

stickpin, 2023-07-03:
@ahd71 that's correct. My groups also got a mixture of bulbs some are with color_temp attribute and some are just regular WW (warm white) bulbs.
It seems to be related to the code refactoring that was done last month:
https://github.com/Koenkk/zigbee-herdsman-converters/commits/master/src/lib/color.ts

evangelo18, 2023-07-04:
same issue here
Debug 2023-07-04 22:27:12Received MQTT message on 'zigbee2mqtt/study_downlights/set' with data '{"state":"ON","transition":1.0,"color_temp":465}'
Debug 2023-07-04 22:27:12Skipping state because of Home Assistant
Debug 2023-07-04 22:27:12Publishing 'set' 'transition' to 'study_downlights'
Debug 2023-07-04 22:27:12Publishing 'set' 'color_temp' to 'study_downlights'
Debug 2023-07-04 22:27:12Requested color_temp 465 is higher than maximum supported 0, using maximum!
Error 2023-07-04 22:27:12Publish 'set' 'color_temp' to 'study_downlights' failed: 'TypeError: Cannot read properties of undefined (reading 'hasOwnProperty')'
Debug 2023-07-04 22:27:12TypeError: Cannot read properties of undefined (reading 'hasOwnProperty') at Function.fromObject (/app/node_modules/zigbee-herdsman-converters/src/lib/color.ts:226:17) at Function.fromMireds (/app/node_modules/zigbee-herdsman-converters/src/lib/color.ts:239:24) at Object.syncColorState (/app/node_modules/zigbee-herdsman-converters/src/lib/color.ts:773:32) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.js:1033:33) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:248:36)

stickpin, 2023-07-04:
@Koenkk thanks a lot for a quick fix!
I can confirm the issue has been resolved.

ahd71, 2023-07-04:
I also confirm it resolved the issue. Thank you very much!

Koenkk, 2023-07-04:
Awesome! I will try to make a hotfix release in the coming days.

closed 2023-07-04 by Koenkk