Koenkk/zigbee2mqtt #18169: Exception while calling fromZigbee converter: 'energy_calibration' is not a number, got string ()}

Nardol, 2023-07-01:
### What happened?

Since version 1.32.0 I have errors about "Exception while calling fromZigbee converter: 'energy_calibration' is not a number, got string ()}"

### What did you expect to happen?

I did not have these error before

### How to reproduce it (minimal and precise)

I did not something else than starting Zigbee2MQTT after I updated it.

### Zigbee2MQTT version

1.32.0

### Adapter firmware version

20230507

### Adapter

Sonoff dongle 2652P not ezsp

### Debug log

debug 01/07/2023 14:48:51: Received Zigbee message from 'prise_entree', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
debug 01/07/2023 14:49:04: Received Zigbee message from 'prise_meuble_bureau', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
debug 01/07/2023 14:49:37: Received Zigbee message from 'prise_meuble_bureau', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,1]}' from endpoint 1 with groupID 0
info  01/07/2023 14:49:37: MQTT publish: topic 'zigbee2mqtt/prise_meuble_bureau', payload '{"child_lock":"UNLOCK","current":0,"elapsed":60079,"energy":0.01,"indicator_mode":"off/on","last_seen":"2023-05-21T16:39:03+02:00","linkquality":58,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":240}'
debug 01/07/2023 14:49:37: Received Zigbee message from 'prise_entree', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,39]}' from endpoint 1 with groupID 0
error 01/07/2023 14:49:37: Exception while calling fromZigbee converter: 'energy_calibration' is not a number, got string ()}
debug 01/07/2023 14:49:37: Error: 'energy_calibration' is not a number, got string ()
    at assertNumber (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:502:42)
    at calibrateAndPrecisionRoundOptions (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:90:5)
    at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/converters/fromZigbee.js:768:34)
    at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/opt/zigbee2mqtt/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
debug 01/07/2023 14:49:37: Received Zigbee message from 'Prise Salle de Bain', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":1,"rmsCurrent":12,"rmsVoltage":238}' from endpoint 1 with groupID 0
info  01/07/2023 14:49:37: MQTT publish: topic 'zigbee2mqtt/Prise Salle de Bain', payload '{"current":0.01,"elapsed":60086,"energy":30.27,"indicator_mode":"off/on","last_seen":"2023-05-21T16:38:27+02:00","linkquality":80,"power":1,"power_outage_memory":"restore","state":"ON","voltage":238}'
debug 01/07/2023 14:49:38: Received Zigbee message from 'Prise Masque Ingrid', type 'attributeReport', cluster 'genBasic', data '{"65506":30,"appVersion":66}' from endpoint 1 with groupID 0
debug 01/07/2023 14:49:46: Received Zigbee message from 'Prise Salle de Bain', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,3027]}' from endpoint 1 with groupID 0
info  01/07/2023 14:49:46: MQTT publish: topic 'zigbee2mqtt/Prise Salle de Bain', payload '{"current":0.01,"elapsed":8026,"energy":30.27,"indicator_mode":"off/on","last_seen":"2023-05-21T16:38:27+02:00","linkquality":76,"power":1,"power_outage_memory":"restore","state":"ON","voltage":238}'
debug 01/07/2023 14:50:05: Received Zigbee message from 'prise_entree', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 01/07/2023 14:50:12: Received Zigbee message from 'prise_entree', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":237}' from endpoint 1 with groupID 0
error 01/07/2023 14:50:12: Exception while calling fromZigbee converter: 'power_calibration' is not a number, got string ()}
debug 01/07/2023 14:50:12: Error: 'power_calibration' is not a number, got string ()
    at assertNumber (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:502:42)
    at calibrateAndPrecisionRoundOptions (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:90:5)
    at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/converters/fromZigbee.js:830:41)
    at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/opt/zigbee2mqtt/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
debug 01/07/2023 14:50:12: Received Zigbee message from 'prise_entree', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,39]}' from endpoint 1 with groupID 0
error 01/07/2023 14:50:12: Exception while calling fromZigbee converter: 'energy_calibration' is not a number, got string ()}
debug 01/07/2023 14:50:12: Error: 'energy_calibration' is not a number, got string ()
    at assertNumber (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:502:42)
    at calibrateAndPrecisionRoundOptions (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:90:5)
    at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/converters/fromZigbee.js:768:34)
    at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
[... 8 lines cut ...]
debug 01/07/2023 14:50:37: Received Zigbee message from 'prise_meuble_bureau', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,1]}' from endpoint 1 with groupID 0
info  01/07/2023 14:50:38: MQTT publish: topic 'zigbee2mqtt/prise_meuble_bureau', payload '{"child_lock":"UNLOCK","current":0,"elapsed":60108,"energy":0.01,"indicator_mode":"off/on","last_seen":"2023-05-21T16:39:03+02:00","linkquality":51,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":240}'
debug 01/07/2023 14:50:38: Received Zigbee message from 'prise_entree', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,39]}' from endpoint 1 with groupID 0
error 01/07/2023 14:50:38: Exception while calling fromZigbee converter: 'energy_calibration' is not a number, got string ()}
debug 01/07/2023 14:50:38: Error: 'energy_calibration' is not a number, got string ()
    at assertNumber (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:502:42)
    at calibrateAndPrecisionRoundOptions (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:90:5)
    at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/converters/fromZigbee.js:768:34)
    at Receive.onDeviceMessage (/opt/zigbee2mqtt/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/opt/zigbee2mqtt/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
debug 01/07/2023 14:50:38: Received Zigbee message from 'Prise Salle de Bain', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":1,"rmsCurrent":12,"rmsVoltage":238}' from endpoint 1 with groupID 0
[... cut ...]
debug 01/07/2023 14:51:00: Received Zigbee message from 'prise_meuble_bureau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1]}' from endpoint 1 with groupID 0
info  01/07/2023 14:51:00: MQTT publish: topic 'zigbee2mqtt/prise_meuble_bureau', payload '{"child_lock":"UNLOCK","current":0,"elapsed":84,"energy":0.01,"indicator_mode":"off/on","last_seen":"2023-05-21T16:39:03+02:00","linkquality":54,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":239}'

keithskillicorn, 2023-07-01:
I also have the same issue, after upgrading.  Every message from Develco Electricity meter is creating an error:

Error 2023-07-01 15:26:58Exception while calling fromZigbee converter: 'power_calibration' is not a number, got string ()}
Debug 2023-07-01 15:26:58Error: 'power_calibration' is not a number, got string () at assertNumber (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:502:42) at calibrateAndPrecisionRoundOptions (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:90:5) at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/converters/fromZigbee.js:752:33) at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/devices/develco.ts:83:40) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51) at EventEmitter.emit (node:events:525:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22) at Controller.<anonymous> (/app/lib/zigbee.ts:106:27) at Controller.emit (node:events:513:28) at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
Info 2023-07-01 15:26:58MQTT publish: topic 'zigbee2mqtt/Develco Electricity Meter', payload '{"battery_low":false,"check_meter":false,"current_summation":0,"energy":1202.191,"interface_mode":"electricity","linkquality":72,"power":309,"pulse_configuration":1000}'

zzz676, 2023-07-01:
In the same way, it constantly writes such an error.

Zigbee2MQTT:error 2023-07-01 19:15:08: Exception while calling fromZigbee converter: 'power_calibration' is not a number, got string ()}
Zigbee2MQTT:error 2023-07-01 19:15:38: Exception while calling fromZigbee converter: 'power_calibration' is not a number, got string ()}

Coldness00, 2023-07-01:
I had to downgrade, as I had same issue to control my IKEA ICPSHC24-10EU-IL-1.

zzz676, 2023-07-01:
> I had to downgrade, as I had same issue to control my IKEA ICPSHC24-10EU-IL-1.

How did you downgrade?

hellcry37, 2023-07-01:
same for me, the problem is i can not downgrade so easily since is not a docker container

alray31, 2023-07-01:
Same:
[image]

lior532, 2023-07-02:
i have the same issue i like to find a solution for this

Koenkk, 2023-07-02:
Make sure in your zigbee2mqtt configuration that e.g. `energy_calibration` is a number not string.

zzz676, 2023-07-02:
> 

> Make sure in your zigbee2mqtt configuration that e.g. `energy_calibration` is a number not string.

I have a repeated error about power_calibration.  The error appeared immediately after updating to 1.32, there were no errors on the previous version.  no new devices connected.

zzz676, 2023-07-02:
[image]

NecoPastor, 2023-07-02:
I got rid of the error by removing the power_calibration: '' line from the zigbee2mqtt config

JackTalisker, 2023-07-02:
I had the same error with a smartplug.

I choose a 0 for the field of _power_calibration_ in the device "Settings (specific)" page and the error seems to be gone

Nardol, 2023-07-02:
Thanks, I also had a `power_calibration: ''` line in my config, error is gone after removing it.
I also remove all other lines I found with value `''` to be sure, for other devices I did not have these and now all looks good.

I don't know why I did not have this error before but it is now fixed.

alray31, 2023-07-02:
> Make sure in your zigbee2mqtt configuration that e.g. `energy_calibration` is a number not string.

I just realized that if you go to the device setting (specific) page of z2m and touch a field when scrolling with a touch screen i.e power_calibtation, Z2m will add automatically a line on your configuration. yaml with no value:

yaml before
[image]

Scrolling and touching the power_calibtation field by mistake:
[image]

Result:
[image]

Is there a way to avoid that apart from not using a touchscreen on z2m?

Nardol, 2023-07-02:
Maybe in  fact this issue should be opened against ZB2MQTT frontend?

hellcry37, 2023-07-02:
I found the issue, had in my device config some temperature_calibration: '' so basically a null or empty string, after removing them all it was ok.

The problem is, I did npt put them there, it was the UI who did when i saved some confs for that sensor. Since i did not put anything there, UI decided to put temperature_calibration: '<empty>'

vadne, 2023-07-02:
I have the same trouble... While I'm waiting for the fix, I rolled back to the old version.

zekje, 2023-07-02:
Solved for my by remplacing

alray31, 2023-07-02:
> Maybe in fact this issue should be opened against ZB2MQTT frontend?

Done
https://github.com/nurikk/zigbee2mqtt-frontend/issues/1750

zzz676, 2023-07-02:
> I got rid of the error by removing the power_calibration: '' line from the zigbee2mqtt config

Thanks. Removed this line, now there are no more errors.

Gybrsh, 2023-07-03:
> Make sure in your zigbee2mqtt configuration that e.g. `energy_calibration` is a number not string.

This did the trick. I had a space in one of my smartplugs power_calibration, replaced it with a 0 (zero).

Coldness00, 2023-07-03:
> 

Hello,
 As I run this from Docker, I just reinstall the DOcker with the specific ver:  koenkk/zigbee2mqtt:1.31.2

Worked well

Nardol, 2023-07-03:
So I close this issue which is more related to the frontend, thanks @alray31 for opening nurikk/zigbee2mqtt-frontend#1750

Koenkk, 2023-07-04:
Pushed a fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Kepro, 2023-07-08:
hey @Koenkk I hate HA with z2m -> 1.32.1-1 which should have `Updated Zigbee2MQTT to version [1.32.1]` which should be included in 1.32.1 release but issue still persist

Koenkk, 2023-07-08:
Can you provide the debug log with 1.32.1?

Kepro, 2023-07-08:
here
```
Zigbee2MQTT:debug 2023-07-08 18:24:22: Received Zigbee message from 'Air Quality', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,242],"type":"Buffer"},"datatype":2,"dp":18}],"seq":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2023-07-08 18:24:22: Exception while calling fromZigbee converter: 'temperature_calibration' is not a number, got string ()}
Zigbee2MQTT:debug 2023-07-08 18:24:22: Error: 'temperature_calibration' is not a number, got string ()
    at toNumber (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:509:15)
    at Object.calibrateAndPrecisionRoundOptions (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:89:29)
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/lib/legacy.ts:3981:44)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
Zigbee2MQTT:info  2023-07-08 18:24:22: MQTT publish: topic 'zigbee2mqtt/Air Quality', payload '{"co2":362,"formaldehyd":2,"humidity":56.2,"last_seen":"2023-07-08T16:24:22.106Z","linkquality":153,"temperature":24,"voc":1}'
Zigbee2MQTT:debug 2023-07-08 18:24:22: Received Zigbee message from 'Air Quality', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,2,46],"type":"Buffer"},"datatype":2,"dp":19}],"seq":0}' from endpoint 1 with groupID 0
```

Koenkk, 2023-07-09:
`''` is indeed not a valid number so the error is correct.

kam821, 2023-07-11:
> `''` is indeed not a valid number so the error is correct.

All you need is click on the value field (of e.g. power_calibration) in Firefox to make the value with '' reappear in YAML configuration file.
This issue is still present in latest and latest-dev branches.

Kepro, 2023-07-11:
this is most likely an issue with the Frotnend part/package not this package... which has more issues with inputs and numbers...

Koenkk, 2023-07-11:
Please create an issue in the frontend repo: https://github.com/nurikk/zigbee2mqtt-frontend

Nardol, 2023-07-11:
Except if there is now something different, nurikk/zigbee2mqtt-frontend#1750 is still opened.

popy2k14, 2023-08-04:
Similar log entries here. I did not changed any device in my network and did not had such messages.
Here is an log entry:

`Debug 2023-08-04 15:38:24Error: 'illuminance_calibration' is not a number, got string () at toNumber (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:509:15) at calibrateAndPrecisionRoundOptions (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:89:29) at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/converters/fromZigbee.js:4052:35) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51) at EventEmitter.emit (node:events:525:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22) at Controller.<anonymous> (/app/lib/zigbee.ts:107:27) at Controller.emit (node:events:513:28) at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14) at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)`

Kepro, 2023-08-04:
@popy2k14 did you check your configuration.yaml?

popy2k14, 2023-08-04:
@Kepro thx for the hint. In my configuration.yaml was illuminance_calibration: '' for one device. Removed it, restarted Z2M and issue is gone. Dont know why this value was in there.

Anyway, it's solved now.
thx a lot

pkasarda, 2023-08-04:
@popy2k14 so same as I wrote above, it is an issue with frontend, adding empty string to config.. issue https://github.com/nurikk/zigbee2mqtt-frontend/issues/1750 

btw if you read a couple of messages about you will find out what to do :-P

popy2k14, 2023-08-05:
Thx for the hint.

Those who can read have a clear advantage 😉😂

closed 2023-09-13 by github-actions[bot]

xelemorf, 2023-10-31:
I had the same with `voltage_calibration: ''`
Did a find/replace for it and seems to be resolved.
```
Find: voltage_calibration: ''
Replace: voltage_calibration: '0'
```

AndKe, 2023-11-26:
updated docker gave me "1.33.2 commit: 9996c93"   - and I still see this issue?

AndKe, 2024-01-21:
still the same issue with 1.35.1 commit: 9eaaa0f
"Failed to apply calibration to 'pressure': 'pressure_calibration' is not a number, got string ()"

is this going to get fixed?

Kepro, 2024-01-22:
@AndKe you didn't even try to read the whole issue, right? ad 1 remove `pressure_calibration` from yaml manually, ad 2, it is or it was issue with frontend not with core package

AndKe, 2024-01-23:
@Kepro thanks, no the issue consists of lots of posts, and none contains a described workaround.
I understood that there is a file in one of the docker volumes, named configuration.yaml 
So far, I can only guess that I am looking for something like:
pressure_calibration : ''<cr>
which should be corrected to 
pressure_calibration :<cr>

- and I still hoped for the frontend or core, to handle such invalid setting from a previous build, simply correcting it on load .

hellcry37, 2024-01-23:
> @Kepro thanks, no the issue consists of lots of posts, and none contains a described workaround. I understood that there is a file in one of the docker volumes, named configuration.yaml So far, I can only guess that I am looking for something like: pressure_calibration : '' which should be corrected to pressure_calibration :
> 
>     * and I still hoped for the frontend or core, to handle such invalid setting from a previous build, simply correcting it on load .

no man just remove in configuration.yaml or devices.yaml or what you have it set to entire line.

for example i have this device:
```txt
'0x001788010b09c6a8':
  friendly_name: Master Bedroom Motion
  temperature_calibration: 2
  temperature_precision: 1
```

now if is set like this will error out:
```txt
'0x001788010b09c6a8':
  friendly_name: Master Bedroom Motion
  temperature_calibration: 2
  temperature_precision: ''
```
because temperature_precision is NULL

For this error to be gone remove the hole line that contains the null string in this case: temperature_precision: ''

Restart zigbee2mqtt

AndKe, 2024-01-25:
Thank you, fixed it after locating the file in docker.  - I hoped this was automatically handled in newer releases.