Koenkk/zigbee2mqtt #18170: MQTT breaks after last update.

Anashost, 2023-07-01:
### What happened?

So after i update to latest zigbee2mqtt version, everything works fine until i restart home assistant, then mqtt breaks (it starts but all devices have Unavailable state and most of them won't work like contact and pir sensors).

Rolling back didn't fix the problem, but if i Rollback then re-upgrade zigbee2mqtt again then it works until i restart home assistant.

### Zigbee2MQTT version

1.32.0-1

### Adapter firmware version

20221102

### Adapter

Sonoff - P

bobofruit1970, 2023-07-01:
How to get back to previous version?

oopiicaa, 2023-07-01:
Thanks. I have exactly the same problem:

> Well, update to 1.32.0 (1.32.0-1 to be exact as it is HA add-on) completely broke down my Zigbee network - Z2M does start normally and everything looks normal (log messages from devices), it's just that MQTT (EMQX) shows all entities as unavailable. I reverted back to 1.31.2, where entities are okay again, but I have a weird problem now with buttons not detecting clicks (multiple Aqara WXKG01LM and IKEA Tradfri E1743 for now) ... Something went completely wrong with this update for me.

Anashost, 2023-07-01:
I've created a Backup. Restore Zigbee2mqtt backup, then restart HA, then upgrade zigbee2mqtt again to latest versio, and everything should work. Don't restart afterwards because it will breaks again, until they fix the issue

anonim-no, 2023-07-01:
> How to get back to previous version?

Settings -> System -> Backups
Click on addon_xxxxxxxx_zigbee2mqtt_1.31.2-1, check and restore
Restart HA

TwanHogervorst, 2023-07-01:
I also just updated my Zigbee2Mqtt HA Addon to 1.32.0-1. It first started with newly added devices breaking and not appearing in Home Assistant. After rebooting (thinking it would fix the problem), all devices stopped working and became unavailable. The MQTT integration also appears to be crashing. Zigbee2Mqtt on it self seems to function properly, but my home assistant logs are filled with MQTT Error. I will share a snippit:

```
Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}], 'brightness': True, 'brightness_scale': 254, 'color_mode': True, 'command_topic': 'zigbee2mqtt/DEVICE_NAME/set', 'device': {'configuration_url': '/api/hassio_ingress/O6QLukZepbPsQF8nFHXA0ES12QcRJ72quZFrow8mIoE/#/device/0xa4c138e7789da260/info', 'identifiers': ['zigbee2mqtt_0xa4c138e7789da260'], 'manufacturer': 'TuYa', 'model': 'Zigbee E14 dimmable smart bulb RGB+CW+WW (TS0505B_1_2)', 'name': 'DEVICE_NAME', 'sw_version': 'z.1.0'}, 'effect': True, 'effect_list': ['blink', 'breathe', 'okay', 'channel_change', 'finish_effect', 'stop_effect'], 'max_mireds': 500, 'min_mireds': 153, 'name': 'DEVICE_NAME', 'schema': 'json', 'state_topic': 'zigbee2mqtt/DEVICE_NAME', 'supported_color_modes': ['xy', 'color_temp'], 'unique_id': '0xa4c138e7789da260_light_zigbee2mqtt', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 270, in async_discover config: DiscoveryInfoType = discovery_schema(discovery_payload)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/voluptuous/validators.py", line 232, in __call__ return self._exec((Schema(val) for val in self.validators), v) 
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/voluptuous/validators.py", line 355, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.11/site-packages/voluptuous/validators.py", line 351, in _exec v = func(v) 
^^^^^^^ File "/usr/local/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) 
^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable return schema(data) 
^^^^^^^^^^^^ File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 41, in validate_mqtt_light_discovery config: ConfigType = schemas[config_value[CONF_SCHEMA]](config_value) 
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/voluptuous/validators.py", line 232, in __call__ return self._exec((Schema(val) for val in self.validators), v) 
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/voluptuous/validators.py", line 355, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.11/site-packages/voluptuous/validators.py", line 351, in _exec v = func(v)
^^^^^^^ File "/usr/local/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data)
^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable return schema(data)
^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data)
^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: invalid url for dictionary value @ data['device']['configuration_url']
```

If some one knows a way to have more readable logs from HA...

bobofruit1970, 2023-07-01:
Is there way to choose version zigbee2mqtt to install? I am not asking how to use backup.

eljotpl, 2023-07-01:
Hello,

I have the same issue like @oopiicaa. In theory everything works, but all devices are listed as unavaliable. My config - sonoff ZBDongle-E with multi-pan firmware, SiliconLabs Zigbee/OpenThread Multiprotocol Add-on.

Anashost, 2023-07-01:
> Is there way to choose version zigbee2mqtt to install? I am not asking how to use backup.

If don't have a Backup then there's no easy way as far as I know.

johnsturgeon, 2023-07-01:
I am also having this issue, I was able to get most devices to recover by physically shutting them off / on but one device refuses to work (Aqara FP1)

Here is the log entry:
```
2023-07-01 05:42:51.534 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_binary_sensor_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Office FP1/availability'}], 'availability_mode': 'all', 'device': {'configuration_url': '/api/hassio_ingress/6kdcgrlQcMPDnlNVq_A368C6nDSrQe-Kfm08U2bdoT4/#/device/0x54ef44100058648a/info', 'identifiers': ['zigbee2mqtt_0x54ef44100058648a'], 'manufacturer': 'Xiaomi', 'model': 'Aqara presence detector FP1 (RTCZCGQ11LM)', 'name': 'Office FP1', 'sw_version': ''}, 'device_class': 'presence', 'json_attributes_topic': 'zigbee2mqtt/Office FP1', 'name': 'Office FP1 presence', 'payload_off': False, 'payload_on': True, 'state_topic': 'zigbee2mqtt/Office FP1', 'unique_id': '0x54ef44100058648a_presence_zigbee2mqtt', 'value_template': '{{ value_json.presence }}', 'platform': 'mqtt'},)
111

wilqq-the, 2023-07-01:
The issue is configuration_url which should be the absolute url but it's only getting the endpoint URL without a base from docker_entrypoint.sh

https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/498

GH2user, 2023-07-01:
I have the same thing and it might be related to the last URL being set in the SETTINGS - Frontend 
"URL on which the frontend can be reached, currently only used for the Home Assistant device configuration page"
I can not change the setting to blank which is the setting in 1.31.0

wilqq-the, 2023-07-01:
@GH2user exactly, you cannot change the URL because entrypoint is overwriting it when the container starts.

Koenkk, 2023-07-01:
Should be fixed in [v1.32.0-2](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/releases/tag/v1.32.0-2)

Anashost, 2023-07-01:
Thanks guys, everything works again👍🏻

magic7s, 2023-07-01:
> Should be fixed in [v1.32.0-2](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/releases/tag/v1.32.0-2)

Has someone verified?

oopiicaa, 2023-07-01:
Yes, working okay for me now.

KoKolaj, 2023-07-01:
> > Should be fixed in [v1.32.0-2](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/releases/tag/v1.32.0-2)
> 
> Has someone verified?

Yes, working okay for me now.

Koenkk, 2023-07-02:
Great, thanks all

closed 2023-07-02 by Koenkk

daaex1, 2023-07-02:
On zigbee2mqtt still not working.

danielo515, 2023-07-10:
I didn't updated zigbee2mqtt in a long time, I only updated HA recently, so how is this a zigbee2mqtt issue only?

simona70, 2023-12-03:
> > Is there way to choose version zigbee2mqtt to install? I am not asking how to use backup.
> 
> If you don't have a Backup then there's no easy way as far as I know.

Using zigbee2mqtt as service over endeavouros I could did "downgrade command" to previous zigbee2mqtt...