Koenkk/zigbee2mqtt #18171: 1.32.0 breaks Namron dimmer switches

mariusjell, 2023-07-01:
### What happened?

After the update the dimmer reports as unsupported and all entities in Home assistant related to the switches dissappear.

### Zigbee2MQTT version

1.32.0

### Adapter firmware version

20221226

### Adapter

zzh!

### Debug log

[image]

mariusjell, 2023-07-01:
This is how the dimmers look in 1.31.2 where they work as expected.
[image]

kiekerjan18, 2023-07-01:
Looks like an EcoDim dimmer. Which is funny. My EcoDim dimmer is broken since latest update. 

[image]

And doesn't exposes anything:
[image]

My other EcoDim seems appear to be working fine. 

Any revert possibilities?

mariusjell, 2023-07-01:
> Looks like an EcoDim dimmer. Which is funny. My EcoDim dimmer is broken since latest update.
> 
> [image]
> 
> And doesn't exposes anything:
> 
> My other EcoDim seems appear to be working fine.
> 
> Any revert possibilities?

I reverted by restoring from backup, I run my HA environment virtual so it's quite quick to do backups and restore of the whole machine. My dimmers are Namron branded, not Ecodim but maybe a rebrand or something. https://www.namron.com/products/namron-zigbee-led-dimmer-250w/

I just tried updating to 132.2 but the issue is still there.

Koenkk, 2023-07-02:
Can you provide the data/database.db entry of your device?

kiekerjan18, 2023-07-02:
NOT WORKING:
`{"id":11,"type":"Router","ieeeAddr":"0xccccccfffe398019","nwkAddr":53989,"manufId":4714,"manufName":"EcoDim B.V","powerSource":"Mains (single phase)","modelId":"Dimmer-Switch-ZB3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Dimmer-Switch-ZB3.0","manufacturerName":"EcoDim B.V","powerSource":1,"zclVersion":3,"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20191105","swBuildId":"3.04"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20191105","swBuildId":"3.04","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1688278129376,"defaultSendRequestWhen":"immediate"}`

WORKING:
`{"id":12,"type":"Router","ieeeAddr":"0x000d6ffffebcdcb8","nwkAddr":24246,"manufId":4714,"manufName":"EcoDim BV","powerSource":"Mains (single phase)","modelId":"Dimmer-Switch-ZB3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Dimmer-Switch-ZB3.0","manufacturerName":"EcoDim BV","powerSource":1,"zclVersion":3,"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20200701","swBuildId":"3.09"}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c04c3e","endpointID":242},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c04c3e","endpointID":242},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c04c3e","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c04c3e","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":1,"maxRepIntval":3600,"repChange":1}],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20200701","swBuildId":"3.09","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1660633525},"lastSeen":1688277874693,"defaultSendRequestWhen":"immediate"}`

[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/11928857/database.zip)

Koenkk, 2023-07-02:
Pushed the fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

You can use the following external converter until the next release is available:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {type: 'Router', manufacturerName: 'EcoDim BV', modelID: 'Dimmer-Switch-ZB3.0'},
        {type: 'Router', manufacturerName: 'EcoDim B.V', modelID: 'Dimmer-Switch-ZB3.0'},
        {type: 'Router', manufacturerName: 'EcoDim BV', modelID: 'EcoDim-Zigbee 3.0', endpoints: [
            {ID: 1, profileID: 260, deviceID: 257, inputClusters: [0, 3, 4, 5, 6, 8, 2821, 4096], outputClusters: [25]},
            {ID: 242, profileID: 41440, deviceID: 97, inputClusters: [], outputClusters: [33]},
        ]},
        {type: 'Router', manufacturerName: 'EcoDim BV', modelID: 'EcoDim-Zigbee 3.0', endpoints: [
            {ID: 1, profileID: 260, deviceID: 257, inputClusters: [0, 3, 4, 5, 6, 8, 2821, 4096], outputClusters: [25]},
            {ID: 67, inputClusters: [], outputClusters: []},
            {ID: 242, profileID: 41440, deviceID: 97, inputClusters: [], outputClusters: [33]},
        ]},
    ],
    model: 'Eco-Dim.07/Eco-Dim.10',
    vendor: 'EcoDim',
    description: 'Zigbee & Z-wave dimmer',
    ota: ota.zigbeeOTA,
    extend: extend.light_onoff_brightness({noConfigure: true, disableEffect: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

closed 2023-07-02 by Koenkk

kiekerjan18, 2023-07-02:
Yes.. my device is working again (with external converter as temp solution).

briis, 2023-07-03:
I have a few clones of this device made by LightSolutions, so I needed to add the below line to the `ext_converter.js` to make it work again.

`        {type: 'Router', manufacturerName: 'Lightsolutions', modelID: 'Dimmer-Switch-ZB3.0'},
`

robertbakke, 2023-07-03:
I tried to use the latest-dev and it did not fix the problem, also the above ext_converter.js with latest did not fix the problem with my Namron AS dimmers. I had to add this line to ext_converter.js to get it working:

{type: 'Router', manufacturerName: 'Namron AS', modelID: 'Dimmer-Switch-ZB3.0'},

Koenkk, 2023-07-04:
Integrated the fixes, thanks!

Credo102, 2023-07-06:
> ```yaml
> external_converters:
>   - ext_converter.js
> ```

Can "Letsleds" aka "Lets leds" be added to this?

I have fixed this for now by adding the following to your (very helpful, thank you) ext_converter.js 
```
 {type: 'Router', manufacturerName: 'Letsleds', modelID: 'Dimmer-Switch-ZB3.0'},
 {type: 'Router', manufacturerName: 'Lets leds', modelID: 'Dimmer-Switch-ZB3.0'},
```

Here are the data/database.db entry of these devices. Thanks again.

`{"id":5,"type":"Router","ieeeAddr":"0x086bd7fffe5d30a3","nwkAddr":23796,"manufId":4714,"manufName":"Letsleds","powerSource":"Mains (single phase)","modelId":"Dimmer-Switch-ZB3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Dimmer-Switch-ZB3.0","manufacturerName":"Letsleds","powerSource":1,"zclVersion":3,"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20200701","swBuildId":"3.09"}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe133fbe","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe133fbe","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":1,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20200701","swBuildId":"3.09","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-526939472},"lastSeen":1688644248848,"defaultSendRequestWhen":"immediate"}`

`{"id":7,"type":"Router","ieeeAddr":"0x086bd7fffe5d3c75","nwkAddr":57063,"manufId":4714,"manufName":"Lets leds","powerSource":"Mains (single phase)","modelId":"Dimmer-Switch-ZB3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Dimmer-Switch-ZB3.0","manufacturerName":"Lets leds","powerSource":1,"zclVersion":3,"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20200701","swBuildId":"3.09"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe133fbe","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe133fbe","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":1,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20200701","swBuildId":"3.09","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-526939472},"lastSeen":1688644318606,"defaultSendRequestWhen":"immediate"}
`

anders3408, 2024-01-14:
I also have a Lightning Solutions dimmer and that's need this line to work:

{type: 'Router', manufacturerName: 'Light Solutions', modelID: 'Dimmer-Switch-ZB3.0'},

So that model exists with different names.