Koenkk/zigbee2mqtt #18173: Z2M doesnt work anymore with HA

xHywel, 2023-07-01:
### What happened?

Hi, 
It seems that since v1.32.0-2, Z2M image can't be found by my home Assistant (I can't restart it and all devices are unavailable)

Image zigbee2mqtt/zigbee2mqtt-amd64:1.32.0-2 does not exist for addon_45df7312_zigbee2mqtt

Is it "normal" ? Something to be fixed by HA ? 

Thanks in advance !

### Zigbee2MQTT version

1.32.0-2

### Adapter firmware version

?

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

chrisbeardy, 2023-07-01:
I have the same problem

Simmarith, 2023-07-02:
Here is a workaround:

- ssh into the linux that is running your homeassistant
- run docker pull `zigbee2mqtt/zigbee2mqtt-amd64:1.32.0-2`
- try to start the zigbee2mqtt addon again

it seems to not automatically pull the latest image

xHywel, 2023-07-02:
Thanks for the reply
I fixed the issue by running this command in Terminal (of Home Assistant) : ha supervisor repair

syberx, 2023-07-06:
same problem i can use 1.32.0-2 oder last dev .. but today last HA Update same ... i use zigbee2mqtt proxy addon ...

closed 2023-08-13 by github-actions[bot]