Koenkk/zigbee2mqtt #18176: Osram lamp stopped working

Th3M1k3y, 2023-07-02:
### What happened?

After updating to 1.32.0-2 one of my lamps stopped working, now I get this error when I try to change anything on it

`Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}`

and the lamp does not change anything.

The lamp is this one https://www.zigbee2mqtt.io/devices/4058075816794.html#osram-4058075816794

### How to reproduce it (minimal and precise)

Any change to the exposed options makes it return the error

### Zigbee2MQTT version

1.32.0-2

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

Zigbee2MQTT:error 2023-07-02 06:35:00: Publish 'set' 'state' to 'workroom_ceiling_lamp' failed: 'Error: 'transition' is not a number, got string (2)'
Zigbee2MQTT:error 2023-07-02 06:35:23: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:35:26: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:35:27: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:35:30: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:35:31: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:35:31: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:36:23: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:36:26: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:36:27: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:36:30: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:36:31: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:36:31: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:37:23: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:37:26: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:37:27: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:37:30: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:37:31: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:37:31: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:38:23: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:38:26: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:38:27: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:38:30: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:error 2023-07-02 06:38:31: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}

Koenkk, 2023-07-02:
Please provide the debug log

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Th3M1k3y, 2023-07-02:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11928898/log.txt)

Koenkk, 2023-07-02:
The `transition` in your `configuration.yaml` seems to be a string and should be a number (change `'2'` to `2`)

Th3M1k3y, 2023-07-02:
Which configuration.yaml should that be in?

Koenkk, 2023-07-02:
Your zigbee2mqtt `configuration.yaml`

Th3M1k3y, 2023-07-02:
I don't follow... There is no transition in there, should it be added in?

```
external_converters: []
devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant:
  legacy_entity_attributes: false
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto:1883
  user: addons
  password: x
  keepalive: 60
  reject_unauthorized: true
  version: 4
serial:
  port: /dev/ttyUSB0
advanced:
  log_level: debug
  pan_id: 6704
  channel: 15
  network_key:
    - x
    - x
    - x
    - x
    - x
    - x
    - x
    - x
    - x
    - x
    - x
    - x
    - x
    - x
    - x
    - x
  availability_blocklist: []
  availability_passlist: []
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
availability: true
```

Koenkk, 2023-07-02:
Check your `groups.yaml` and `devices.yaml` files

etvanp, 2023-07-02:
Sorry to jump in, but have same issue with Livarno Lux E27 bulb CCT, TS0502A also after updating zigbee2mqtt to 1.32.0-2 on Homeassistant. Errors are 
```
08:31:06: Publish 'set' 'brightness' to 'Bedroom-BedLight' failed: 'Error: 'transition' is not a number, got string ()'
Zigbee2MQTT:error 2023-07-02 08:31:10: Publish 'set' 'brightness' to 'Bedroom-BedLight' failed: 'Error: 'transition' is not a number, got string ()'
Zigbee2MQTT:error 2023-07-02 08:31:14: Publish 'set' 'brightness' to 'Bedroom-BedLight' failed: 'Error: 'transition' is not a number, got string ()'
``` 
Please note that I haven't changed anything for months regarding the configuration of the device, so some breaking change was in this minor release.

Th3M1k3y, 2023-07-02:
Found it, it was inside devices.yaml, but I wonder what caused it to stop working. Yesterday I turned the light on by pressing another zigbee button, and after updating zigbee2mqtt it could no longer be controlled.

etvanp, 2023-07-02:
Something changed, before state and brightness could be a string but not anymore.

Th3M1k3y, 2023-07-02:
So the lamp started working again, but there is another problem, which is also in the log.txt from before.

I am getting a lot of `Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}` which seems to be coming from this one https://www.zigbee2mqtt.io/devices/TS0601_din_1.html#tuya-ts0601_din_1 which is called `network_cabinet`

Chaoscontrol, 2023-07-03:
Similar problem with a Gledopto GL-C-008P. Worked fine for 6 months since set up. Stopped responding a couple days ago (I don't recall the update but I guess I did). Issue seems to be just with this device. 

Errors here for different commands:
`Error 2023-07-03 10:12:14Publish 'set' 'state' to 'Bed LED' failed: 'Error: 'transition' is not a number, got string ()'
Info 2023-07-03 10:12:14MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Bed LED' failed: 'Error: 'transition' is not a number, got string ()'","meta":{"friendly_name":"Bed LED"},"type":"zigbee_publish_error"}'
Error 2023-07-03 10:12:18Publish 'set' 'brightness' to 'Bed LED' failed: 'Error: 'transition' is not a number, got string ()'
Info 2023-07-03 10:12:18MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'Bed LED' failed: 'Error: 'transition' is not a number, got string ()'","meta":{"friendly_name":"Bed LED"},"type":"zigbee_publish_error"}'
Error 2023-07-03 10:12:21Publish 'set' 'color_temp' to 'Bed LED' failed: 'Error: 'transition' is not a number, got string ()'
Info 2023-07-03 10:12:21MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'color_temp' to 'Bed LED' failed: 'Error: 'transition' is not a number, got string ()'","meta":{"friendly_name":"Bed LED"},"type":"zigbee_publish_error"}'
Error 2023-07-03 10:12:21Publish 'set' 'brightness' to 'Bed LED' failed: 'Error: 'transition' is not a number, got string ()'
Info 2023-07-03 10:12:21MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'Bed LED' failed: 'Error: 'transition' is not a number, got string ()'","meta":{"friendly_name":"Bed LED"},"type":"zigbee_publish_error"}'`

Here my Z2M config.yaml: 
`homeassistant: true
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: redacted
serial:
  port: /dev/ttyUSB0
frontend:
  port: 8099
devices:
  '0x00124b0024c95cf8':
    friendly_name: Sonoff plug
  '0x00158d0008f12938':
    friendly_name: Front door
  '0x00158d0008f129b0':
    friendly_name: Back door
  '0xec1bbdfffe940e04':
    friendly_name: Living room lights
  '0x60a423fffecd6f10':
    friendly_name: Bed LED
    transition: ''
  '0xa4c1381977fabda3':
    friendly_name: Kitchen lights
  '0xa4c1384908ab4c72':
    friendly_name: Guest bedroom lights
  '0x00158d0006f44f5c':
    friendly_name: Utility room motion sensor
    occupancy_timeout: 180
  '0xa4c1387577fc1208':
    friendly_name: Utility room lights
  '0xa4c1386e67aaf99d':
    friendly_name: Entrance lights
  '0x00158d000896cb33':
    friendly_name: Vibration sensor
    vibration_timeout: 5
  '0x00158d0008f0f9aa':
    friendly_name: Stairs motion sensor
    occupancy_timeout: 60
  '0x00158d0009410a8d':
    friendly_name: Kitchen cooker weather
    disabled: false
    homeassistant: {}
    optimistic: true
  '0x00158d00091a46ba':
    friendly_name: Guest room weather
  '0xe0798dfffe9bedcc':
    friendly_name: Living room knob
  '0x70ac08fffe20d2f7':
    friendly_name: Bed 4-switch
  '0x70ac08fffe735ba4':
    friendly_name: Guest room remote
  '0xa4c13869f69a17a6':
    friendly_name: Studio lights
  '0xa4c1389449660cb9':
    friendly_name: Landing switch
permit_join: true
advanced:
  log_level: info
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 514
    protocol: udp4
    type: '5424'
  log_rotation: true
`

etvanp, 2023-07-03:
@Chaoscontrol remove the transition: line on the yaml. That's the issue .

Chaoscontrol, 2023-07-03:
@etvanp Thanks! 
I did once, restarted HA but it repapered, but I did again and now it seems it worked. Why was that there though? I didn't add that myself. I didn't even know of the existence of this file till today. It must still be an issue from this new version.

etvanp, 2023-07-03:
That line on the yaml reflects this Web UI config regarding the particular device. At some point I remember I tried some transition value and then removed it (hence the '' on the yaml). Until now this was fine, but with the latter release of Z2M something changed and default must be now 0 and not ''. This is my best guess

[image]

Chaoscontrol, 2023-07-03:
Fair enough. Might have played with it myself back when I set it up.

closed 2023-07-07 by Th3M1k3y