Koenkk/zigbee2mqtt #18177: Zigbee2mqtt not starting Conbee II

Imaginous, 2023-07-02:
### What happened?

Zigbee2mqtt doesn't start, gives a herdsman error 

### What did you expect to happen?

Zigbee2mqtt to start 

### How to reproduce it (minimal and precise)

Just start it with a Conbee II stick

### Zigbee2MQTT version

1.32.0

### Adapter firmware version

0x264a0700

### Adapter

Conbee II 

### Debug log

`Jul 02 06:36:44 domoticapi1 systemd[1]: Started zigbee2mqtt.
Jul 02 06:36:46 domoticapi1 npm[3084]: > zigbee2mqtt@1.32.0 start
Jul 02 06:36:46 domoticapi1 npm[3084]: > node index.js
Jul 02 06:36:55 domoticapi1 npm[3084]: /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js:79
Jul 02 06:36:55 domoticapi1 npm[3084]:         if (converter.options) {
Jul 02 06:36:55 domoticapi1 npm[3084]:                       ^
Jul 02 06:36:55 domoticapi1 npm[3084]: TypeError: Cannot read properties of undefined (reading 'options')
Jul 02 06:36:55 domoticapi1 npm[3084]:     at Object.addDefinition [as addDeviceDefinition] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js:79:23)
Jul 02 06:36:55 domoticapi1 npm[3084]:     at new ExternalConverters (/opt/zigbee2mqtt/lib/extension/externalConverters.ts:15:17)
Jul 02 06:36:55 domoticapi1 npm[3084]:     at new Controller (/opt/zigbee2mqtt/lib/controller.ts:84:58)
Jul 02 06:36:55 domoticapi1 npm[3084]:     at start (/opt/zigbee2mqtt/index.js:106:18)
Jul 02 06:36:55 domoticapi1 systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jul 02 06:36:55 domoticapi1 systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Jul 02 06:36:55 domoticapi1 systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jul 02 06:36:55 domoticapi1 systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Jul 02 06:36:55 domoticapi1 systemd[1]: Stopped zigbee2mqtt.`

Imaginous, 2023-07-02:
It's not the Conbee II stick, but an external herdsman converter that stopped working.

It's missing the "options" parameter if I understand correctly.

Is there a way to fix this external converter?

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const valueConverter = {
    phaseA: {
        from: (v) => {
            return {current: ((v[4] | v[3] << 8) / 1000), voltage: ((v[1] | v[0] << 8) / 10), power: ((v[7] | v[6] << 8))}
        },
    },
    threshold: {
        from: (v) => {
            const lookup = {0: 'not_set', 1: 'over_current', 3: 'over_voltage'};
            
            return {
                threshold_1: lookup[v[0]],
                threshold_1_value : (v[3] | v[2] << 8),
                threshold_2: lookup[v[4]],
                threshold_2_value : (v[7] | v[6] << 8),                            
            };  
        },
    },
    alarm: tuya.valueConverterBasic.lookup({'clear': 0, 'over_current': 1, 'over_power': 2, 'over_voltage': 4, 'wrong_frequency': 8}),
};

const tuyaExposes = {
    threshold: (l) => exposes.enum(`threshold_${l}`, ea.STATE, ['not_set', 'over_current', 'over_voltage'])
        .withDescription(`Treshold ${l} state`),
    tresholdValue: (l) => exposes.numeric(`threshold_${l}_value`, ea.STATE)
        .withDescription(`Treshold ${l} value`),
    alarm: () => exposes.enum(`alarm`, ea.STATE, ['clear', 'over_current', 'over_voltage', 'over_power', 'wrong_frequency'])
        .withDescription(`Alarm state`),  
};

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_lsanae15']),
    model: 'TS0601_din_2',
    vendor: 'TuYa',
    description: 'Zigbee DIN energy meter',
    fromZigbee: [tuya.fzDataPoints],
    toZigbee: [tuya.tzDataPoints],
    configure: tuya.configureMagicPacket,
    exposes: [ e.energy(), e.power(), e.voltage(), e.current(), 
        tuyaExposes.threshold('1'), tuyaExposes.threshold('2'), tuyaExposes.tresholdValue('1'), tuyaExposes.tresholdValue('2'),
        tuyaExposes.alarm()],
    meta: {
        tuyaDatapoints: [
            [1, 'energy', tuya.valueConverter.divideBy100],
            [3, null, null], //??? - [22,11,22,11,0,0,0,0]
            [4, null, null], //??? - [11,4,11,4,0,0,0,0]
            [6, null, valueConverter.phaseA], // voltage, power and current
            [10, 'alarm', valueConverter.alarm],
            [11, null, null], //??? - [1,0]
            [16, 'state', tuya.valueConverter.onOff],
            [17, null, valueConverter.threshold],
            [18, null, null], //Serial number
        ],
    },
};

module.exports = definition;`

Koenkk, 2023-07-02:
`_TZE200_lsanae15` is supported out-of-the-box, you can remove your ext converter.

Imaginous, 2023-07-02:
This, ofcourse, works... but I would like to understand why it doensn't work anymore.

I use this custom implementation to block the switch function of this device. Since it's installed in the mains powerbox you could kill the entire house of power when turning it off.

I disabled that functionality in the custom converter.

Koenkk, 2023-07-04:
External converters are meant for temporary testing, they can break if the used zigbee-herdsman-converters code changes (which it did a lot last release)

Imaginous, 2023-08-04:
It seems due to a Herdsman change.
Still would like to know how to fix it. But it is not a bug.

closed 2023-08-04 by Imaginous