Koenkk/zigbee2mqtt #18178: EcoDim Dimmer-Switch-ZB3.0 does not work anymore  after update 1.32.0-2

Roy75, 2023-07-02:
### What happened?

EcoDim Dimmer-Switch-ZB3.0 does not work anymore  after update EcoDim Dimmer-Switch-ZB3.0 does not work anymore  after update 1.32.0-2

### What did you expect to happen?

i hope that it will work again.

### Zigbee2MQTT version

EcoDim Dimmer-Switch-ZB3.0 does not work anymore  after update 1.32.0-2

### Adapter firmware version

20210120

### Adapter

zStack3x0

KDanielK, 2023-07-02:
Same with this device that earlier was identified as EcoDim Dimmer-Switch-ZB3.0 

This device is a external cord dimmer
Zigbee-modell
Dimmer-Switch-ZB3.0
Zigbee-manufacturer
MoooN AB

Koenkk, 2023-07-02:
Please provide the data/database.db entry of your device.

Roy75, 2023-07-02:
you mean this?

{"id":25,"type":"Router","ieeeAddr":"0x086bd7fffe5de612","nwkAddr":15191,"manufId":4714,"manufName":"EcoDim BV","powerSource":"Mains (single phase)","modelId":"Dimmer-Switch-ZB3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Dimmer-Switch-ZB3.0","manufacturerName":"EcoDim BV","powerSource":1,"zclVersion":3,"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20200701","swBuildId":"3.09"}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":191}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b95378","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b95378","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":1,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20200701","swBuildId":"3.09","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1660633525},"lastSeen":1688279504317,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-07-02:
> EcoDim Dimmer-Switch-ZB3.0 does not work anymore after update 1.32.0-2

Can you explain what doesn't work anymore?

KDanielK, 2023-07-02:
Reported as 'not supported', nothing works

Roy75, 2023-07-02:
its online, but can not control it. It is "not supported"

Koenkk, 2023-07-02:
Does the ext converter from https://github.com/Koenkk/zigbee2mqtt/issues/18171#issuecomment-1616416833 fix it?

Roy75, 2023-07-02:
I'm not at home right now, so i will try this later

KDanielK, 2023-07-02:
I still get Zigbee2MQTT:warn  2023-07-02 10:11:00: Received message from unsupported device with Zigbee model 'Dimmer-Switch-ZB3.0' and manufacturer name 'MoooN AB'

Please note that I have a different manufacturer

{"id":8,"type":"Router","ieeeAddr":"0x8cf681fffe9d5845","nwkAddr":43236,"manufId":4714,"manufName":"MoooN AB","powerSource":"Mains (single phase)","modelId":"Dimmer-Switch-ZB3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":0}},"genBasic":{"attributes":{"zclVersion":3,"manufacturerName":"MoooN AB","hwVersion":1,"modelId":"Dimmer-Switch-ZB3.0","appVersion":1,"stackVersion":6,"powerSource":1,"dateCode":"20210802"}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"genOta":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b78e83","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b78e83","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":1,"dateCode":"20210802","swBuildId":"3.09","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-526939472},"lastSeen":1688285106856,"defaultSendRequestWhen":"immediate"}

KDanielK, 2023-07-02:
Same manufId but different manufName that is

KDanielK, 2023-07-02:
Changed to 
        {type: 'Router', manufacturerName: 'EcoDim BV', modelID: 'Dimmer-Switch-ZB3.0'},
        {type: 'Router', manufacturerName: 'EcoDim B.V', modelID: 'Dimmer-Switch-ZB3.0'},
        {type: 'Router', manufacturerName: 'MoooN AB', modelID: 'Dimmer-Switch-ZB3.0'},
        {type: 'Router', manufacturerName: 'EcoDim BV', modelID: 'EcoDim-Zigbee 3.0', endpoints: [
And now it works as expected even though I get the wrong picture of the device

KDanielK, 2023-07-02:
I guess that there isnt a 1:1 between manufId and manufName

KDanielK, 2023-07-02:
This is my device btw, I guess the hw is tha same but the formfactor is different 
https://www.kjell.com/se/produkter/smarta-hem/fjarrkontroller/smarta-dimmers/hzc-sladdimmer-zigbee-vit-p51917

markxroberts, 2023-07-02:
I have a similar problem.  Worked up to the last version.  Database entry is:
```
{"id":11,"type":"Router","ieeeAddr":"0x0c4314fffefc2102","nwkAddr":41694,"manufId":4098,"manufName":"Smart Dim","powerSource":"Mains (single phase)","modelId":"Dimmer-Switch-ZB3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Dimmer-Switch-ZB3.0","manufacturerName":"Smart Dim","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":6,"hwVersion":1,"dateCode":"20201223","swBuildId":"1.2"}},"genLevelCtrl":{"attributes":{"currentLevel":127}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dedb45","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dedb45","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":1,"dateCode":"20201223","swBuildId":"1.2","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-526939472},"lastSeen":1688290388364,"defaultSendRequestWhen":"immediate"}
```
My device is: 
https://www.amazon.co.uk/gp/product/B09627JQ31

robertbakke, 2023-07-02:
I have the Namron AS version of these dimmers, and they all stopped working after upgrading to 1.32. They could not be controlled and Zigbee2Mqtt said they were unsupported. Rolling back to 1.31.2 fixes the problem. 

Zigbee Model
: Dimmer-Switch-ZB3.0
Zigbee Manufacturer
: Namron AS
Description
: Zigbee & Z-wave dimmer

Firmware build date
: 20191105
Firmware version
: 3.04
Manufacturer
: [EcoDim](https://www.zigbee2mqtt.io/supported-devices/#v=EcoDim)
Model
: [Eco-Dim.07/Eco-Dim.10](https://www.zigbee2mqtt.io/devices/Eco-Dim.07_Eco-Dim.10.html#ecodim-eco-dim.07%252Feco-dim.10)

Caine72, 2023-07-02:
I have the same "visual" model as @KDanielK, but the maker shyugj.

Added that maker as well to @Koenkk supplied fix above in the fingerprint section. . And now it's visible again. I can turn it on/off and change brightness level.

Though, I did not have time to check more things before leaving for work.

Dissimilis, 2023-07-02:
Same with Thorgeon dimmer ([https://thorgeon.com/10003](https://thorgeon.com/10003) ) after upgrade to 1.32.0 - device is not supported anymore.

turrican944, 2023-07-02:
Idinio Zigbee dont supported in 1.32.0.
Geräte-Typ 
:Router
Zigbee Modell
:Dimmer-Switch-ZB3.0
Zigbee Hersteller
:ldinio
Beschreibung
:Zigbee & Z-wave dimmer
Unterstützungsstatus
Unterstützt

IEEE Addresse
0x086bd7fffe5d6569
Netzwerk Adresse
0x517F
Firmware-Datum
20200701
Firmware-Version
3.09

I added {type: 'Router', manufacturerName: 'ldinio', modelID: 'Dimmer-Switch-ZB3.0'},  in ext_converter.js and it works with 1.32.0.

robertbakke, 2023-07-03:
I had to add this line to the ext_converter.js to get it working with my Namron AS dimmers:

{type: 'Router', manufacturerName: 'Namron AS', modelID: 'Dimmer-Switch-ZB3.0'},

Caine72, 2023-07-03:
@Koenkk   I saw that your closed this issue: https://github.com/Koenkk/zigbee2mqtt/issues/18171#issuecomment-1616416833

It seems like we are a lot of users with the same modelID but different manufacturerName. 
What is the recommended way moving forward with this issue? 
Open up a pull request for each device ( where the manufacturerName is different), or something else?

KDanielK, 2023-07-03:
@Caine72 as I understand the zigbee standard the manufacturer id is unique/registered but the name is a free text field.
_"The manufacturer code field of the node descriptor is sixteen bits in length and specifies a manufacturer code that is allocated by the ZigBee Alliance, relating the manufacturer to the device.
The manufacturer name field has a variable length and contains a character string representing the name of the manufacturer of the device."_

We have manufacturer id 4714 with several white label manufacturer names associated. I guess many who white label also will register a manufacturer id but not sure what the process looks like.

Caine72, 2023-07-03:
For sure @KDanielK . 

But what do we do then? 
Seems like we need to add the manufacturerName to the fingerprint for it to be recognized correctly in z2m. 

This way we'll have an ever growing fingerprint array added to the EcoDim.
Or we could add each in lack of better word,manufacturer name,  like a unique device. 
Or there is some other way of doing it that I do not know about ( which is likely as my knowledge of adding new devices to z2m is rudimentary at best.)

Either way, I assume there is some guidelines regarding best practice regarding similar cases? I assume this is not the first instance of this happening. Or maybe it is. I do not know. =)

So what I am searching for is,  what is the next sustainable step in making sure that this is included in the next release. And is not removed again.

What is officially added right now, https://github.com/Koenkk/zigbee2mqtt/issues/18171#issuecomment-1616416833,  will not work for all of us in the thread, as we all have unique fingerprints stemming from the different manufacturerName(s).

Do you have a proposal of what to do? :) E.g Can one lean on the ID instead?

palsbo, 2023-07-04:
My device was in v 1.31.2 registered as:
Producer=EcoDim, Model=Eco-Dim.07/Eco-Dim.10
in v 1.32.0 it says:
Producer=not supported, Model=Dimmer-Switch-ZB3.0
I had to go back to 1.31.2 until solved.

MDSDM, 2023-07-04:
I'm having similar issues. However, all my EcoDim 10 devices are reported as EcoDim 7. And all EcoDim 7 devices are reported as unsupported.

KDanielK, 2023-07-04:
@Caine72 I changed it to match with id instead of name and it works for me at least

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {type: 'Router', manufacturerID: 4714, modelID: 'Dimmer-Switch-ZB3.0'},
        {type: 'Router', manufacturerName: 'EcoDim BV', modelID: 'EcoDim-Zigbee 3.0', endpoints: [
            {ID: 1, profileID: 260, deviceID: 257, inputClusters: [0, 3, 4, 5, 6, 8, 2821, 4096], outputClusters: [25]},
            {ID: 242, profileID: 41440, deviceID: 97, inputClusters: [], outputClusters: [33]},
        ]},
        {type: 'Router', manufacturerName: 'EcoDim BV', modelID: 'EcoDim-Zigbee 3.0', endpoints: [
            {ID: 1, profileID: 260, deviceID: 257, inputClusters: [0, 3, 4, 5, 6, 8, 2821, 4096], outputClusters: [25]},
            {ID: 67, inputClusters: [], outputClusters: []},
            {ID: 242, profileID: 41440, deviceID: 97, inputClusters: [], outputClusters: [33]},
        ]},
    ],
    model: 'Eco-Dim.07/Eco-Dim.10',
    vendor: 'EcoDim',
    description: 'Zigbee & Z-wave dimmer',
    ota: ota.zigbeeOTA,
    extend: extend.light_onoff_brightness({noConfigure: true, disableEffect: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint);
    },
};

module.exports = definition;

Caine72, 2023-07-04:
@KDanielK Did the same, seems to work.

I guess we should create a pull request then if the maintainer does not want to change it himself like he did in the other tread?

Koenkk, 2023-07-04:
This should be fixed in the dev branch now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

KDanielK, 2023-07-04:
I created a pull request to support MoooN AB and other white label versions
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5946

Arkoxs, 2023-07-05:
@Koenkk Can you explain what happend? I had the same issue with my older EcoDim dimmer. Normally I don't just update a working config but this time the 1.32.0-2  "Fix issue with configuration URL" seemed to make sense. I missed the earlier update to 1.32.0-1 where Zigbee2Mqtt updated to 1.32.0 and EcoDim went in the pooper.

Koenkk, 2023-07-05:
@Arkoxs a mistake was made during refactoring. I will make a hotfix release later this week.

This issue should be fixed in the dev branch now, @ all please confirm.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

robertbakke, 2023-07-06:
> @Arkoxs a mistake was made during refactoring. I will make a hotfix release later this week.
>
> This issue should be fixed in the dev branch now, @ all please confirm.
>
> Changes will be available in [...]

Im running Zigbee2MQTT in docker on Unraid. 
I just stopped the container, removed my ext_converter.js reference in configuration.yaml, changed repository to zigbee2mqtt:latest-dev and started the container. My Namron AS dimmers are still broken with the latest dev. I need to use the temp solution with ext_converter.js or roll back to 1.31.2 to get them working (supported) again.

[image]

Thanks for all the effort everyone!

KDanielK, 2023-07-06:
@robertbakke id you look in database.db what is the manufId for Namron? EcoDim whitelabel units should be "manufId":4714
I would guess that namron should be in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/namron.ts instead of EcoDim

robertbakke, 2023-07-06:
@KDanielK Yes, looks like my Namron ones are also 4714. 
This is one of the dimmers working with 1.32 using ext_converter.js:

{"id":29,"type":"Router","ieeeAddr":"0x086bd7fffe5cd208","nwkAddr":2832,"manufId":4714,"manufName":"Namron AS","powerSource":"Mains (single phase)","modelId":"Dimmer-Switch-ZB3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Dimmer-Switch-ZB3.0","manufacturerName":"Namron AS","powerSource":1,"zclVersion":3,"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20191220","swBuildId":"3.05"}},"genLevelCtrl":{"attributes":{"currentLevel":51}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00259079a1","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00259079a1","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20191220","swBuildId":"3.05","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-526939472},"lastSeen":1688630401025,"defaultSendRequestWhen":"immediate"}

KDanielK, 2023-07-06:
@robertbakke if you change your ext_converter.js to the following, does it work?

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {type: 'Router', manufacturerID: 4714, modelID: 'Dimmer-Switch-ZB3.0'},
        {type: 'Router', manufacturerName: 'EcoDim BV', modelID: 'EcoDim-Zigbee 3.0', endpoints: [
            {ID: 1, profileID: 260, deviceID: 257, inputClusters: [0, 3, 4, 5, 6, 8, 2821, 4096], outputClusters: [25]},
            {ID: 242, profileID: 41440, deviceID: 97, inputClusters: [], outputClusters: [33]},
        ]},
        {type: 'Router', manufacturerName: 'EcoDim BV', modelID: 'EcoDim-Zigbee 3.0', endpoints: [
            {ID: 1, profileID: 260, deviceID: 257, inputClusters: [0, 3, 4, 5, 6, 8, 2821, 4096], outputClusters: [25]},
            {ID: 67, inputClusters: [], outputClusters: []},
            {ID: 242, profileID: 41440, deviceID: 97, inputClusters: [], outputClusters: [33]},
        ]},
    ],
    model: 'Eco-Dim.07/Eco-Dim.10',
    vendor: 'EcoDim',
    description: 'Zigbee & Z-wave dimmer',
    ota: ota.zigbeeOTA,
    extend: extend.light_onoff_brightness({noConfigure: true, disableEffect: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint);
    },
};

module.exports = definition;

robertbakke, 2023-07-06:
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee'); const tz = require('zigbee-herdsman-converters/converters/toZigbee'); const exposes = [...]

YES! This works in ext_converter.js with 1.32!

Thanks

KDanielK, 2023-07-06:
@robertbakke Strange, as I understand it this is the same code that is in dev. Since this is my first contribution to the project I might have done something wrong. Not sure

robertbakke, 2023-07-06:
@KDanielK Thanks. Maybe @Koenkk can answer? Are we sure the dev version I get when using latest-dev includes your code? Im not too familiar with Github and how this works :)

Anyways, thank you so much for trying to help. Its not the biggest hazzle if I need to include the ext_converter.js with every version, but it would be good to have a non customized version :)

Credo102, 2023-07-06:
> This is my device btw, I guess the hw is the same but the formfactor is different https://www.kjell.com/se/produkter/smarta-hem/fjarrkontroller/smarta-dimmers/hzc-sladdimmer-zigbee-vit-p51917

I have exactly the same issue with (almost) the same product, although my manufacturer name is "Letsleds" or "Lets leds" (depending on which of the 2 dimmer switches i use).
It used to be considered as a model: 'Eco-Dim.07/Eco-Dim.10' before update and worked perfectly well but now is unsupported".

How did you fix this? i am totally confused by the herdsman converters instructions and this thread.

KDanielK, 2023-07-06:
@Koenkk Since this problem just recently started appearing I looked into the changes in ecodim.ts and found that this line was removed:
**zigbeeModel: ['Dimmer-Switch-ZB3.0']** 

It looks like it was due to a white label version of the dimmer in this bug https://github.com/Koenkk/zigbee-herdsman-converters/issues/5859 

Since that device also has manufacturerID 4714 it should most likely be treated as a white label version of ecodim same as Mooon AB, letsleds, namron etc

It's not really great with one manufacturer ID connected to so many names.. perhaps a logic with a catch all with a generic name and then the possibillity to add more specific versions with pictures and extras? I can see that there is a whitelabel annotation in the framework but I dont know if its applicable here

Koenkk, 2023-07-06:
@KDanielK this is already fixed, all device with manufacturerID 4714 will be detected as supported again (https://github.com/Koenkk/zigbee-herdsman-converters/blob/06d2ac8c08e783d9914c3b3c92626d4a5960f005/src/devices/ecodim.ts#L40). I will try to make a hotfix release in the coming days.

jamesround86, 2023-07-23:
> I have a similar problem. Worked up to the last version. Database entry is:
>
> ```
> {"id":11,"type":"Router","ieeeAddr":"0x0c4314fffefc2102","nwkAddr":41694,"manufId":4098,"manufName":"Smart [...]

Is there any further update to this? I have the same device as quoted here too, which worked fine prior to 1.32.0-1 but ceased working after that simply showing as 'unsupported' despite it showing as supported here: https://www.zigbee2mqtt.io/devices/Eco-Dim.07_Eco-Dim.10.html

I'm afraid my technical skill is nowhere near the rest of you, so I'd be grateful to understand when a release that's a standard public update rather than dev release will address this issue and restore the functionality of the device? Or any simple workarounds in the meantime?

I did also test with 1.32.0-2 and 1.32.1-1 and I had to roll both back as the device changes to unsupported each time the update is performed, so I'm stuck with 1.31.2-1. Any advice or guidance appreciated :)

LXR85, 2023-09-25:
The issue itself has been fixed, but they are now showing up as a different device compared to before (Candeo C210):

[image]

closed 2023-11-04 by github-actions[bot]