Koenkk/zigbee2mqtt #18180: "dict object' has no attribute" after update to latest zigbee2mqtt for IR remote controller

bobofruit1970, 2023-07-02:
### What happened?

After update every restart of HA I get Warning:

"  Template variable warning: 'dict object' has no attribute 'ir_code_to_send' when rendering '{{ value_json.ir_code_to_send }}'  "

for TUYA Infrared recorder/sender

I am not using in automations any json commands. It probably comes internally from new zigbee2mqtt.

2023-07-02 07:24:46.557 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'ir_code_to_send' when rendering '{{ value_json.ir_code_to_send }}'

[image]

### What did you expect to happen?

I do not want to see this warning just like I haven't seen it for the past two years.

### How to reproduce it (minimal and precise)

Restart HA when using TUYA zigbee IR remote controller
[image]

### Zigbee2MQTT version

1.32.0-2

### Adapter firmware version

26780700

### Adapter

dresden_elektronik ConBee_II

Gugulator, 2023-07-03:
facing same issue.
it occurs up to 6 times immediately when sending any payload command to IR device via z2m.

Home Assistant 2023.6.3
Supervisor 2023.06.4
Operating System 10.3
Zigbee2MQTT version: 1.32.0-2

bobofruit1970, 2023-07-07:
I have deleted all form
automations.yaml
scripts.yaml
configuration.yaml

After restart of such clean configuration still have warning from zigbee2mqtt 
[image]

bobofruit1970, 2023-07-08:
After last update to Current version: 1.32.1-1 warning still exist.

[image]

Gugulator, 2023-07-09:
confirm same issue

jocamane, 2023-07-15:
I confirm same issue with 1.32.1-1
My router is the same and this warning appears every 10 minutes

2023-07-15 10:21:01.356 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'ir_code_to_send' when rendering '{{ value_json.ir_code_to_send }}'
2023-07-15 10:31:30.025 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'ir_code_to_send' when rendering '{{ value_json.ir_code_to_send }}'
2023-07-15 10:41:31.112 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'ir_code_to_send' when rendering '{{ value_json.ir_code_to_send }}'
2023-07-15 10:51:32.195 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'ir_code_to_send' when rendering '{{ value_json.ir_code_to_send }}'

jocamane, 2023-07-17:
@Koenkk Please help. since 00:00 I have more than 100 of these Warnings and the log is full.
I have one more device like this to configure but until this is resolved I can't
thanks

Koenkk, 2023-07-18:
I pushed a fix, please let me know if it works.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Koenkk, 2023-07-19:
Great! Next version will be released on 1 August.

closed 2023-07-19 by Koenkk