Koenkk/zigbee2mqtt #18184: Since version 1.32.0 I have errors about "Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}"

NecoPastor, 2023-07-02:
### What happened?

I did not have these error before

### How to reproduce it (minimal and precise)

I did not something else than starting Zigbee2MQTT after I updated it.

### Zigbee2MQTT version

1.32.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZBDongle-E

### Debug log

info  2023-07-02 10:36:16: MQTT publish: topic 'zigbee2mqtt/DDS238-2', payload '{"ac_frequency":50.01,"current":2.38,"energy":3696.72,"last_seen":"2023-07-02T07:36:16.175Z","linkquality":255,"power":342,"power_factor":62.5,"produced_energy":0,"state":"ON","voltage":226.9}'
debug 2023-07-02 10:36:17: Received Zigbee message from 'DDS238-2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,5,164,8],"type":"Buffer"},"datatype":2,"dp":101}],"seq":49664}' from endpoint 1 with groupID 0
error 2023-07-02 10:36:17: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
debug 2023-07-02 10:36:17: TypeError: Cannot read properties of null (reading 'from')
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:980:34)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
    at EZSPAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
    at EZSPAdapter.emit (node:events:513:28)

object, 2023-07-03:
I also get "Exception while calling fromZigbee" after upgrading to 1.32 (they may occur in relation to another issue https://github.com/Koenkk/zigbee2mqtt/issues/18199).

Koenkk, 2023-07-04:
Should be fixed now!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

fotis3d, 2023-07-05:
I also get similar errors after latest update
[image]

Sonoff Dongle Coordinator

closed 2023-08-13 by github-actions[bot]