Koenkk/zigbee2mqtt #18185: After upgrade to 1.32 WSDCGQ01LM temperature and humidity sensors stopped reporting temperature and humidity.

ramelito, 2023-07-02:
### What happened?

After upgrade to 1.32 WSDCGQ01LM temperature and humidity sensors stopped reporting temperature and humidity.
Rejoin did not help. Affected all WSDCGQ01LM sensors.

### What did you expect to happen?

After upgrade i have expected sensors to continue reporting T/H.

### Zigbee2MQTT version

1.32.0-2

### Adapter firmware version

20220219

### Adapter

Electrolama

### Debug log

[core-ssh 2023-07-02.11-33-17]$ grep aaa2 log.txt                                                                                                                                                                                                     
info  2023-07-02 11:33:23: 0x00158d000408aaa2 (0x00158d000408aaa2): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)                                                                                                               
info  2023-07-02 11:33:26: MQTT publish: topic 'homeassistant/sensor/0x00158d000408aaa2/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000408aaa2"],"manufacturer":"Xi
aomi","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","name":"0x00158d000408aaa2","sw_version":"3000-0001"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x00158d
000408aaa2","name":"0x00158d000408aaa2 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d000408aaa2","unique_id":"0x00158d000408aaa2_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}
'                                                                                                                                                                                                                                                     
info  2023-07-02 11:33:26: MQTT publish: topic 'homeassistant/sensor/0x00158d000408aaa2/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000408aaa2"],"manufacturer"
:"Xiaomi","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","name":"0x00158d000408aaa2","sw_version":"3000-0001"},"device_class":"temperature","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x00158d000408aaa2","name":"0x0
0158d000408aaa2 temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d000408aaa2","unique_id":"0x00158d000408aaa2_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'         
info  2023-07-02 11:33:26: MQTT publish: topic 'homeassistant/sensor/0x00158d000408aaa2/humidity/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000408aaa2"],"manufacturer":"X
iaomi","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","name":"0x00158d000408aaa2","sw_version":"3000-0001"},"device_class":"humidity","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x00158d000408aaa2","name":"0x00158d0
00408aaa2 humidity","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d000408aaa2","unique_id":"0x00158d000408aaa2_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'                         
info  2023-07-02 11:33:26: MQTT publish: topic 'homeassistant/sensor/0x00158d000408aaa2/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000408aaa2"],"manufacturer":"Xi
aomi","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","name":"0x00158d000408aaa2","sw_version":"3000-0001"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x00158
d000408aaa2","name":"0x00158d000408aaa2 voltage","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d000408aaa2","unique_id":"0x00158d000408aaa2_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}
"}'                                                                                                                                                                                                                                                   
info  2023-07-02 11:33:26: MQTT publish: topic 'homeassistant/sensor/0x00158d000408aaa2/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000408aaa2"],"manufacturer"
:"Xiaomi","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","name":"0x00158d000408aaa2","sw_version":"3000-0001"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00158d
000408aaa2","name":"0x00158d000408aaa2 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d000408aaa2","unique_id":"0x00158d000408aaa2_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.li
nkquality }}"}'                                                                                                                                                                                                                                       
info  2023-07-02 11:33:26: MQTT publish: topic 'homeassistant/sensor/0x00158d000408aaa2/last_seen/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000408aaa2"],"manufacturer":"
Xiaomi","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","name":"0x00158d000408aaa2","sw_version":"3000-0001"},"device_class":"timestamp","enabled_by_default":true,"entity_category":"diagnostic","icon":"mdi:clock","json_attributes_topic
":"zigbee2mqtt/0x00158d000408aaa2","name":"0x00158d000408aaa2 last seen","state_topic":"zigbee2mqtt/0x00158d000408aaa2","unique_id":"0x00158d000408aaa2_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'                        
info  2023-07-02 11:33:33: MQTT publish: topic 'zigbee2mqtt/0x00158d000408aaa2', payload '{"battery":83,"humidity":44.8,"last_seen":"2023-07-02T11:29:42+03:00","linkquality":80,"power_outage_count":35,"temperature":24.3,"voltage":2975}'          
debug 2023-07-02 11:33:33: Received MQTT message on 'homeassistant/sensor/0x00158d000408aaa2/temperature/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000408aaa2"],"manufacturer":"Xiaomi","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","name":"0x00158d000408aaa2","sw_version":"3000-0001"},"device_class":"temperature","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x00158d000408aaa2","name":"0x00158d000408aaa2 temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d000408aaa2","unique_id":"0x00158d000408aaa2_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
debug 2023-07-02 11:33:33: Received MQTT message on 'homeassistant/sensor/0x00158d000408aaa2/humidity/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000408aaa2"],"manufacturer":"Xiaomi","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","name":"0x00158d000408aaa2","sw_version":"3000-0001"},"device_class":"humidity","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x00158d000408aaa2","name":"0x00158d000408aaa2 humidity","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d000408aaa2","unique_id":"0x00158d000408aaa2_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
debug 2023-07-02 11:33:33: Received MQTT message on 'homeassistant/sensor/0x00158d000408aaa2/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000408aaa2"],"manufacturer":"Xiaomi","model":"MiJia temperature & humidity sensor (WSDCGQ01LM)","name":"0x00158d000408aaa2","sw_version":"3000-0001"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x00158d000408aaa2","name":"0x00158d000408aaa2 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d000408aaa2","unique_id":"0x00158d000408aaa2_battery_zigbee2mqtt","unit_of_measurement":"%","value_templa [... cut ...]
info  2023-07-02 11:34:09: MQTT publish: topic 'zigbee2mqtt/0x00158d000408aaa2', payload '{"battery":83,"humidity":44.8,"last_seen":"2023-07-02T11:34:09+03:00","linkquality":72,"power_outage_count":35,"temperature":24.3,"voltage":2975}'
debug 2023-07-02 11:34:09: Received Zigbee message from '0x00158d000408aaa2', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":4374}' from endpoint 1 with groupID 0
info  2023-07-02 11:34:09: MQTT publish: topic 'zigbee2mqtt/0x00158d000408aaa2', payload '{"battery":83,"humidity":44.8,"last_seen":"2023-07-02T11:34:09+03:00","linkquality":72,"power_outage_count":35,"temperature":24.3,"voltage":2975}'

ramelito, 2023-07-02:
One notice, temperature and humidity are not chaning over time. I am getting the same value for hours.

ramelito, 2023-07-02:
More debug logs

Zigbee2MQTT:debug 2023-07-02 13:32:16: Received Zigbee message from '0x60a423fffe39df47', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":4168}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2023-07-02 13:32:16: Exception while calling fromZigbee converter: 'humidity_precision' is not a number, got object ([object Object])}
Zigbee2MQTT:debug 2023-07-02 13:32:16: Error: 'humidity_precision' is not a number, got object ([object Object])
    at assertNumber (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:502:42)
    at calibrateAndPrecisionRoundOptions (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:103:5)
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/converters/fromZigbee.js:440:35)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)

ramelito, 2023-07-02:
I have to comment out this lines

'0x00158d000408aaa2':
  #  temperature_precision:
  #  '10': 1
  #  '40': 0
  #humidity_precision:
  #  '10': 1
  #  '80': 0

and after this T/H started to update values

tishcufksips, 2023-08-08:
Unfortunately I have the same or very similar issue with all my WSDCGQ01LM sensor. My issue is that I've got 4 pcs and I can pair them without a problem, they work ok for couple of hours and then all stop updating at the same time.
I've tried mentioned solutions, but unfortunately they don't work for me.

arteck, 2023-09-14:
the new one with problem .. other all without
[image]

@Koenkk i see the herdsman has new battery functions... is here the problem ??

Koenkk, 2023-09-14:
Can you provide the debug log when you trigger the sensor? (e.g. blow into it to trigger humidity sensor)

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

arteck, 2023-09-15:
strange ... 
[image]

in the past you could do this with the button press..

arteck, 2023-09-15:
here other device same problem

[image]

here debug info 
[neu1.txt](https://github.com/Koenkk/zigbee2mqtt/files/12617937/neu1.txt)

Koenkk, 2023-09-16:
Does it still not report any battery after a day of working?

arteck, 2023-09-16:
sure.. but not on press the button.. like before

Koenkk, 2023-09-16:
can you post the debug log when downgrading z2m snd this still worked?

closed 2023-10-25 by github-actions[bot]

DooMMasteR, 2024-03-21:
I have had the issue today...