Koenkk/zigbee2mqtt #18186: Error flashing CC2652R_coordinator_20230507 firmware ZZH!

fjcerdan27, 2023-07-02:
### What happened?

Hello:

When I try to flash the new firmware with TI Flash Programmer 2 I get this message in Log:

>Initiate access to target: COM4 using 2-pin cJTAG.
>Reading file: C:/CC2652R_coordinator_20230507.hex.
>Unknown record type: 3.
>Reset target ...
>Reset of target successful.

My device is an Electroalma Electroalma ZZH! 

The last firmware flashed succesfully is CC2652R_coordinator_20220928.hex

I downloaded the firmware a few times, with de same result.

### What did you expect to happen?

To flash the new firmware succesfully.

### How to reproduce it (minimal and precise)

Flash the device with Texas Instruments Flash Programmer 2.

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

CC2652R_coordinator_20230507

### Adapter

Electroalma ZZH! CC2652R Chip

### Debug log

>Initiate access to target: COM4 using 2-pin cJTAG.
>Reading file: C:/CC2652R_coordinator_20230507.hex.
>Unknown record type: 3.
>Reset target ...
>Reset of target successful.

lux73, 2023-07-02:
flashing with cc2538-bsl.py Script should work -> https://www.zigbee2mqtt.io/guide/adapters/#flashing-cc1352-cc2652-cc2538-based-adapters

https://github.com/JelmerT/cc2538-bsl

fjcerdan27, 2023-07-02:
Thanks @lux73 , it works.

I've downloaded python script and flashed with the sentence:

$ sudo python3 cc2538-bsl.py -p /dev/ttyUSB0 -evw CC2652R_coordinator_20230507.hex

👍

closed 2023-08-09 by github-actions[bot]