Koenkk/zigbee2mqtt #18187: SYMFONISK sound remote gen2 stops working after OTA

weaselchops, 2023-07-02:
### What happened?

I updated the using OTA to version 16777269 firmware.
The buttons press were being registered correctly and were being passed to HA correctly

I then did a re-register to make sure the new structure was being picked up (but was probably working correctly)
The device registers and interview works.
Everything looks ok, but no button presses are being registered in the Z2M gui or via the Home Assistant!

This working after the OTA, but not after the re-register! 

### What did you expect to happen?

Button presses to be picked up

### How to reproduce it (minimal and precise)

Remove device and re add.

### Zigbee2MQTT version

1.32.0

### Adapter firmware version

20210708

### Adapter

Sonoff zigbee 3.0 USB dongle plus E

### Debug log

Zigbee2MQTT:info  2023-07-02 12:15:01: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x540f57fffe21f248","ieee_address":"0x540f57fffe21f248"},"type":"device_announce"}'
Zigbee2MQTT:debug 2023-07-02 12:15:03: Received Zigbee message from '0x540f57fffe21f248', type 'readResponse', cluster 'genPollCtrl', data '{"checkinInterval":172800}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-07-02 12:15:03: Successfully interviewed '0x540f57fffe21f248', device has successfully been paired
Zigbee2MQTT:info  2023-07-02 12:15:03: Device '0x540f57fffe21f248' is supported, identified as: IKEA SYMFONISK sound remote gen2 (E2123)
Zigbee2MQTT:info  2023-07-02 12:15:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"SYMFONISK sound remote gen2","exposes":[{"access":5,"description":"Remaining battery in %, can take up to 24 hours before reported.","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["toggle","track_previous","track_next","volume_up","volume_down","volume_up_hold","volume_down_hold","dots_1_initial_press","dots_2_initial_press","dots_1_long_press","dots_2_long_press","dots_1_short_release","dots_2_short_release","dots_1_long_release","dots_2_long_release","dots_1_double_press","dots_2_double_press"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"E2123","options":[{"access":2,"description":"Set to false to disable the legacy integration (highly recommended), will change structure of the published payload (default true).","name":"legacy","property":"legacy","type":"binary","value_off":false,"value_on":true}],"supports_ota":true,"vendor":"IKEA"},"friendly_name":"0x540f57fffe21f248","ieee_address":"0x540f57fffe21f248","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:debug 2023-07-02 12:15:41: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2023-07-02 12:15:51: Received MQTT message on 'zigbee2mqtt/bridge/request/device/options' with data '{"id":"0x540f57fffe21f248","options":{"friendly_name":"0x540f57fffe21f248","homeassistant":{"name":"Ikea Remote"},"legacy":false,"optimistic":true},"transaction":"hv830-6"}'
Zigbee2MQTT:info  2023-07-02 12:15:51: Changed config for device 0x540f57fffe21f248
Zigbee2MQTT:info  2023-07-02 12:15:51: MQTT publish: topic 'homeassistant/sensor/0x540f57fffe21f248/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x540f57fffe21f248"],"manufacturer":"IKEA","model":"SYMFONISK sound remote gen2 (E2123)","name":"Ikea Remote","sw_version":"1.0.35"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"Ikea Remote battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x540f57fffe21f248","unique_id":"0x540f57fffe21f248_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:info  2023-07-02 12:15:51: MQTT publish: topic 'homeassistant/sensor/0x540f57fffe21f248/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x540f57fffe21f248"],"manufacturer":"IKEA","model":"SYMFONISK sound remote gen2 (E2123)","name":"Ikea Remote","sw_version":"1.0.35"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"Ikea Remote action","state_topic":"zigbee2mqtt/0x540f57fffe21f248","unique_id":"0x540f57fffe21f248_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Zigbee2MQTT:info  2023-07-02 12:15:51: MQTT publish: topic 'homeassistant/sensor/0x540f57fffe21f248/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x540f57fffe21f248"],"manufacturer":"IKEA","model":"SYMFONISK sound remote gen2 (E2123)","name":"Ikea Remote","sw_version":"1.0.35"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Ikea Remote linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x540f57fffe21f248","unique_id":"0x540f57fffe21f248_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2023-07-02 12:15:51: MQTT publish: topic 'homeassistant/sensor/0x540f57fffe21f248/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x540f57fffe21f248"],"manufacturer":"IKEA","model":"SYMFONISK sound remote gen2 (E2123)","name":"Ikea Remote","sw_version":"1.0.35"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","name":"Ikea Remote update state","state_topic":"zigbee2mqtt/0x540f57fffe21f248","unique_id":"0x540f57fffe21f248_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Zigbee2MQTT:info  2023-07-02 12:15:51: MQTT publish: topic 'homeassistant/binary_sensor/0x540f57fffe21f248/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x540f57fffe21f248"],"manufacturer":"IKEA","model":"SYMFONISK sound remote gen2 (E2123)","name":"Ikea Remote","sw_version":"1.0.35"},"device_class":"update","enabled_by_default":false,"entity_category":"diagnostic","name":"Ikea Remote update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x540f57fffe21f248","unique_id":"0x540f57fffe21f248_update_available_zigbee2mqtt","value_template":"{{ value_json['update']['state'] == \"available\" }}"}'
Zigbee2MQTT:info  2023-07-02 12:15:51: MQTT publish: topic 'homeassistant/update/0x540f57fffe21f248/update/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/bridge/request/device/ota_update/update","device":{"identifiers":["zigbee2mqtt_0x540f57fffe21f248"],"manufacturer":"IKEA","model":"SYMFONISK sound remote gen2 (E2123)","name":"Ikea Remote","sw_version":"1.0.35"},"device_class":"firmware","entity_picture":"https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png","latest_version_template":"{{ value_json['update']['latest_version'] }}","latest_version_topic":"zigbee2mqtt/0x540f57fffe21f248","name":"Ikea Remote","payload_install":"{\"id\": \"0x540f57fffe21f248\"}","state_topic":"zigbee2mqtt/0x540f57fffe21f248","unique_id":"0x540f57fffe21f248_update_zigbee2mqtt","value_template":"{{ value_json['update']['installed_version'] }}"}'
Zigbee2MQTT:info  2023-07-02 12:15:51: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{"legacy":false},"id":"0x540f57fffe21f248","restart_required":false,"to":{"homeassistant":{"name":"Ikea Remote"},"legacy":false,"optimistic":true}},"status":"ok","transaction":"hv830-6"}'
Zigbee2MQTT:debug 2023-07-02 12:15:51: Received MQTT message on 'homeassistant/sensor/0x540f57fffe21f248/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x540f57fffe21f248"],"manufacturer":"IKEA","model":"SYMFONISK sound remote gen2 (E2123)","name":"Ikea Remote","sw_version":"1.0.35"},"device_class":"battery","enabled_b [... cut ...] _version":"1.0.35"},"device_class":"firmware","entity_picture":"https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png","latest_version_template":"{{ value_json['update']['latest_version'] }}","latest_version_topic":"zigbee2mqtt/0x540f57fffe21f248","name":"Ikea Remote","payload_install":"{\"id\": \"0x540f57fffe21f248\"}","state_topic":"zigbee2mqtt/0x540f57fffe21f248","unique_id":"0x540f57fffe21f248_update_zigbee2mqtt","value_template":"{{ value_json['update']['installed_version'] }}"}'

weaselchops, 2023-07-02:
Hmm this seems to happen after a re-register:

2023-07-02 13:14:17Configuring '0x540f57fffe21f248'

Error 2023-07-02 13:14:29Failed to configure '0x540f57fffe21f248', attempt 1 (Error: Bind 0x540f57fffe21f248/1 genLevelCtrl from '0x00124b00257e07f4/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))

weaselchops, 2023-07-02:
Hmm, many reboots and re-registers...
It finally works!
No idea why! 😪

closed 2023-07-02 by weaselchops

philippsandhaus, 2024-01-06:
Hmm, I have the same problem currently with two remotes. When I try to re-register, I get the following log:
```
2024-01-06 21:36:02Device '0x94deb8fffe9b679f' announced itself
Info 2024-01-06 21:36:02MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x94deb8fffe9b679f","ieee_address":"0x94deb8fffe9b679f"},"type":"device_announce"}'
Info 2024-01-06 21:36:02MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x94deb8fffe9b679f"},"type":"device_announced"}'
Debug 2024-01-06 21:36:04Retrieving state of '0x94deb8fffe9b679f' after reconnect
Info 2024-01-06 21:36:10MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe9b679f', payload '{"action":null,"last_seen":"2024-01-06T21:36:10+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-06 21:36:10MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe9b679f', payload '{"action":null,"last_seen":"2024-01-06T21:36:10+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-06 21:36:10MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe9b679f', payload '{"action":null,"last_seen":"2024-01-06T21:36:10+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-06 21:36:10MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe9b679f', payload '{"action":null,"last_seen":"2024-01-06T21:36:10+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-06 21:36:10MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe9b679f', payload '{"action":null,"last_seen":"2024-01-06T21:36:10+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-06 21:36:10MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe9b679f', payload '{"action":null,"last_seen":"2024-01-06T21:36:10+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-06 21:36:10MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe9b679f', payload '{"action":null,"last_seen":"2024-01-06T21:36:10+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-06 21:36:10MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe9b679f', payload '{"action":null,"last_seen":"2024-01-06T21:36:10+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-06 21:36:10MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe9b679f', payload '{"action":null,"last_seen":"2024-01-06T21:36:10+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-06 21:36:10MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe9b679f', payload '{"action":null,"last_seen":"2024-01-06T21:36:10+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-06 21:36:10MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe9b679f', payload '{"action":null,"last_seen":"2024-01-06T21:36:10+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-06 21:36:10MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe9b679f', payload '{"action":null,"last_seen":"2024-01-06T21:36:10+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-06 21:36:10MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe9b679f', payload '{"action":null,"last_seen":"2024-01-06T21:36:10+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-01-06 21:36:10MQTT publish: topic 'zigbee2mqtt/0x94deb8fffe9b679f', payload '{"action":null,"last_seen":"2024-01-06T21:36:10+01:00","linkquality":0,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
```

The remotes do successfully pair, but I get no events when pressing any button. I was however abet to update the remotes' firmware. Does anybody have an idea how to solve this?