Koenkk/zigbee2mqtt #18190: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}

tax987, 2023-07-02:
### What happened?

Hi,

I'm using Zigbee2mqtt from about 3years without problems, from 1.32 version start to popup this alarm:

`Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}`

Right now I'm using last version 1.32.0.2, I've tryed also with Edge but nothing changes.
Below you can find part of logs, hope that will be helpful.

Thanks

### Zigbee2MQTT version

1.32.0.2

### Adapter firmware version

20230507

### Adapter

ITEAD Sonoff ZigBee 3.0 USB Dongle Plus v1 - zStack3x0

### Debug log

Zigbee2MQTT:debug 2023-07-02 20:32:47: Received Zigbee message from 'EnM_Genarale', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,1,3,161],"type":"Buffer"},"datatype":2,"dp":109}],"seq":36608}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2023-07-02 20:32:47: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:debug 2023-07-02 20:32:47: TypeError: Cannot read properties of null (reading 'from')
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:980:34)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
    at ZStackAdapter.emit (node:events:513:28)
Zigbee2MQTT:info  2023-07-02 20:32:47: MQTT publish: topic 'zigbee2mqtt/EnM_Genarale', payload '{"ac_frequency":49.97,"current":2.96,"energy":1419.69,"last_seen":"2023-07-02T20:32:47+02:00","linkquality":43,"power":641,"power_factor":90.7,"produced_energy":0,"state":"ON","voltage":241.3}'
Zigbee2MQTT:debug 2023-07-02 20:32:47: Received Zigbee message from 'EV_GP1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,60],"type":"Buffer"},"datatype":2,"dp":108}],"seq":25856}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-07-02 20:32:47: MQTT publish: topic 'zigbee2mqtt/EV_GP1', payload '{"battery":60,"cycle_irrigation_interval":0,"cycle_irrigation_num_times":0,"irrigation_end_time":"07:35:23","irrigation_start_time":"07:25:23","irrigation_target":600,"last_irrigation_duration":"00:10:00","last_seen":"2023-07-02T20:32:47+02:00","linkquality":193,"mode":"duration","state":"OFF","water_consumed":39}'
Zigbee2MQTT:debug 2023-07-02 20:32:48: Received Zigbee message from 'EnM_Genarale', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,165],"type":"Buffer"},"datatype":2,"dp":110}],"seq":36864}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2023-07-02 20:32:48: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:debug 2023-07-02 20:32:48: TypeError: Cannot read properties of null (reading 'from')
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:980:34)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
    at ZStackAdapter.emit (node:events:513:28)
Zigbee2MQTT:info  2023-07-02 20:32:48: MQTT publish: topic 'zigbee2mqtt/EnM_Genarale', payload '{"ac_frequency":49.97,"current":2.96,"energy":1419.69,"last_seen":"2023-07-02T20:32:48+02:00","linkquality":43,"power":641,"power_factor":90.7,"produced_energy":0,"state":"ON","voltage":241.3}'
Zigbee2MQTT:debug 2023-07-02 20:32:49: Received Zigbee message from 'EnM_Genarale', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,162],"type":"Buffer"},"datatype":2,"dp":111}],"seq":37120}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-07-02 20:32:49: MQTT publish: topic 'zigbee2mqtt/EnM_Genarale', payload '{"ac_frequency":49.97,"current":2.96,"energy":1419.69,"last_seen":"2023-07-02T20:32:49+02:00","linkquality":43,"power":641,"power_factor":93,"produced_energy":0,"state":"ON","voltage":241.3}'
Zigbee2MQTT:debug 2023-07-02 20:32:51: Received Zigbee message from 'EnM_Genarale', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":5,"dp":9}],"seq":37376}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2023-07-02 20:32:51: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:debug 2023-07-02 20:32:51: TypeError: Cannot read properties of null (reading 'from')
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:980:34)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
[... 12 lines cut ...]
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
    at ZStackAdapter.emit (node:events:513:28)
Zigbee2MQTT:info  2023-07-02 20:32:52: MQTT publish: topic 'zigbee2mqtt/EnM_Genarale', payload '{"ac_frequency":49.97,"current":2.96,"energy":1419.69,"last_seen":"2023-07-02T20:32:52+02:00","linkquality":43,"power":641,"power_factor":93,"produced_energy":0,"state":"ON","voltage":241.3}'
Zigbee2MQTT:debug 2023-07-02 20:32:52: Received Zigbee message from 'EnM_Genarale', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,5,0,100,0,5,0,30,0,60,0,0,0,0,0,0,0],"type":"Buffer"},"datatype":0,"dp":18}],"seq":37888}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2023-07-02 20:32:52: Exception while calling fromZigbee converter: Cannot read properties of null (reading 'from')}
Zigbee2MQTT:debug 2023-07-02 20:32:52: TypeError: Cannot read properties of null (reading 'from')
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:980:34)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
    at ZStackAdapter.emit (node:events:513:28)
Zigbee2MQTT:info  2023-07-02 20:32:52: MQTT publish: topic 'zigbee2mqtt/EnM_Genarale', payload '{"ac_frequency":49.97,"current":2.96,"energy":1419.69,"last_seen":"2023-07-02T20:32:52+02:00","linkquality":40,"power":641,"power_factor":93,"produced_energy":0,"state":"ON","voltage":241.3}'
Zigbee2MQTT:debug 2023-07-02 20:32:56: Received Zigbee message from 'EnM_Genarale', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,19,135],"type":"Buffer"},"datatype":2,"dp":105}],"seq":38144}' from endpoint 1 with groupID 0
[... cut ...]
Zigbee2MQTT:info  2023-07-02 20:33:03: MQTT publish: topic 'zigbee2mqtt/sm_cucina', payload '{"battery":100,"device_temperature":32,"illuminance":56,"illuminance_lux":56,"last_seen":"2023-07-02T20:33:03+02:00","linkquality":102,"occupancy":true,"power_outage_count":46,"temperature":30,"voltage":3065}'
Zigbee2MQTT:info  2023-07-02 20:33:03: MQTT publish: topic 'zigbee2mqtt/sm_cucina', payload '{"battery":100,"device_temperature":32,"illuminance":56,"illuminance_lux":56,"last_seen":"2023-07-02T20:33:03+02:00","linkquality":102,"occupancy":true,"power_outage_count":46,"temperature":30,"voltage":3065}'

chris94440, 2023-07-03:
Hello
Same behaviour ...
Chris94440

Meishop, 2023-07-04:
Hello,
same here...
Thank you for looking into it!

chris94440, 2023-07-04:
same issue as https://github.com/Koenkk/zigbee2mqtt/issues/18184 ... fix in dev branch

tax987, 2023-07-05:
Hi,

Probably I don't do the right procedure because as mentioned in first post I've already tried with ZigBee 2Mqtt Edge version on hassos, but error remain.

Could you please help me to solve?

Thanks