Koenkk/zigbee2mqtt #18192: Stability problems with SLZB-06

ict-one-nl, 2023-07-03:
### What happened?

I'm having more and more stability issues with the SLZB-06. Occasionaly Zigbee2Mqtt loses connectivity all together and Zigbee2MQTT restarts because the connection to the adapter was gone (causing stuff to go unavailable in HA, which triggers some rules). I have created a ticket for this to make Z2M a bit more robust.

Now I'm experiencing something else, see the logs below. Z2M can't send any data anymore and just stops working. Restarting Z2M solves it for a while but then it occurs again. This is even more annoying because it doesn't rectify itself.

### What did you expect to happen?

Z2M keeps working and handles a momentary loss of connectivity more gracefully.

### How to reproduce it (minimal and precise)

Don't know, problem is intermittent.

### Zigbee2MQTT version

1.32.0 commit: 69638d7

### Adapter firmware version

20221226

### Adapter

SLZB-06 0.9.9 (Mar 3 2023)

### Debug log

```
Zigbee2MQTT:error 2023-07-03 10:09:53: Publish 'set' 'state' to 'Philips Hue light office Sander printer' failed: 'Error: Command 0x001788010b2766fd/11 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Cannot request when znp has not been initialized yet)'
Zigbee2MQTT:info  2023-07-03 10:09:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Philips Hue light office Sander printer' failed: 'Error: Command 0x001788010b2766fd/11 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Cannot request when znp has not been initialized yet)'","meta":{"friendly_name":"Philips Hue light office Sander printer"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2023-07-03 10:09:53: Publish 'set' 'state' to 'Philips Hue light office Sander desk left' failed: 'Error: Command 0x001788010b277b9f/11 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Cannot request when znp has not been initialized yet)'
Zigbee2MQTT:info  2023-07-03 10:09:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Philips Hue light office Sander desk left' failed: 'Error: Command 0x001788010b277b9f/11 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Cannot request when znp has not been initialized yet)'","meta":{"friendly_name":"Philips Hue light office Sander desk left"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2023-07-03 10:09:53: Publish 'set' 'state' to 'Tradfri switch pump floor heating' failed: 'Error: Command 0xbc33acfffe69c735/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Cannot request when znp has not been initialized yet)'
Zigbee2MQTT:info  2023-07-03 10:09:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Tradfri switch pump floor heating' failed: 'Error: Command 0xbc33acfffe69c735/1 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Cannot request when znp has not been initialized yet)'","meta":{"friendly_name":"Tradfri switch pump floor heating"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2023-07-03 10:09:53: Publish 'set' 'state' to 'Tradfri switch fans AEG inverter' failed: 'Error: Command 0x84b4dbfffec50962/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Cannot request when znp has not been initialized yet)'
Zigbee2MQTT:info  2023-07-03 10:09:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Tradfri switch fans AEG inverter' failed: 'Error: Command 0x84b4dbfffec50962/1 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Cannot request when znp has not been initialized yet)'","meta":{"friendly_name":"Tradfri switch fans AEG inverter"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2023-07-03 10:10:03: Publish 'set' 'state' to 'Tradfri switch pump floor heating' failed: 'Error: Command 0xbc33acfffe69c735/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Cannot request when znp has not been initialized yet)'
Zigbee2MQTT:info  2023-07-03 10:10:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Tradfri switch pump floor heating' failed: 'Error: Command 0xbc33acfffe69c735/1 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Cannot request when znp has not been initialized yet)'","meta":{"friendly_name":"Tradfri switch pump floor heating"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2023-07-03 10:10:13: Publish 'set' 'state' to 'Tradfri switch pump floor heating' failed: 'Error: Command 0xbc33acfffe69c735/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Cannot request when znp has not been initialized yet)'
Zigbee2MQTT:info  2023-07-03 10:10:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Tradfri switch pump floor heating' failed: 'Error: Command 0xbc33acfffe69c735/1 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Cannot request when znp has not been initialized yet)'","meta":{"friendly_name":"Tradfri switch pump floor heating"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2023-07-03 10:10:23: Publish 'set' 'state' to 'Tradfri switch pump floor heating' failed: 'Error: Command 0xbc33acfffe69c735/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Cannot request when znp has not been initialized yet)'
Zigbee2MQTT:info  2023-07-03 10:10:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Tradfri switch pump floor heating' failed: 'Error: Command 0xbc33acfffe69c735/1 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disab [... cut ...]
Zigbee2MQTT:info  2023-07-03 10:10:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Tradfri switch pump floor heating' failed: 'Error: Command 0xbc33acfffe69c735/1 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Cannot request when znp has not been initialized yet)'","meta":{"friendly_name":"Tradfri switch pump floor heating"},"type":"zigbee_publish_error"}'
```

sergge1, 2023-07-04:
1. Zigbee firmware is not the latest one
2. try to change Zigbee channel to something different from 11

ict-one-nl, 2023-08-04:
Switched back to the development board over usb. Works better.

rsporsche, 2023-11-19:
I think I may also be having a similar issue although infrequently.

I'm using an SLZB-06 in LAN mode with zigbee2mqtt and back in July while I was on holiday the SLZB-06 stopped responding. I was able to remotely power cycle it via the POE switch and it's been working fine until last night when it did the same thing.

I have 0.9.9 and not sure which zigbee firmware but I did update it earlier this year. Is there a way to tell the version of the zigbee firmware?

weelad, 2024-01-27:
> Switched back to the development board over usb. Works better.

@sandervandegeijn when you say you've switched back to the "development board" and "over usb" are you saying:

You're using a development firmware and running it over usb as opposed to PoE. 

Which firmware version are you using?

I'm having the same issues and it's getting quite frustrating.

sandervandegeijn, 2024-01-27:
Haven't been using it since. The whole point was to use it through PoE but connections failed too many times and Zigbee2mqtt is not robust at all when it comes to connection management.

The USB options are much more stable though not perfect as well. For me this is the one and only drawback of Zigbee2mqtt, throughout the years it has has very stable periods alternated with connection issues once a month - 2 months. But if you're on holiday when that happens it's extremely annoying.

I really would like more robustness.

sandervandegeijn, 2024-01-27:
I can reopen the issue if that helps. I can't provide more debug info since I'm not using it anymore.

infestdead, 2024-11-27:
I think I hit the issue you are describing - also over PoE. What firmware and zigbee firmware should I be using? I'm using the ones that the device came with: core v2.5.4.dev, ZigBee 20221226.

sandervandegeijn, 2024-12-01:
I'm not using this adapter anymore, but had similar problems with my other adapter. Updating to the latest firmware and the new zigbee2mqtt releases has improved stability remarkably. It's rock solid for a few months now!

federicotravaini, 2024-12-16:
Hi, I' facing the same issue but I want to describe it:
* coordinator is online, web ui is working, fw updated to latest version (20240710 and v2.6.8.dev16)
* z2m starts fine (v 1.42), I can add devices and they work
* after a while (might be hours) it just stops working
* coordinator is still online and web ui is working
* z2m restart and can't connect to coordinator anymore

Super frustrating, might just go back and switch to some old USB stick that has been working for years just fine.

mschingel, 2024-12-18:
@federicotravaini I am also in need of a solution, I am connected via POE, I havent connected to USB / WIFI.  But suddenly My zigbee devices are not responding, I can goto the web ui and see its on.  I goto HA and zigbee2mqtt and the add-on is like in a frozen state.

martini-k, 2025-01-06:
Same issue and after restart of SL-ZB06 I see the following logs (on SL-ZB06 logs page):
[19:52:20] setup | Starting firmware: v2.5.8
[19:52:21] ConfigHelper | LittleFS mounted
[19:52:21] ConfigHelper | load config
[19:52:21] ConfigHelper | config open: Ok
[19:52:21] initConfig | Config loaded
[19:52:21] setup | Reboot reason: SOFTWARE
[19:52:21] setup | Coordinator mode: USB
[19:52:21] setup | Device type: SLZB-06
[19:52:21] setup | Radio mode: "ZB COORD" Radio FW version: 20221226 Radio FW CH: PROD
[19:52:21] Network | init
[19:52:21] Network | Try ETH
[19:52:21] L_Y,L_B | status: 1
[19:52:22] setup | Filesystem size: 3456
[19:52:22] setup | Filesystem used: 184
[19:52:22] setup | done
[19:52:23] Network | EVENT_ETH_START
[19:52:23] Network | EVENT_ETH_CONNECTED
[19:52:23] Network | ETH init: OK
[19:52:23] Network | ETH static IP config: OK
[19:52:23] Network | Use ETH
[19:52:23] Network | [MDNS] Started
[19:52:23] Network | EVENT_ETH_GOT_IP
[19:52:23] Network | ETH MAC: FC:E8:C0:F8:xx.xx IPv4: 192.168.1.40 GW: 192.168.1.1 Speed: 100Mbps DNS1: 8.8.8.8 DNS2: 8.8.4.4
[19:52:23] Network | fireNetworkUp
[19:52:23] Web | Webserver started
[19:52:23] time | time sync start
[20:52:23] time | timezone: CET-1CEST,M3.5.0,M10.5.0/3
[20:52:23] time | Monday, January 06 2025 20:52:23
[20:52:23] taskTimeSync | Heap: 1504
[20:52:23] time | stop task
[20:52:30] internet | connected
[20:52:34] Network | [POST] result: 200
[20:52:34] Network | [POST] response: Previous records removed successfully.New record [on] created successfullyNew record [lo] created successfully
[20:52:34] stats | Statistics sent
[20:52:34] stats | Heap: 2384
[20:52:41] [ZBCHK] | Wrong answer
[20:52:41] [ZBVER] | Unknown
[20:52:42] [ZBCHK] | done
[20:52:55] readFactoryIeee | sblSync failed

Also, there's an error on the Z2M and ZHA page:
[image]

mschingel, 2025-01-06:
I fixed mine by factory resetting. Re writing the original address. Then updating the firmware.

martini-k, 2025-01-06:
> I fixed mine by factory resetting. Re writing the original address. Then updating the firmware.

I wrote the original address, updated to the most recent firmware and now I see this:
[image]

What a piece of crap :( I'm switching back to Conbee II. It's been working for years.

mschingel, 2025-01-06:
i had very similar issue with the sonoff controller so go figure.  I think
it was a combination of issues frankly but I was as upset as you seem.

martini-k, 2025-01-06:
> i had very similar issue with the sonoff controller so go figure. I think it was a combination of issues frankly but I was as upset as you seem.
> […](#)

I wanted to move my HA box to my server rack, but since I don't have a good signal there, I tried switching my ZB network to SL-ZB06 wired and powered by PoE. Apparently, this dongle is not stable enough for my purpose. I'll stay on my HA box in my living room with Conbee II, and I'll create a small separate SL-ZB06 network with non-critical devices for testing purposes.

fkuiper, 2025-01-08:
I've got a lot of problems with this coordinator as well. Bought it to be able to virtualize my HA setup (now running on a RP4) but have nothing but troubles ever since. Rebuild and repaired all of my 80+ devices. Mostly lights, switches and power plugs from IKEA, Philips and INNR which are all supported by Z2M and all worked fine with my years old Conbee II USB dongle. The network seems stable for a couple of hours (sometimes 24hours) but then devices begin to drop off and become unavailable. Restarting Z2M doesn't work and I have to powercycle the SLZB-06 before devices become available again... I've moved the coordinator around (it was originally in the same spot as my Conbee II has been for years), switched channels (which made me repair most of my devices again) increased transmitter power, decreased it again... 

To make things worse: with this coordinator I somehow am unable to join switches to groups in and then direct bind them so lights will work in case the HA or coordinator goes down... and since this doesn't work and the coordinator goes down every 24hours or so... my spouse-approval-rate has taken a serious hit.

I've decided to switch back to the Conbee next weekend and rebuild the entire network... again...

sandervandegeijn, 2025-01-08:
I've been able to switch out the coordinators by overwriting the IEEE/mac addresses during flashing. Saved me from repairing 60 devices...

Kingofthecarts, 2025-02-28:
I've been crawling forums for a bit and noticed no one said anything about this setting - Has anyone tried? I think if web is off, no web to crash?

[image]

I'm also trying the HA integration disabled this week. I've been trying a few different firmwares, but I also have two SLZB-06 ready to go as extras on standby to give further testing.

Thoughts on that setting or has anyone tested this? Now with the most recent firmware at least it reboots when things start to hang up

danielherrero, 2025-12-17:
I have been using the dual antenna version for a couple of days. After 24 hours the webUI stop working and I have to reset the device. Any idea? Its connected with ethernet and USB powered