Koenkk/zigbee2mqtt #18209: After 1.32.0-2 update,JS file became unavailable

lcooop, 2023-07-04:
### What happened?

**After updating Z2M to version 1.32.0-2, myexternal converter.JS file caused Z2M to **fail to start.**** 
However, this JS file was able to start Z2M and function properly in version 1.30. 

**_Here is my Logs of Z2M Add-on._**

[22:54:38] INFO: Preparing to start...
[22:54:39] INFO: Socat not enabled
[22:54:41] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:79
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

**_Here is my JS file._**
const fz = require('zigbee-herdsman-converters/converters/fromZigbee')
const tz = require('zigbee-herdsman-converters/converters/toZigbee')
const exposes = require('zigbee-herdsman-converters/lib/exposes')
const reporting = require('zigbee-herdsman-converters/lib/reporting')
const extend = require('zigbee-herdsman-converters/lib/extend')
const utils = require('zigbee-herdsman-converters/lib/utils')
const e = exposes.presets
const ea = exposes.access
const tuya = require('zigbee-herdsman-converters/lib/tuya')

//utils.mapNumberRange(value, 1, 255, 0, 1000)

const fzLocal = {
  tuya_switch_8k: {
    cluster: 'manuSpecificTuya',
    type: ['commandDataResponse', 'commandDataReport'],
    convert: (model, msg, publish, options, meta) => {
      const dpValue = tuya.firstDpValue(msg, meta, 'tuya_switch')
      const dp = dpValue.dp
      const value = tuya.getDataValue(dpValue)
      const state = value ? 'ON' : 'OFF'
      const lookup = { 1: 'l1', 2: 'l2', 3: 'l3', 4: 'l4' }
      let endpoint = lookup[dp]
      if (endpoint in model.endpoint(msg.device)) {
        return { [`state_${endpoint}`]: state }
      }

      const lookupB = { 144: 'l1', 145: 'l2', 146: 'l3', 147: 'l4' }
      endpoint = lookupB[dp]
      if (endpoint) {
        return {
          [`brightness_${endpoint}`]: value,
        }
      }

      const lookupS = { 128: 'p1', 129: 'p2', 130: 'p3', 131: 'p4' }
      endpoint = lookupS[dp]
      if (endpoint) {
        return { [`action_${endpoint}`]: 'click' }
      }

      return null
    },
  },
}

const tzLocal = {
  tuya_switch_state: {
    key: ['state'],
    convertSet: async (entity, key, value, meta) => {
      const lookup = { l1: 1, l2: 2, l3: 3, l4: 4 }
      const keyid = lookup[meta.endpoint_name]
      await tuya.sendDataPointBool(entity, keyid, value === 'ON')
    },
  },
  tuya_switch_brightness: {
    key: ['brightness'],
    convertSet: async (entity, key, value, meta) => {
      const lookup = { l1: 144, l2: 145, l3: 146, l4: 147 }
      const keyid = lookup[meta.endpoint_name]
      await tuya.sendDataPointValue(entity, keyid, value)
    },
  },
}

const definition = {
  // Since a lot of Tuya devices use the same modelID, but use different data points
  // it's usually necessary to provide a fingerprint instead of a zigbeeModel
  fingerprint: [
    {modelID: 'TS0601', manufacturerName: '_TZE200_bxiw6yq3'},
  ],
  model: 'STS0601',
 // model: 'SH-BP8L4Z',
  vendor: 'Sicoo',
  description: '8keys-4gang-4scene',
  fromZigbee: [
    fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
    //fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    fzLocal.tuya_switch_8k,
  ],
  toZigbee: [
    tzLocal.tuya_switch_state,
    tzLocal.tuya_switch_brightness,
    tz.tuya_data_point_test,
  ],
  onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
  configure: async (device, coordinatorEndpoint, logger) => {
    const endpoint = device.getEndpoint(1)
    await reporting.bind(endpoint, coordinatorEndpoint, [
      'genBasic',
      'manuSpecificTuya',
      'genLevelCtrl',
    ])
  },
  meta: { multiEndpoint: true },
  endpoint: (device) => {
    // Endpoint selection is made in tuya_switch_state
    return { l1: 1, l2: 1, l3: 1, l4: 1 }
  },
  exposes: [
    e
      .light_brightness()
      .withEndpoint('l1')
      .setAccess('state', ea.STATE_SET)
      .setAccess('brightness', ea.STATE_SET),
    e
      .light_brightness()
      .withEndpoint('l2')
      .setAccess('state', ea.STATE_SET)
      .setAccess('brightness', ea.STATE_SET),
    e
      .light_brightness()
      .withEndpoint('l3')
      .setAccess('state', ea.STATE_SET)
      .setAccess('brightness', ea.STATE_SET),
    e
      .light_brightness()
      .withEndpoint('l4')
      .setAccess('state', ea.STATE_SET)
      .setAccess('brightness', ea.STATE_SET),
    exposes.enum('action_p1', ea.STATE, ['click']),
    exposes.enum('action_p2', ea.STATE, ['click']),
    exposes.enum('action_p3', ea.STATE, ['click']),
    exposes.enum('action_p4', ea.STATE, ['click']),
  ],
}

module.exports = definition

### What did you expect to happen?

**I'm in urgent need of using it, and I hope there's a kind person who can help me modify this JS file. I'm extremely grateful for any assistance provided here!！！！！！！！**

### How to reproduce it (minimal and precise)

无

### Zigbee2MQTT version

1.32.0-2

### Adapter firmware version

20221226

### Adapter

CC2652P

closed 2023-08-12 by github-actions[bot]

ramalamadingdong, 2024-04-02:
Did you ever find a fix for this?

lccox, 2024-04-02:
> Did you ever find a fix for this?你找到解决办法了吗

I rewrote the  converter .js file,then the problem was solved.It is caused by the change of some parameters in tuya.js file