Koenkk/zigbee2mqtt #18211: Z2M don't use the right white color with new version

Henke19822, 2023-07-04:
### What happened?

After updating Z2M from 1.31.2-1 to version 1.32.0-2, Z2M don't use the right white color that is predifined in the scene that is being called on. The scene is set to use a warmer white light. It worked until 1.32.0-2.

Found some interesting in the log files though: Publish 'set' 'color_temp' to 'Bottenvåning, Z2M' failed: 'TypeError: Cannot read properties of undefined (reading 'hasOwnProperty')'

### What did you expect to happen?

I expected the the scene to get activated and the color of the white light should be warmer. 

### How to reproduce it (minimal and precise)

I only have a bunch of IKEA TRådfri lights, and they are affected. A friend of mine did not have this issue though, so it is hard to find a way to reproduce. 
I downgraded to 1.31.2-1 to test, and in that version everything works. But after upgrading to 1.32.0-2 again, i still have the same issue with whitest white.

### Zigbee2MQTT version

1.32.0-2

### Adapter firmware version

20220219

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

HA log: https://pastebin.com/1VDSsCUC
Z2M log: https://pastebin.com/1C3Z7KUN

ahd71, 2023-07-04:
https://github.com/Koenkk/zigbee2mqtt/issues/18168

Koenkk, 2023-07-04:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/18168 indeed