Koenkk/zigbee2mqtt #18212: Aqara SRTS-A01 - sensor_temp Error: Value is not a number, got string since upgrade to 1.32.0

chris94440, 2023-07-04:
### What happened?

Error when updating sensor temperature through endpoint sensor_temp => Error: Value is not a number, got string (25.41)
And temperature is not updated

### What did you expect to happen?

Temperature would be updated (local_temperature)

### Zigbee2MQTT version

1.30.0 commit 69638d79

### Adapter firmware version

20220219

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P

### Debug log

Debug 2023-07-04 17:20:03Received MQTT message on 'zigbee2mqtt/Z2M_XIAOMI_Vanne_Thermostatique_Chambre_Noe/set' with data '{"sensor_temp":"25.41"}'
Debug 2023-07-04 17:20:03Publishing 'set' 'sensor_temp' to 'Z2M_XIAOMI_Vanne_Thermostatique_Chambre_Noe'
Error 2023-07-04 17:20:03Publish 'set' 'sensor_temp' to 'Z2M_XIAOMI_Vanne_Thermostatique_Chambre_Noe' failed: 'Error: Value is not a number, got string (25.41)'
Debug 2023-07-04 17:20:03Error: Value is not a number, got string (25.41) at Object.assertNumber (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:502:42) at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:463:27) at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:248:52) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/opt/zigbee2mqtt/lib/eventBus.ts:109:22) at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.ts:140:27) at MqttClient.emit (node:events:513:28) at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1547:12) at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:535:12) at work (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:438:12)

chris94440, 2023-07-04:
Sorry error in z2m version 
Not
` Zigbee2MQTT version
1.30.0 commit https://github.com/Koenkk/zigbee2mqtt/commit/69638d7956b4991ffcbc5b61828d201fe912e8b9
`
but 
`Zigbee2MQTT version
1.32.0 commit https://github.com/Koenkk/zigbee2mqtt/commit/69638d7956b4991ffcbc5b61828d201fe912e8b9`

Koenkk, 2023-07-04:
Pushed a fix

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-07-04 by Koenkk

chris94440, 2023-07-04:
Thx a lot