Koenkk/zigbee2mqtt #18220: LQI readings are flawed with SLZB-06 coordinator

TC1666, 2023-07-04:
### What happened?

Setup a SLZB-06 coordinator on my test bench paired with a mains powered device... and varied the distance from 0 to 1 meter... and have never gotten a LQI reading above 200. I can do the same thing with the ConBeeII stick and deconz, and get a 255 LQI when the devices are near each other.

I then selected two of my devices that are physically on the edge of my Zigbee network. These are identical battery powered sensors located within one meter of each other. I left one on the ConBeeII/deconz network and moved the other over to the Z2M/SLZB-06.

The device on the ConBee/deconz has an LQI reading in the low 200s. When there has been problems in the past (interference, someone moved objects in front of the sensor, my Wifi channel clobbered it) I could see the LQI drop into the 100's which gave me a clue why it was not reporting in. By changing the environment I could see the LQI number improve and restore network stability.

Its twin device on the Z2M/SLZB-06 has an LQI of 0. And it has never budged above that. Yet it paired fine, reports in on a regular basis, etc. There are zero issues with it.

So what happens when I *do* have a problem with a device like it?? How do I troubleshoot it? How do I make changes to the environment and be able to objectively gauge the results regarding signal strength data?

I've read claims here that the ConBee/deconz LQI readings were never calculated correctly and are inflated. Fine, I'm willing to believe that.

But if that is the case, then I propose the Z2M/SLZB-06 LQI readings are *also* flawed and greatly deflated, with no bottom-end headroom, making it useless to troubleshoot with. If a device is already reporting 0 LQI then there's no way to tell if the signal has degraded.  With the ConbeeII/deconz I could tell when the signal degraded.

### What did you expect to happen?

In my opinion, based on real world testing, the LQI calculation should be re-examined, or there should be a mechanism to calibrate the expected LQI range to fit/fill the 8-bit Zigbee LQI range. If a coordinator and a device are next to each other then one should expect an LQI reading close to 255, which would then elevate devices on the lower-end.

### How to reproduce it (minimal and precise)

See above...

### Zigbee2MQTT version

1.31.2 commit: 21f5125 

### Adapter firmware version

20221226 

### Adapter

SMLight SLZB-06 

### Debug log

N/A

TEF2one, 2023-07-29:
I reported the same issue to the support team by email as I was having issues migrating from ConbeeII...
Their response was the following:
> Regarding LQI, this parameter is not clearly standardized in ZIGBEE, so different controllers will give different values of this parameter at the same power. This does not affect the actual output power.

So it doesn't look like they have any intention to change it...

TC1666, 2023-07-29:
After doing some additional testing, it's pretty clear to me that the SLZB-06 is an inferior transceiver (specifically receiver) when compared to the ConBeeII. Having that huge 5db gain antennae and 20dbm amplifier setting doesn't help at all since your battery-powered endpoints are basically whispering back to the controller. Like I previously stated, I've got endpoints on the perimeter of my network that the ConBeeII has zero problem communicating with, and it has no external antennae! 

My advice to people is don't waste your money on the SLZB-06, its only advantage is the multiple interfaces it allows. It's a lousy receiver.

TEF2one, 2023-07-29:
My goal was to avoid USB dongles to allow my server to fail over...
Any other recommendations for using ethernet?
The first one listed on Z2M is TubesZB but doesn't seem to be for sale in Europe...

TC1666, 2023-07-29:
My only recommendation, and I realize it may not be feasible given the supply chain issues, is create your own network transceiver. In other words, take a USB dongle and plug it into a RPi, and also put the Z2M stack on there (possibly also MQTT). That is currently what I'm doing with the SLZB-06 since I already had a RPi4 that was running Node-red and my UniFi network controller.

closed 2023-09-05 by github-actions[bot]

FreakErn, 2024-09-17:
I have a similar issue. I had a Conbee 2 which had 255 LQI on close devices (now my conbee 2 died). 

Now i have a sonoff CC2652P and i have the SLZB-06, both adapters have a really bad LQI (highest is a power plug, 60 cm away from the coordinator with a LQI of ~155 and i have 3-6 devices, out of 60, which have a LQI of 0, with most others between 60 and 80).

But i don't think this is a LQI Range mapping problem, because my battery driven devices, with the highest distance die in 2-4 weeks with a bad LQI compared to 6-9 months with a mediocre LQI of 120-160 i had with my conbee 2
(14-30 days compared to 180-260something days!). I switched the coordinators (with remapping of aaaaaaallllll devices) a couple of times, always with the same result.

So my Sonoff zigbee plus CC2652P and my SLZB-06 will be replaced by a conbee 2 asap.

wdonders, 2025-12-28:
I just migrated from a Sonoff zigbee dongle plus-E with the EFR32MG21 chipset to the SLZB-06U.
With the sonoff I had many devices hitting 240 signal strength and the worst onces about 120
Now the best one gets 167... The worst onces struggle to keep a connection at 18 to 35..
If this is the real signal strength, then I'm pretty disappointed in my 'upgrade' :-(