Koenkk/zigbee2mqtt #18221: 1.32 npm start fails at philips.tz.effect

JC38, 2023-07-04:
### What happened?

npm start and then an error occurs
D:\zigbee2mqtt-zigate\lib\extension\publish.ts:22
    philips.tz.effect, // Support Hue effects for groups
            ^
TypeError: Cannot read properties of undefined (reading 'tz')
    at Object.<anonymous> (D:\zigbee2mqtt-zigate\lib\extension\publish.ts:22:13)
    at Module._compile (node:internal/modules/cjs/loader:1255:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1309:10)
    at Module.load (node:internal/modules/cjs/loader:1113:32)
    at Function.Module._load (node:internal/modules/cjs/loader:960:12)
    at Module.require (node:internal/modules/cjs/loader:1137:19)
    at require (node:internal/modules/helpers:121:18)
    at Object.<anonymous> (D:\zigbee2mqtt-zigate\lib\controller.ts:14:1)
    at Module._compile (node:internal/modules/cjs/loader:1255:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1309:10)

### What did you expect to happen?

Being able to update from 1.31.2 to 1.32

### How to reproduce it (minimal and precise)

Deleted the folder node_modules and run npm ci before lauching npm start

### Zigbee2MQTT version

1.32

### Adapter firmware version

2103

### Adapter

zigate

Koenkk, 2023-07-05:
Did you do a `npm ci` after updating (in the z2m dir)?

JC38, 2023-07-05:
Hello
Yes, I did many tries.
First, npm ci from the folder --> failure
Deleted the node_modules folder and then a npm ci --> failure again
Reinstall the previous version, did a npm ci then updated z2m and re-run npm ci --> failure

Koenkk, 2023-07-06:
> First, npm ci from the folder --> failure

What is the error?

JC38, 2023-07-06:
Sorry maybe i didn't write enough explanation.
Stage 1 : npm ci  --> ok
Stage 2 : npm start then the error below happens
D:\zigbee2mqtt-zigate\lib\extension\publish.ts:22
philips.tz.effect, // Support Hue effects for groups
^
TypeError: Cannot read properties of undefined (reading 'tz')
at Object. (D:\zigbee2mqtt-zigate\lib\extension\publish.ts:22:13)
at Module._compile (node:internal/modules/cjs/loader:1255:14)
at Object.Module._extensions..js (node:internal/modules/cjs/loader:1309:10)
at Module.load (node:internal/modules/cjs/loader:1113:32)
at Function.Module._load (node:internal/modules/cjs/loader:960:12)
at Module.require (node:internal/modules/cjs/loader:1137:19)
at require (node:internal/modules/helpers:121:18)
at Object. (D:\zigbee2mqtt-zigate\lib\controller.ts:14:1)
at Module._compile (node:internal/modules/cjs/loader:1255:14)
at Object.Module._extensions..js (node:internal/modules/cjs/loader:1309:10)

Koenkk, 2023-07-06:
What is the output of `git status`?

JC38, 2023-07-06:
I am using Windows without git.
Just getting the archive from github, expanding the zip and copying the files into the "production" folder.

JC38, 2023-07-06:
I never did that.
I just read the explanation there [https://www.zigbee2mqtt.io/guide/installation/05_windows.html#configuring](url)
Did I miss something ? I never got any problem (except crashes but this is another story).

JC38, 2023-07-11:
Hello
As an updated version was published somedays ago, I try this one and still get the error message about philips.tz.effect (see screenshot below)
I am keeping the running version and the non-working in separate folders, so it's easy now for me to test something on the non-working if needed. Just tell me what you would like me to test.
[image]

Koenkk, 2023-07-13:
Did you already try running a `npm run build`?

Koenkk, 2023-07-15:
The guide assumes running your setup directly from the `git clone` directory, in that case `npm run build` will run automatically. When copying from the downloaded zip, `npm run build` will only run if the `dist` folder does not exist yet. Therefore in case you update z2m by copying over the files manually, `npm run build` is not executed and therefore you are not using the latest version of the code.

closed 2023-07-15 by Koenkk