Koenkk/zigbee2mqtt #18222: [New device support]: _TZE204_ue3rzddr Tuya Smart Knob Dimmer Switch, I try to add myself but lost direction...

Keele520, 2023-07-05:
### Link

https://www.alibaba.com/product-detail/Smart-Knob-Dimmer-Switch-WiFi-Tuya_1600574868597.html

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c138099d6e6eda","nwkAddr":21784,"manufId":4417,"manufName":"_TZE204_ue3rzddr","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1688527822816,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the supporting new device page but got stuck at sending data to zigbee, have found out this dimmer datapoints and function by tuya IoT platform.

datapoint 1 is switch, datatype is 1, ON/OFF function is OK.
datapoint 2 is the brightness, datatype is 2, when turn the dimmer knob from 0% to 100%. 
   - position "0%", data buffer is [0,0,0,0], valueConverter.raw get 0
   - position "100%", data buffer is [0,0,3,232], valueConverter.raw get 1000
so using valueConverter.divideBy10, can get the precise value of the brightness, so far read back is OK,
but when play the Brightness bar at the dashboard page, drag it to 100%, but next cycle it will change back to the bar position around 10% of the full bar.

here is the debug log
2023-07-05 11:55:44 Zigbee2MQTT:debug 
2023-07-05 11:55:44: Received MQTT message on 'zigbee2mqtt/dimmer/set' with data '{"brightness":254}'
2023-07-05 11:55:44 Zigbee2MQTT:debug 
2023-07-05 11:55:44: Publishing 'set' 'brightness' to 'dimmer'
2023-07-05 11:55:44 Zigbee2MQTT:info  
2023-07-05 11:55:44: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"backlight_switch":"OFF","brightness":254,"countdown":0,"light_type":181,"linkquality":138,"max_brightness":100,"min_brightness":0,"power_on_behavior":"previous","state":"ON"}'
2023-07-05 11:55:44 Zigbee2MQTT:debug 
2023-07-05 11:55:44: Received Zigbee message from 'dimmer', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,236],"type":"Buffer"},"datatype":2,"dp":2}],"seq":14336}' from endpoint 1 with groupID 0
2023-07-05 11:55:44 Zigbee2MQTT:info  
2023-07-05 11:55:44: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"backlight_switch":"OFF","brightness":23.6,"countdown":0,"light_type":181,"linkquality":138,"max_brightness":100,"min_brightness":0,"power_on_behavior":"previous","state":"ON"}'

or trigger by MQTT "zigbee2mqtt/dimmer/set" with payload {"brightness":100}
2023-07-05 12:06:47 Zigbee2MQTT:debug 
2023-07-05 12:06:47: Received MQTT message on 'zigbee2mqtt/dimmer/set' with data '{"brightness": 100}'
2023-07-05 12:06:47 Zigbee2MQTT:debug 
2023-07-05 12:06:47: Publishing 'set' 'brightness' to 'dimmer'
2023-07-05 12:06:49 Zigbee2MQTT:info  
2023-07-05 12:06:49: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"backlight_switch":"OFF","brightness":100,"countdown":0,"light_type":181,"linkquality":138,"max_brightness":100,"min_brightness":0,"power_on_behavior":"previous","state":"OFF"}'
2023-07-05 12:06:49 Zigbee2MQTT:debug 
2023-07-05 12:06:49: Received Zigbee message from 'dimmer', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":19200}' from endpoint 1 with groupID 0
2023-07-05 12:06:49 Zigbee2MQTT:info 
2023-07-05 12:06:49: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"backlight_switch":"OFF","brightness":100,"countdown":0,"light_type":181,"linkquality":138,"max_brightness":100,"min_brightness":0,"power_on_behavior":"previous","state":"ON"}'
2023-07-05 12:06:49 Zigbee2MQTT:debug 
2023-07-05 12:06:49: Received Zigbee message from 'dimmer', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,232],"type":"Buffer"},"datatype":2,"dp":2}],"seq":19456}' from endpoint 1 with groupID 0
2023-07-05 12:06:49 Zigbee2MQTT:info  
2023-07-05 12:06:49: MQTT publish: topic 'zigbee2mqtt/dimmer', payload '{"backlight_switch":"OFF","brightness":23.2,"countdown":0,"light_type":181,"linkquality":142,"max_brightness":100,"min_brightness":0,"power_on_behavior":"previous","state":"ON"}'

I'm doubt it is the datatype problem? 
From debug the data is 4*1 byte [0,0,0,232], and the last byte '232’ divide by 10 is 23.2.
When send out from MQTT to zigbee, value 100 will multiple 10 become 1000, and value 1000 to 4-byte binary is [0,0,3,232], but only last byte is sending so value become 232.

Try to solve this but lack skill, looking for help. Thanks a lot.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_ue3rzddr',
        },
    ],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'DIMMER knob with 4 state button',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.lightBrightnessWithMinMax(), 
        e.power_on_behavior(),
        tuya.exposes.countdown(), 
        tuya.exposes.lightType()
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
            [2, 'brightness', tuya.valueConverter.divideBy10],
            [3, 'min_brightness', tuya.valueConverter.divideBy10],
            [4, 'light_type', tuya.valueConverter.raw],
            [5, 'max_brightness', tuya.valueConverter.divideBy10],
            [6, 'countdown', tuya.valueConverter.countdown],
            [10, 'light_type_2', tuya.valueConverter.raw],
            [14, 'power_on_behavior', tuya.valueConverter.powerOnBehavior],
            [26, 'backlight_switch', tuya.valueConverter.onOff]
        ],
    },
};

module.exports = definition;
```

closed 2023-08-13 by github-actions[bot]

torinnguyen, 2023-11-05:
@Keele520 I'm interested to try adding support for this device, but I have not bought it yet. Did you try to use it with native Tuya first? I just want to know whether the % display can be updated via ZigBee in the first place?

Keele520, 2023-11-08:
@torinnguyen Yes, the % display can be updated via ZigBee. The device with the latest firmware already resolve my problem, they already change the scale range from 0-1000 to 0-255.

yabdali, 2024-02-01:
> @torinnguyen Yes, the % display can be updated via ZigBee. The device with the latest firmware already resolve my problem, they already change the scale range from 0-1000 to 0-255.

Hi, is the light you are controlling is a smart light/zigbee or attached to a zigbee LED driver? I bought something similar to this but my LED light is a just standard LED downlight with a normal driver and I haven't been able to make use of the dimmer except to turn on and off the light!

treetopped, 2024-11-14:
Is the issue resolved yet? I mean does above definition work?

I have one of these, tried to add external definition .js file (above), but still no go.

If this should work please point me to some walk-through, I am probably doing sth. wrong.

treetopped, 2025-05-26:
the file _TZE204_ue3rzddr.js , located in zigbee2mqtt/external_converters
I also have newer model, _TZE284_gokck1v7.js, it is also working with the same converter

there is no control of backlight. I somehow managed to turn off blue ring light entirely and dim the backlight on the older model, but I don't remember how it happened. it is bright and with blue ring on the newer model

just now I am using z2m version 2.3.0

working definition:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
//const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_ue3rzddr',
        },
    ],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'DIMMER knob with 4 state button',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.lightBrightnessWithMinMax(), 
        e.power_on_behavior(),
        tuya.exposes.countdown(), 
        tuya.exposes.lightType()
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
            [2, 'brightness', tuya.valueConverter.scale0_254to0_1000],
            [3, 'min_brightness', tuya.valueConverter.divideBy10],
            [4, 'light_type', tuya.valueConverter.raw],
            [5, 'max_brightness', tuya.valueConverter.divideBy10],
            [6, 'countdown', tuya.valueConverter.countdown],
            [10, 'light_type_2', tuya.valueConverter.raw],
            [14, 'power_on_behavior', tuya.valueConverter.powerOnBehavior],
            [26, 'backlight_switch', tuya.valueConverter.onOff]
        ],
    },
};

module.exports = definition;

assrianon, 2025-05-27:
[image]

assrianon, 2025-05-27:
[image]

treetopped, 2025-05-28:
[image]

treetopped, 2025-05-28:
BTW, this dimmer works great, I use it to control kitchen hood with HA automations.
Just as a knob, nothing is connected to physical output. You can see the actual setpoint.
I am waiting for official support in Z2M.

[image]

assrianon, 2025-06-12:
> [image]
There really was an extra gap. I removed and everything worked! Thank you!