Koenkk/zigbee2mqtt #18223: HZC Zigbee cord dimmer used to work, does not with 1.32.0

albatorsk, 2023-07-05:
### What happened?

In Home Assistant, I updated the zigbee2mqtt add-on from version 1.31.2-1 to version 1.32.0-2 and my HZC Zigbee cord dimmer that has worked just fine for the last few months stopped being recognized and is now unsupported and can not be used any more.

I believe it might be related to this issue that was resolved with the latest version of zigbee2mqtt: https://github.com/Koenkk/zigbee-herdsman-converters/issues/5859

After downgrading to version 1.31.2-1 the cord dimmer is working as expected again and can be controlled from zigbee2mqtt.

### What did you expect to happen?

After updating to 1.32.0-2, the cord dimmer should continue to work as it has with the version of zigbee2mqtt before this.

### How to reproduce it (minimal and precise)

1. Use zigbee2mqtt 1.31.2-1 Home Assistant addon
2. Pair the HZC Zigbee cord dimmer
3. Verify that the dimmer works as expected
4. Update zigbee2mqtt Home Assistant add-on to version 1.32.0-2
5. The cord dimmer is no longer recognized as a supported device

### Zigbee2MQTT version

1.32.0

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Debug log

[zigbee2mqtt.log](https://github.com/Koenkk/zigbee2mqtt/files/11957838/zigbee2mqtt.log)

ronniebach, 2023-07-06:
I also experienced the exact same behaviour with the D160-ZG dimmer.

They appeared in Zigbee2MQTT as a generic dimmer - but no connection / exposed functions.

Also reverting to 1.31.2 fixed the problem here.
So it does confirm that it has something to do with the identification of the dimmer I think.

KDanielK, 2023-07-06:
FYI https://github.com/Koenkk/zigbee2mqtt/issues/18178

albatorsk, 2023-07-07:
> FYI #18178

Thank you. This is definitely a duplicate of the issue you linked to. I should have looked more thoroughly before logging mine.

closed 2023-07-07 by Koenkk