Koenkk/zigbee2mqtt #18228: Lidl HG06336 battery=100 and battery_low=true

Bastian007, 2023-07-05:
### What happened?

After Battery change the device Lidl HG06336
shows  battery 100% but battery_low icon.
[image]

{
    **"battery": 100,
    "battery_low": true,**
    "contact": true,
    "linkquality": 43,
    "tamper": false,
    "elapsed": 3631179
}

### What did you expect to happen?

"battery_low": false after battery change.

### Zigbee2MQTT version

1.32.0-2

### Adapter firmware version

Z-Stack_3.x.0 coordinator 20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

Bastian007, 2023-08-05:
issue still valid

Bastian007, 2023-09-06:
issue still exist

comprido, 2023-09-17:
still reproduced

Bastian007, 2023-10-18:
issue still exist

Bastian007, 2024-04-18:
issue still exist with 1.36.1

Bastian007, 2024-06-22:
issue still exist with 1.38.0-1

comprido, 2024-06-22:
can this issue be fixed.
It would be great to know when the battery is getting low

Koenkk, 2024-06-24:
Is `battery_low` always `true`? I think the device doesn't report it properly, so we should ignore it.

Bastian007, 2024-06-25:
No, it is not always true
[image]

Koenkk, 2024-06-25:
Removed it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-06-25 by Koenkk