Koenkk/zigbee2mqtt #18229: XHK1-TC Not arming

tylermenezes, 2023-07-06:
### What happened?

When trying to arm the UHK1-UC as per the documentation, the keypad does not appear to respond in any way (to either `entry_delay`, `exit_delay`, or `arm_day_zones`) and I eventually get an error that `request.frame.Payload.filter is not a function` in zigbee2mqtt logs.

The keypad _does_ respond correctly to responses to keypad-initiated events (i.e. with `arm_mode.transaction`).

Possibly related to #11113 or #8630

### What did you expect to happen?

The keypad should beep when receiving `entry_delay` or `exit_delay`.

### How to reproduce it (minimal and precise)

Set `zigbee2mqtt/[name]/set`with `{"arm_mode": {"mode": "exit_delay"}}` and/or `{"arm_mode": {"mode": "arm_day_zones"}}`

### Zigbee2MQTT version

1.32.0 commit: 69638d7

### Adapter firmware version

20220219

### Adapter

athom.tech

### Debug log

```
debug 2023-07-05 20:14:29Received MQTT message on 'zigbee2mqtt/0x000d6f0012586edc/set' with data '{"arm_mode": {"mode": "exit_delay"}}'
debug 2023-07-05 20:14:29Publishing 'set' 'arm_mode' to '0x000d6f0012586edc'
debug 2023-07-05 20:14:49Received MQTT message on 'zigbee2mqtt/0x000d6f0012586edc/set' with data '{"arm_mode": {"mode": "arm_day_zones"}}'
debug 2023-07-05 20:14:49Publishing 'set' 'arm_mode' to '0x000d6f0012586edc'
debug 2023-07-05 20:14:54Device '0x000d6f0012586edc' announced itself
info 2023-07-05 20:14:54MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0012586edc","ieee_address":"0x000d6f0012586edc"},"type":"device_announce"}'
info 2023-07-05 20:14:54MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x000d6f0012586edc"},"type":"device_announced"}'
debug 2023-07-05 20:14:54Received Zigbee message from '0x000d6f0012586edc', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
info 2023-07-05 20:14:54MQTT publish: topic 'zigbee2mqtt/0x000d6f0012586edc', payload '{"action":null,"action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"battery_low":false,"contact":true,"linkquality":32,"occupancy":true,"presence":null,"tamper":false,"temperature":28,"voltage":6400}'
error 2023-07-05 20:14:54Error: CommandResponse 0x000d6f0012586edc/1 ssIasAce.getPanelStatusRsp({"panelstatus":1,"secondsremain":0,"audiblenotif":0,"alarmstatus":0}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (request.frame.Payload.filter is not a function)
debug 2023-07-05 20:14:54TypeError: CommandResponse 0x000d6f0012586edc/1 ssIasAce.getPanelStatusRsp({"panelstatus":1,"secondsremain":0,"audiblenotif":0,"alarmstatus":0}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (request.frame.Payload.filter is not a function) at Endpoint.filterRequests (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:346:67) at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:383:18) at Endpoint.commandResponse (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:865:24) at Object.onEvent (/app/node_modules/zigbee-herdsman-converters/src/devices/technicolor.ts:42:45) at OnEvent.callOnEvent (/app/lib/extension/onEvent.ts:47:37) at EventEmitter.<anonymous> (/app/lib/extension/onEvent.ts:13:60) at EventEmitter.emit (node:events:525:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22) at Controller.<anonymous> (/app/lib/zigbee.ts:106:27) at Controller.emit (node:events:513:28)
```

closed 2023-08-13 by github-actions[bot]

tylermenezes, 2024-02-19:
I think some of them are just unreliable. I bought a second one which works fine.

thecobra666, 2024-02-21:
> I think some of them are just unreliable. I bought a second one which works fine.

Probably a firmware issue then? You wouldn't be able to see if they have different firmware?

thecobra666, 2024-02-27:
Updating a little more. So I've added the keypad to ZHA and tried again. With ZHA I've found how to send a cluster command 

Entry_delay command:
service: zha.issue_zigbee_cluster_command
data:
  ieee: xxxxxxxxxxxxxxxxxxxxxxx
  cluster_id: 1281
  command: 4
  command_type: client
  cluster_type: out
  args: [5, 10, 1, 0]
  endpoint_id: 1

And then the keypad does beep when it's armed for away mode. So the device is okay, the bug is in zigbee2mqtt. 

Probably the same command would work when it's arming for exit_delay

Exit_delay command:
service: zha.issue_zigbee_cluster_command
data:
  ieee: xxxxxxxxxxxxxxxxxxxxxxx
  cluster_id: 1281
  command: 4
  command_type: client
  cluster_type: out
  args: [4, 10, 1, 0]
  endpoint_id: 1