Koenkk/zigbee2mqtt #18237: [New device support]: for Tuya M100S with 24G radar

DosPH, 2023-07-07:
### Link

https://shopee.ph/ZigBee-Wifi-24G-MmWave-Radar-Human-Presence-Motion-Sensor-With-Luminosity-Distance-Detection-5-110-220V-Tuya-Smart-Life-Home-DIY-i.458442746.21881172707?sp_atk=c89bca14-cb63-4ccb-acdf-36f123ccff3e&xptdk=c89bca14-cb63-4ccb-acdf-36f123ccff3e

### Database entry

{"id":65,"type":"Router","ieeeAddr":"0xa4c13855222f4904","nwkAddr":2607,"manufId":4417,"manufName":"_TZE204_ijxvkhd0","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE204_ijxvkhd0","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1688723423810,"defaultSendRequestWhen":"immediate"}

### Comments

anyone have tried to have this new tuya m100s with 24G radar working? this is the same square wallmount type with a Type C connector as power source.

vitaleg, 2023-07-12:
I tried to connect using quirks in ZHA based on previous device models, and also with zigbee2mqtt using a converter from previous models. Nothing worked.
Looks like it's a brand new sensor, works on a completely different principle.

So, +1, add support, please.

DosPH, 2023-07-13:
yes this new one. Tried ZHA and latest converter also not working.

rpearl, 2023-07-23:
Try this https://gist.github.com/rpearl/553184b310b8aa50d7e37f6a000fac51 (for TZE204_qasjif9e at least)

gevorgmansuryan, 2023-07-24:
> Try this https://gist.github.com/rpearl/553184b310b8aa50d7e37f6a000fac51 (for TZE204_qasjif9e at least)

this one is not working correctly. _TZE204_ijxvkhd0 supports new events, move and presence

rpearl, 2023-07-24:
So it goes. This was for TZE204_qasjif9e, and I don't have _TZE204_ijxvkhd0.

I was able to identify these datapoints by observing the `Datapoint 'X' with value 'Y' not defined for '_TZE204_qasjif9e'` log messages in zigbee2mqtt. I recommend trying the same for your device. This is not as easy as using the tuya gateway stuff but it can work.

VladKorr, 2023-07-27:
I have _TZE204_ijxvkhd0 and tried to get datapoints by means of tuya gateway.
I got the results for DP:
DP; DP Name
1; Switch 1
2; Switch 2
7; Timer 1
8; Timer 2
101; Work Mode
102; Bright Value
103; Temp Value
104; Illuminance Value
105; 感应状态 (? inductive state)
106; 运动灵敏度 (? motion sensitivity)
107; Longest distance
109; target_distence
110; Delay time
111; 静止灵敏度 (?static sensitivity)
112; 有无人状态 (?unoccupied state)
Some DPs are in Chinese - (Google translation in brackets)

VladKorr, 2023-07-27:
_TZE204_ijxvkhd0 is 24GHz and and it's different from a similar 5 GHz sensor.
I tried to make a converter for Z2M, but it doesn't work quite right with the sensor settings (Delay time, static sensitivity) and there is no way to set the minimum sensor response distance.
The current code of my converter is here:
https://github.com/VladKorr/ZY-M100-24G/blob/main/ZY-M100-24G.js

vincentBesseau, 2023-07-28:
Hello @rpearl , how can I add your file for  the  "_TZE204_qasjif9e" device on my zigbee2mqtt ?
Do you have any process?
Thx in advance !

rpearl, 2023-07-28:
https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters

Carabin79, 2023-08-03:
Hello, please help me add this new device _TZE204_ijxvkhd0 in Z2q , it's a new device that works in 24g.

Igor-Andrusevic, 2023-08-04:
Hello!! Add this new device _TZE204_ijxvkhd0 in zigbee2mqtt

VladKorr, 2023-08-04:
> Hello, please help me add this new device _TZE204_ijxvkhd0 in Z2q , it's a new device that works in 24g.

Hello, try to add external converters to your z2m from link:
https://github.com/VladKorr/ZY-M100-24G/blob/main/ZY-M100-24G.js
How to do it you can see at:
https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters

Igor-Andrusevic, 2023-08-04:
Благодарю. Буду проверять, данный момент работает.

> Hello, try to add external converters to your z2m from link: https://github.com/VladKorr/ZY-M100-24G/blob/main/ZY-M100-24G.js How to do it you can see at: [...]

Carabin79, 2023-08-04:
Thank you it works

Koenkk, 2023-08-05:
I've added the device for out-of-the-box support (from 1 September release). Thanks @VladKorr

BeFygo, 2023-08-05:
> I've added the device for out-of-the-box support (from 1 September release). Thanks @VladKorr
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Hi, 
Great work guys.

I noticed with the latest-dev docker that the Tuya M100S gets recognized out-of-the-box but throws (no converter available for '') for all the adjustable parameters except "fading_time".

Koenkk, 2023-08-06:
@BeFygo fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

leroyloren, 2023-08-06:
Please, does this device flood the network like the previous model (ZY-M100-S_1)?

MTokarev, 2023-08-06:
> Try this https://gist.github.com/rpearl/553184b310b8aa50d7e37f6a000fac51 (for TZE204_qasjif9e at least)

I am a new user here and missing almost everything ;)
I have tried to add a custom converter and it works perfect.
The only thing here is illuminance. Looks like the value is coming divided by 10. If real illuminance is 40 then it shows 4.

Again, since I don't know how it works inside (I believe it should be fixed somewhere down in the code) I made a nasty workaround (pay attention to lookup_dictionary and the basic converter):

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const maxLuxAvailableDividedBy10 = 100
const lookup_dictionary = {};
const range = [...Array(maxLuxAvailableDividedBy10).keys()];
range.forEach(key => lookup_dictionary[key * 10] = key);

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_qasjif9e']),
    model: 'ZY-M100-S',
    vendor: 'TuYa',
    description: 'Mini Human Breathe Sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.illuminance_lux(), e.presence(),
        e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
        e.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1)
            .withDescription('sensitivity of the radar'),
        e.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Minimum range').withUnit('m'),
        e.numeric('maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Maximum range').withUnit('m'),
        e.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1)
            .withDescription('Detection delay').withUnit('s'),
        e.numeric('fading_time', ea.STATE_SET).withValueMin(0.5).withValueMax(1500).withValueStep(1)
            .withDescription('Fading time').withUnit('s'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence', tuya.valueConverter.trueFalse1],
            [9, 'target_distance', tuya.valueConverter.divideBy100],
           
	    // This is a nasty fix to change illuminance report. We must multiply the value by 10
            // to get accurate value. The problem here is that the step = 10. So you won't see the value like 11 or 15
            // Only 10, 20, 30...
            [104, 'illuminance_lux', tuya.valueConverterBasic.lookup(lookup_dictionary)],
            
	    [2, 'radar_sensitivity', tuya.valueConverter.raw],
            [4, 'maximum_range', tuya.valueConverter.divideBy100],
            [3, 'minimum_range', tuya.valueConverter.divideBy100],
            [102, 'fading_time', tuya.valueConverter.divideBy10],
            [101, 'detection_delay', tuya.valueConverter.divideBy10],
        ],
    },
};

module.exports = definition;
```

BeFygo, 2023-08-07:
> @BeFygo fixed!
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Confirmed, great work!

BeFygo, 2023-08-07:
@VladKorr great work!

I noticed when trying to change the values for  "move_sensitivity" & "presence_sensitivity" that over time the changes get overwritten to standard values.

device: _TZE204_ijxvkhd0

BeFygo, 2023-08-07:
> Please, does this device flood the network like the previous model (ZY-M100-S_1)?

In my opinion it does

VladKorr, 2023-08-07:
@BeFygo, unfortunately, yes. 
It happens both with an external converter and out-of-the-box (dev).
If you look at the logs, you can see that when we change values, zigbee2mqtt sends the values and they are written to the device. 
A few minutes pass and they go back to the previous values. And sometimes it is possible to change them and they become "standard".
I can't find a pattern or reason why this happens. 
I have an assumption that there are hidden DPs, where some value is stored and old values are returned by it. 
There is part of log where I change move_sensitivity from 4 to 7  (time 2023-08-07 20:22:07) & presence_sensitivity from 6 to 9 (time 2023-08-07 20:22:14) 
There are records "Zigbee2MQTT:debug 2023-08-07 20:22:07: Received MQTT message on 'zigbee2mqtt/0xa4c138a050a55304/set' with data '{"motion_sensitivity":7}'
Zigbee2MQTT:debug 2023-08-07 20:22:07: Publishing 'set' 'motion_sensitivity' to '0xa4c138a050a55304'"
and "Zigbee2MQTT:debug 2023-08-07 20:22:14: Received MQTT message on 'zigbee2mqtt/0xa4c138a050a55304/set' with data '{"presence_sensitivity":9}'
Zigbee2MQTT:debug 2023-08-07 20:22:14: Publishing 'set' 'presence_sensitivity' to '0xa4c138a050a55304'"

but at 20:23:32 we get message "Zigbee2MQTT:debug 2023-08-07 20:23:32: Received Zigbee message from '0xa4c138a050a55304', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,40],"type":"Buffer"},"datatype":2,"dp":106}],"seq":49408}' from endpoint 1 with groupID 0"
and 'motion_sensitivity' was returned to old value 4.
The same happened at 20:23:32 with the presence_sensitivity.

[Part Log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12283912/Part.Log.txt)

mindtripper, 2023-08-17:
Try setting move_sensitivity and presence_sensitivity between 10-90. Unsure if it requires increments of 10.
This sticks when using the device via Hubitat Zigbee GW.
I've noticed this through trial and error.

mindtripper, 2023-08-19:
> I have tried setting move_sensitivity and presence_sensitivity between 10 and 90 in both 10 and 1 change increments. This does not help. After a while the values revert to the previous values. The values to which these settings revert depend on the sensitivity level that was set before in Smart Life. There are five sensitivity levels that can be set for this sensor in Smart Life. Each sensitivity level has its own move_sensitivity and presence_sensitivity values. L1 - 7 move_sensitivity and 9 presence_sensitivity L2 - 6 and 7 respectively L3 - 4 and 6 L4 - 2 and 4 L5 - 2 and 3 But in which dp the value of the sensitivity level is written and then retrieved I can't determine yet чт, 17 авг. 2023 г. в 20:47, mindtripper ***@***.***>:

You are correct in regard to setting 10-90. It does not work.
But on Hubitat GW the device reports back that the setting is *10. Meaning if I set it to 3 it reports back after a while that it is 30. I have not found anything in the driver code that would do this multiplication when the device reports back.
The setting does however stick. At least for hours and even when I refresh the device. But I've paused further development on Hubitat because the device is constantly reporting presence when there is none. No matter how I configure move_sensitivity and presence_sensitivity.

Getting a couple of SONOFF sticks that I will deploy to continue the work using Z2M and your driver instead.

As to your findings. It seems based on your description of L1-5 that "L" is only a template in the app that will set move_sensitivity and presence_sensitivity only. But I could definitely be wrong. Dont have Tuya GW, so I can't really check :/

Your effort is much appreciated! And you seem to be the only one working on getting this device to work.

Please continue sharing your findings and I will test using my SONOFF's.

mindtripper, 2023-08-21:
With the following external converter (based on your code), I can adjust move_sensitivity and presence_sensitivity between 1-9. And it sticks. At least for quite a while (1 hour and counting). The number reported back will be multiplied 10.

Continuing investigations, as it actually does not make a whole lot of sense atm.

```

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: [
        'switch_1', 'switch_2', 'countdown_1', 'countdown_2', 'work_mode', 'bright_value', 'temp_value', 'illuminance_value', 
          'presence_state', 'move_sensitivity', 'motion_sensitivity', 'detection_distance_max', 'detection_distance_min', 'far_detection', 'target_dis_closest', 'fading_time', 'presence_sensitivity', 'man_state' 
    ],
}

const definition = {
    fingerprint: [ { modelID: 'TS0601', manufacturerName: '_TZE204_ijxvkhd0'}],
    model: 'ZY-M100-24G',
    vendor: 'TuYa',
    description: 'Micro Motion Sensor v1.2 ',
    configure: tuya.configureMagicPacket,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tzDatapoints],
	exposes: [
		e.illuminance_lux(), e.presence(),
		e.enum('presence_state', ea.STATE, ['none', 'present', 'moving']).withDescription('Presence state'),
		e.numeric('motion_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(100).withValueStep(1).withDescription('Motion sensitivity'),
		e.numeric('detection_distance_max', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withDescription('Max detection distance').withUnit('m'),
		e.numeric('detection_distance_min', ea.STATE).withDescription('Min detection distance').withUnit('m'),
		e.numeric('fading_time', ea.STATE_SET).withValueMin(1).withValueMax(15).withValueStep(1).withDescription('Delay time').withUnit('s'),
		e.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(100).withValueStep(1).withDescription('Presence sensitivity'),
	],
	meta: {
		tuyaDatapoints: [
			[104, 'illuminance_lux', tuya.valueConverter.raw],
			// [105, 'presence_state', tuya.valueConverterBasic.lookup({'none': tuya.enum(0), 'present': tuya.enum(1), 'moving': tuya.enum(2)})],
			[105, 'presence_state', tuya.valueConverterBasic.lookup({'None': tuya.enum(0), 'Present': tuya.enum(1), 'Moving': tuya.enum(2)})],
			// [106, 'motion_sensitivity', tuya.valueConverter.divideBy10],
			[106, 'motion_sensitivity', tuya.valueConverter.raw],
			[107, 'detection_distance_max', tuya.valueConverter.divideBy100],
			[109, 'detection_distance_min', tuya.valueConverter.divideBy100],
			[110, 'fading_time', tuya.valueConverter.raw],
			[111, 'presence_sensitivity', tuya.valueConverter.raw],
			// [111, 'presence_sensitivity', tuya.valueConverter.divideBy10],
			// [112, 'presence',  tuya.valueConverterBasic.lookup({'nobody': tuya.enum(0), 'exist': tuya.enum(1) })],
			[112, 'presence', tuya.valueConverter.trueFalse1],
		],
	},
};

module.exports = definition;

```

Elkiff71, 2023-08-25:
Hello, I don't have much experience but I observe the following:

The presence detector (Mod.ZY-M100-S_1) is publishing messages every 2 seconds in the broker, non-stop and floods the network with these messages,
In principle the messages are correct, I mean, they do not present errors. The records are always the same.

These are my logs:

2023-08-25 11:40:59MQTT publish: topic 'zigbee2mqtt/DecBan', payload '{"detection_delay":0.5,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"DecBan","hardwareVersion":1,"ieeeAddr":"0xa4c13899b0f3ed7e","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":63938,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":2,"illuminance_lux":9,"last_seen":"2023-08-25T09:40:59.365Z","linkquality":91,"maximum_range":5.55,"minimum_range":0.3,"presence":false,"radar_sensitivity":7,"target_distance":0}'
Info 2023-08-25 11:41:00MQTT publish: topic 'zigbee2mqtt/DecBan', payload '{"detection_delay":0.5,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"DecBan","hardwareVersion":1,"ieeeAddr":"0xa4c13899b0f3ed7e","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":63938,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":2,"illuminance_lux":9,"last_seen":"2023-08-25T09:41:00.962Z","linkquality":102,"maximum_range":5.55,"minimum_range":0.3,"presence":false,"radar_sensitivity":7,"target_distance":0}'
Info 2023-08-25 11:41:02MQTT publish: topic 'zigbee2mqtt/DecBan', payload '{"detection_delay":0.5,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"DecBan","hardwareVersion":1,"ieeeAddr":"0xa4c13899b0f3ed7e","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":63938,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":2,"illuminance_lux":9,"last_seen":"2023-08-25T09:41:02.559Z","linkquality":105,"maximum_range":5.55,"minimum_range":0.3,"presence":false,"radar_sensitivity":7,"target_distance":0}'
Info 2023-08-25 11:41:04MQTT publish: topic 'zigbee2mqtt/DecBan', payload '{"detection_delay":0.5,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"DecBan","hardwareVersion":1,"ieeeAddr":"0xa4c13899b0f3ed7e","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":63938,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":2,"illuminance_lux":9,"last_seen":"2023-08-25T09:41:04.172Z","linkquality":102,"maximum_range":5.55,"minimum_range":0.3,"presence":false,"radar_sensitivity":7,"target_distance":0}'
Info 2023-08-25 11:41:05MQTT publish: topic 'zigbee2mqtt/DecBan', payload '{"detection_delay":0.5,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"DecBan","hardwareVersion":1,"ieeeAddr":"0xa4c13899b0f3ed7e","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":63938,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":2,"illuminance_lux":9,"last_seen":"2023-08-25T09:41:05.756Z","linkquality":102,"maximum_range":5.55,"minimum_range":0.3,"presence":false,"radar_sensitivity":7,"target_distance":0}'
Info 2023-08-25 11:41:07MQTT publish: topic 'zigbee2mqtt/DecBan', payload '{"detection_delay":0.5,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"DecBan","hardwareVersion":1,"ieeeAddr":"0xa4c13899b0f3ed7e","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":63938,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":2,"illuminance_lux":9,"last_seen":"2023-08-25T09:41:07.352Z","linkquality":94,"maximum_range":5.55,"minimum_range":0.3,"presence":false,"radar_sensitivity":7,"target_distance":0}'
Info 2023-08-25 11:41:08MQTT publish: topic 'zigbee2mqtt/DecBan', payload '{"detection_delay":0.5,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"DecBan","hardwareVersion":1,"ieeeAddr":"0xa4c13899b0f3ed7e","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":63938,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":2,"illuminance_lux":9,"last_seen":"2023-08-25T09:41:08.950Z","linkquality":83,"maximum_range":5.55,"minimum_range":0.3,"presence":false,"radar_sensitivity":7,"target_distance":0}'
Info 2023-08-25 11:41:10MQTT publish: topic 'zigbee2mqtt/DecBan', payload '{"detection_delay":0.5,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"DecBan","hardwareVersion":1,"ieeeAddr":"0xa4c13899b0f3ed7e","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":63938,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":2,"illuminance_lux":9,"last_seen":"2023-08-25T09:41:10.549Z","linkquality":91,"maximum_range":5.55,"minimum_range":0.3,"presence":false,"radar_sensitivity":7,"target_distance":0}'
Info 2023-08-25 11:41:12MQTT publish: topic 'zigbee2mqtt/DecBan', payload '{"detection_delay":0.5,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"DecBan","hardwareVersion":1,"ieeeAddr":"0xa4c13899b0f3ed7e","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":63938,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":2,"illuminance_lux":9,"last_seen":"2023-08-25T09:41:12.145Z","linkquality":94,"maximum_range":5.55,"minimum_range":0.3,"presence":false,"radar_sensitivity":7,"target_distance":0}'

THX

nickgoh, 2023-09-14:
> @VladKorr great work!
> 
> I noticed when trying to change the values for "move_sensitivity" & "presence_sensitivity" that over time the changes get overwritten to standard values.
> 
> device: _TZE204_ijxvkhd0

Same. Cant seems to change ,it will overwrite to standard value for

Motion sensitivity
Presence sensitivity

[Presence] is always False, [Presence state] is always moving depite restaring the device

BeFygo, 2023-09-14:
> With the following external converter (based on your code), I can adjust move_sensitivity and presence_sensitivity between 1-9. And it sticks. At least for quite a while (1 hour and counting). The number reported back will be multiplied 10.
> 
> Even tried to "Reconfigure" the device and forcefully remove it and adding it back. move_sensitivity and presence_sensitivity between 1-9 still sticks.
> 
> Continuing investigations, as it actually does not make a whole lot of sense atm.
> 
> ```
> 
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const tzDatapoints = {
>     ...tuya.tz.datapoints,
>     key: [
>         'switch_1', 'switch_2', 'countdown_1', 'countdown_2', 'work_mode', 'bright_value', 'temp_value', 'illuminance_value', 
>           'presence_state', 'move_sensitivity', 'motion_sensitivity', 'detection_distance_max', 'detection_distance_min', 'far_detection', 'target_dis_closest', 'fading_time', 'presence_sensitivity', 'man_state' 
>     ],
> }
> 
> const definition = {
>     fingerprint: [ { modelID: 'TS0601', manufacturerName: '_TZE204_ijxvkhd0'}],
>     model: 'ZY-M100-24G',
>     vendor: 'TuYa',
>     description: 'Micro Motion Sensor v1.2 ',
>     configure: tuya.configureMagicPacket,
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tzDatapoints],
> 	exposes: [
> 		e.illuminance_lux(), e.presence(),
> 		e.enum('presence_state', ea.STATE, ['none', 'present', 'moving']).withDescription('Presence state'),
> 		e.numeric('motion_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(100).withValueStep(1).withDescription('Motion sensitivity'),
> 		e.numeric('detection_distance_max', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withDescription('Max detection distance').withUnit('m'),
> 		e.numeric('detection_distance_min', ea.STATE).withDescription('Min detection distance').withUnit('m'),
> 		e.numeric('fading_time', ea.STATE_SET).withValueMin(1).withValueMax(15).withValueStep(1).withDescription('Delay time').withUnit('s'),
> 		e.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(100).withValueStep(1).withDescription('Presence sensitivity'),
> 	],
> 	meta: {
> 		tuyaDatapoints: [
> 			[104, 'illuminance_lux', tuya.valueConverter.raw],
> 			// [105, 'presence_state', tuya.valueConverterBasic.lookup({'none': tuya.enum(0), 'present': tuya.enum(1), 'moving': tuya.enum(2)})],
> 			[105, 'presence_state', tuya.valueConverterBasic.lookup({'None': tuya.enum(0), 'Present': tuya.enum(1), 'Moving': tuya.enum(2)})],
> 			// [106, 'motion_sensitivity', tuya.valueConverter.divideBy10],
> 			[106, 'motion_sensitivity', tuya.valueConverter.raw],
> 			[107, 'detection_distance_max', tuya.valueConverter.divideBy100],
> 			[109, 'detection_distance_min', tuya.valueConverter.divideBy100],
> 			[110, 'fading_time', tuya.valueConverter.raw],
> 			[111, 'presence_sensitivity', tuya.valueConverter.raw],
> 			// [111, 'presence_sensitivity', tuya.valueConverter.divideBy10],
> 			// [112, 'presence',  tuya.valueConverterBasic.lookup({'nobody': tuya.enum(0), 'exist': tuya.enum(1) })],
> 			[112, 'presence', tuya.valueConverter.trueFalse1],
> 		],
> 	},
> };
> 
> module.exports = definition;
> ```

I can confirm this.
Using the code and description from @mindtripper the settings stick for days now.

Igor-Andrusevic, 2023-09-20:
Hello. 
1. Presence work incorrectly, if body not move, presence state nobody, but body inside room. 
2. illuminance_lux very low indicator, my room max 30 lux, another's sensors show much more.

BeFygo, 2023-09-20:
> Hello.
> 
> 1. Presence work incorrectly, if body not move, presence state nobody, but body inside room.
> 2. illuminance_lux very low indicator, my room max 30 lux, another's sensors show much more.

1. Play a little with the positioning, the "Motion sensitivity" and "Presence sensitivity" settings to improve the detection.
2. Shining a light directly at the little holes gives a higher lux value, is presume that one of the holes is too little too let all the light through.

Igor-Andrusevic, 2023-09-20:
> 1. Play a little with the positioning, the "Motion sensitivity" and "Presence sensitivity" settings to improve the detection.
> 2. Shining a light directly at the little holes gives a higher lux [...]

I try but its adding it back (default number ) move_sensitivity and presence_sensitivity

BeFygo, 2023-09-20:
> I try but its adding it back (default number ) move_sensitivity and presence_sensitivity

Try the code and instructions from [mindtripper] in this issue, it works for me.

PetzJohannes, 2023-09-24:
@thx1e I can fully confirm the behavior you described in your last comment. Yesterday I setup the presence sensor and worked around the issues. with the fix of @mindtripper everything works fine.

I think the issue should be reopened.

Dillton, 2023-10-03:
@PetzJohannes I can also confirm that everything works only with code provided by @mindtripper.
@mindtripper First thank you for the code and second were you able to make some additional changes?
@Koenkk Can you reopen this issue as these devices still not working 100% out of the box?

mindtripper, 2023-10-03:
> @PetzJohannes I can also confirm that everything works only with code provided by @mindtripper. @mindtripper First thank you for the code and second were you able to make some additional changes? @Koenkk Can you reopen this issue as these devices still not working 100% out of the box?

I have parked the devices in my growing IoT drawer for now. But that is not due to the driver.
I'm just not happy with the device functionality. Specifically setting detection distance. Which is not fine grained enough for my use cases. Therefore, I have not spent any more time on improving the driver. As this is a device limitation.

Koenkk, 2023-10-03:
@mindtripper would you mind making a pr?

EvertonCa, 2023-10-03:
I tried the converter provided by @mindtripper, but the device is not working properly yet.

- it still won't stick any settings that I change.
- presence is now showing True all the time (I left the room entirely it did not changed to False)

I have the older 5ghz version and it was working great. I ended up buying 3 of this 24ghz version and now I'm stuck.. Does anyone had any luck with another converter that solved all the problems?

patrex0304, 2023-10-16:
I have bought one, and it has the described problems:
Only setting of fading time possible and everything else gots reseted after a short time.
Presence is always false.
Would it help to sniff the communication between the sensor and a tuya gateway? I would consider doing that if chances are high to fix the problems this way.

Gutifarra, 2023-10-22:
I'm having the same issues described previously. Strongly considering returning it as the settings do not stick, and none of the readings are accurate (neither presence nor presence state).

euroman2007, 2023-10-23:
Just got this device. Connected it with my home assistant using my sonoff Zigbee dongle. In combination with the addon zigbee2mqtt. It worked out of the box. It worked accurate. Until I have tried to decrease the sensitivity. After that it stoped working with detecting presences. Lux stil works. I am gonna try the above solution now with z2m edge. Hope it will work for the time being. But I hope there will be a permanent solution soon 🙏🏻🤞

ippo112, 2023-10-24:
I'm the same issues descibed in the first post.
I use the device in HA with zigbee sonoff dongle and Z2M add-on. 
I try to upload the external connector by @mindtripper (slightly modified) and now it's better, 
the presence and presence state work, but it's not possible to set "Motion sensitivity" and "Presence sensitivity", because these parameters automatically return to default values after few seconds.
I hope hope there is a definitive fix to the problem

GormaH, 2023-10-25:
> Just got this device. Connected it with my home assistant using my sonoff Zigbee dongle. In combination with the addon zigbee2mqtt. It worked out of the box. It worked accurate. Until I have tried [...]

Hi,

Same here, tried @mindtripper solution, worked for 10mn, then stopped detecting presence or movements. Detection distance min stay at 0. Lux still work.

nickgoh, 2023-10-25:
i have modified the converter base on the param of tuya app, so far is working good .

motion_sensitivity : Level 2-7 (base on tuya dev device log only, not sure how accurate)
present_sensitivity : Level 3-9 (base on tuya dev device log only, not sure how accurate)
sensitivity value: lowest is the most sensitive 
distance : 1.5-5.5m ( base on tuya app)

issue :
i am not sure why the value of motion & present sensitivity will mutiply by 10 after setting

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: [
        'switch_1', 'switch_2', 'countdown_1', 'countdown_2', 'work_mode', 'bright_value', 'temp_value', 'illuminance_value', 
          'presence_state', 'move_sensitivity', 'motion_sensitivity', 'detection_distance_max', 'detection_distance_min', 'far_detection', 'target_dis_closest', 'fading_time', 'presence_sensitivity', 'man_state' 
    ],
}

const definition = {
    fingerprint: [ { modelID: 'TS0601', manufacturerName: '_TZE204_ijxvkhd0'}],
    model: 'ZY-M100-24G',
    vendor: 'TuYa',
    description: 'Micro Motion Sensor v1.2 ',
    configure: tuya.configureMagicPacket,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tzDatapoints],
	exposes: [
		e.illuminance_lux(), e.presence(),
		e.enum('presence_state', ea.STATE, ['none', 'present', 'moving']).withDescription('Presence state'),
		e.numeric('motion_sensitivity', ea.STATE_SET).withValueMin(2).withValueMax(7).withValueStep(1).withDescription('Motion sensitivity'),
		e.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(3).withValueMax(9).withValueStep(1).withDescription('Presence sensitivity'),
		e.numeric('detection_distance_max', ea.STATE_SET).withValueMin(1.5).withValueMax(5.5).withValueStep(1).withDescription('Max detection distance').withUnit('m'),
		e.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(1500).withValueStep(1).withDescription('Delay time').withUnit('s'),
		e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),

	],

	meta: {
		tuyaDatapoints: [
			[104, 'illuminance_lux', tuya.valueConverter.raw],
			// [105, 'presence_state', tuya.valueConverterBasic.lookup({'none': tuya.enum(0), 'present': tuya.enum(1), 'moving': tuya.enum(2)})],
			[105, 'presence_state', tuya.valueConverterBasic.lookup({'None': tuya.enum(0), 'Present': tuya.enum(1), 'Moving': tuya.enum(2)})],
			[106, 'motion_sensitivity', tuya.valueConverter.raw], //2-7
// 			[106, 'motion_sensitivity', tuya.valueConverter.raw],
			[107, 'detection_distance_max', tuya.valueConverter.divideBy100],
			[109, 'target_distance', tuya.valueConverter.divideBy100],
			[110, 'fading_time', tuya.valueConverter.raw],
			[111, 'presence_sensitivity', tuya.valueConverter.raw],// 2-9
			[112, 'presence', tuya.valueConverter.trueFalse1],
		],
	},
};

module.exports = definition;
```

GormaH, 2023-10-25:
> i have modified the converter base on the param of tuya app, so far is working good .
> 
> motion_sensitivity : Level 2-7 (base on tuya dev device log only, not sure how accurate) presence_sensitivity : Level 3-9 (base on tuya dev device log only, not sure how accurate) sensitivity value: lowest is the most sensitive distance : 1.5-5.5m ( base on tuya app) fading Time : 1-1500 sec (base on tuya app)
> 
> issue : i am not sure why the value of motion & present sensitivity will mutiply by 10 after setting
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const tzDatapoints = {
>     ...tuya.tz.datapoints,
>     key: [
>         'switch_1', 'switch_2', 'countdown_1', 'countdown_2', 'work_mode', 'bright_value', 'temp_value', 'illuminance_value', 
>           'presence_state', 'move_sensitivity', 'motion_sensitivity', 'detection_distance_max', 'detection_distance_min', 'far_detection', 'target_dis_closest', 'fading_time', 'presence_sensitivity', 'man_state' 
>     ],
> }
> 
> const definition = {
>     fingerprint: [ { modelID: 'TS0601', manufacturerName: '_TZE204_ijxvkhd0'}],
>     model: 'ZY-M100-24G',
>     vendor: 'TuYa',
>     description: 'Micro Motion Sensor v1.2 ',
>     configure: tuya.configureMagicPacket,
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tzDatapoints],
> 	exposes: [
> 		e.illuminance_lux(), e.presence(),
> 		e.enum('presence_state', ea.STATE, ['none', 'present', 'moving']).withDescription('Presence state'),
> 		e.numeric('motion_sensitivity', ea.STATE_SET).withValueMin(2).withValueMax(7).withValueStep(1).withDescription('Motion sensitivity'),
> 		e.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(3).withValueMax(9).withValueStep(1).withDescription('Presence sensitivity'),
> 		e.numeric('detection_distance_max', ea.STATE_SET).withValueMin(1.5).withValueMax(5.5).withValueStep(1).withDescription('Max detection distance').withUnit('m'),
> 		e.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(1500).withValueStep(1).withDescription('Delay time').withUnit('s'),
> 		e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
> 
> 	],
> 
> 	meta: {
> 		tuyaDatapoints: [
> 			[104, 'illuminance_lux', tuya.valueConverter.raw],
> 			[105, 'presence_state', tuya.valueConverterBasic.lookup({'None': tuya.enum(0), 'Present': tuya.enum(1), 'Moving': tuya.enum(2)})],
> 			[106, 'motion_sensitivity', tuya.valueConverter.raw], //2-7
> 			[107, 'detection_distance_max', tuya.valueConverter.divideBy100],
> 			[109, 'target_distance', tuya.valueConverter.divideBy100],
> 			[110, 'fading_time', tuya.valueConverter.raw],
> 			[111, 'presence_sensitivity', tuya.valueConverter.raw],// 2-9
> 			[112, 'presence', tuya.valueConverter.trueFalse1],
> 		],
> 	},
> };
> 
> module.exports = definition;
> ```

Thank you, it works, but yes the values multiply by 10 after a while when set. But it seem to move in the right direction !

ippo112, 2023-10-25:
> i have modified the converter base on the param of tuya app, so far is working good .
>
> motion_sensitivity : Level 2-7 (base on tuya dev device log only, not sure how accurate) presence_sensitivity : [...]

modify two row ..

			[106, 'motion_sensitivity', tuya.valueConverter.divideBy10],

			[111, 'presence_sensitivity', tuya.valueConverter.divideBy10],

now the two value of  motion & present sensitivity are corrects

euroman2007, 2023-10-25:
Thanks That Worked for me. the Script is working perfectly. i did change 1 smal thing to. the 1 to 100 value from the sliders. ive changed them back to 10 and now the sensetivity slider is also correct. an working without flaws. 

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: [
        'switch_1', 'switch_2', 'countdown_1', 'countdown_2', 'work_mode', 'bright_value', 'temp_value', 'illuminance_value', 
        'presence_state', 'move_sensitivity', 'motion_sensitivity', 'detection_distance_max', 'detection_distance_min', 'far_detection', 'target_dis_closest', 'fading_time', 'presence_sensitivity', 'man_state' 
    ],
}

const definition = {
    fingerprint: [ { modelID: 'TS0601', manufacturerName: '_TZE204_ijxvkhd0'}],
    model: 'ZY-M100-24G',
    vendor: 'TuYa',
    description: 'Micro Motion Sensor v1.2 ',
    configure: tuya.configureMagicPacket,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tzDatapoints],
	exposes: [
		e.illuminance_lux(), e.presence(),
		e.enum('presence_state', ea.STATE, ['none', 'present', 'moving']).withDescription('Presence state'),
		e.numeric('motion_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(10).withValueStep(1).withDescription('Motion sensitivity'),
		e.numeric('detection_distance_max', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withDescription('Max detection distance').withUnit('m'),
		e.numeric('detection_distance_min', ea.STATE).withDescription('Min detection distance').withUnit('m'),
		e.numeric('fading_time', ea.STATE_SET).withValueMin(1).withValueMax(15).withValueStep(1).withDescription('Delay time').withUnit('s'),
		e.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(10).withValueStep(1).withDescription('Presence sensitivity'),
	],
	meta: {
		tuyaDatapoints: [
			[104, 'illuminance_lux', tuya.valueConverter.raw],
			// [105, 'presence_state', tuya.valueConverterBasic.lookup({'none': tuya.enum(0), 'present': tuya.enum(1), 'moving': tuya.enum(2)})],
			[105, 'presence_state', tuya.valueConverterBasic.lookup({'None': tuya.enum(0), 'Present': tuya.enum(1), 'Moving': tuya.enum(2)})],
			// [106, 'motion_sensitivity', tuya.valueConverter.divideBy10],
			[106, 'motion_sensitivity', tuya.valueConverter.divideBy10],
			[107, 'detection_distance_max', tuya.valueConverter.divideBy100],
			[109, 'detection_distance_min', tuya.valueConverter.divideBy100],
			[110, 'fading_time', tuya.valueConverter.raw],
			[111, 'presence_sensitivity', tuya.valueConverter.divideBy10],
			// [111, 'presence_sensitivity', tuya.valueConverter.divideBy10],
			// [112, 'presence',  tuya.valueConverterBasic.lookup({'nobody': tuya.enum(0), 'exist': tuya.enum(1) })],
			[112, 'presence', tuya.valueConverter.trueFalse1],
		],
	},
};

module.exports = definition;
`

1,  		e.numeric('motion_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(10).withValueStep(1).withDescription('Motion sensitivity'),

2, 		e.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(10).withValueStep(1).withDescription('Presence sensitivity'),

euroman2007, 2023-10-25:
sorry dont know how to ad a script here the right way yet :(

Dillton, 2023-10-25:
> modify two row ..
>
> ```
> 		[106, 'motion_sensitivity', tuya.valueConverter.divideBy10],
>
> 		[111, 'presence_sensitivity', tuya.valueConverter.divideBy10],
> ```
>
> now the two value of motion & present [...]

Confirm works better with this code.

But question is does anyone have the ceiling version of this sensor?
Because the settings not working for 100% for it.
I can set any value for motion and presence sensitivity but is always change to 10:
[image]

dinde, 2023-10-25:
Hi,
Same issues here.

Tested the latest converter with following settings:
- Motion Sensitivity: 2 (minimum I could set)
- Presence Sensitivity: 3 (same minimum)
- Detection Distance Max: 1,5m (same minimum)
- Fading time: 10s

Door is closed when I pass by the closed door, it detects motion and presence.
I can't set less sentivity or detection distance less than these values.

Hope it helps.

nickgoh, 2023-10-26:
> Hi, Same issues here. Tested with the Ceiling version 24G and Wall version 24G Both are using the external converter provided here.
> 
> Tested the latest converter with following settings:
> 
> * Motion Sensitivity: 2 (minimum I could set)
> * Presence Sensitivity: 3 (same minimum)
> * Detection Distance Max: 1,5m (same minimum)
> * Fading time: 10s
> 
> Door is closed when I pass by the closed door, it detects motion and presence. I can't set less sentivity or detection distance less than these values.
> 
> Quick question: is the 5.8G version working correctly in z2m ?
> 
> Hope it helps.

try set the sensitivity the other way around

nickgoh, 2023-10-26:
> > > i have modified the converter base on the param of tuya app, so far is working good .
> > > motion_sensitivity : Level 2-7 (base on tuya dev device log only, not sure how accurate) presence_sensitivity : Level 3-9 (base on tuya dev device log only, not sure how accurate) sensitivity value: lowest is the most sensitive distance : 1.5-5.5m ( base on tuya app) fading Time : 1-1500 sec (base on tuya app)
> > > issue : i am not sure why the value of motion & present sensitivity will mutiply by 10 after setting
> > > ```js
> > > const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> > > const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> > > const exposes = require('zigbee-herdsman-converters/lib/exposes');
> > > const reporting = require('zigbee-herdsman-converters/lib/reporting');
> > > const extend = require('zigbee-herdsman-converters/lib/extend');
> > > const ota = require('zigbee-herdsman-converters/lib/ota');
> > > const tuya = require('zigbee-herdsman-converters/lib/tuya');
> > > const utils = require('zigbee-herdsman-converters/lib/utils');
> > > const globalStore = require('zigbee-herdsman-converters/lib/store');
> > > const e = exposes.presets;
> > > const ea = exposes.access;
> > > 
> > > const tzDatapoints = {
> > >     ...tuya.tz.datapoints,
> > >     key: [
> > >         'switch_1', 'switch_2', 'countdown_1', 'countdown_2', 'work_mode', 'bright_value', 'temp_value', 'illuminance_value', 
> > >           'presence_state', 'move_sensitivity', 'motion_sensitivity', 'detection_distance_max', 'detection_distance_min', 'far_detection', 'target_dis_closest', 'fading_time', 'presence_sensitivity', 'man_state' 
> > >     ],
> > > }
> > > 
> > > const definition = {
> > >     fingerprint: [ { modelID: 'TS0601', manufacturerName: '_TZE204_ijxvkhd0'}],
> > >     model: 'ZY-M100-24G',
> > >     vendor: 'TuYa',
> > >     description: 'Micro Motion Sensor v1.2 ',
> > >     configure: tuya.configureMagicPacket,
> > >     fromZigbee: [tuya.fz.datapoints],
> > >     toZigbee: [tzDatapoints],
> > > 	exposes: [
> > > 		e.illuminance_lux(), e.presence(),
> > > 		e.enum('presence_state', ea.STATE, ['none', 'present', 'moving']).withDescription('Presence state'),
> > > 		e.numeric('motion_sensitivity', ea.STATE_SET).withValueMin(2).withValueMax(7).withValueStep(1).withDescription('Motion sensitivity'),
> > > 		e.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(3).withValueMax(9).withValueStep(1).withDescription('Presence sensitivity'),
> > > 		e.numeric('detection_distance_max', ea.STATE_SET).withValueMin(1.5).withValueMax(5.5).withValueStep(1).withDescription('Max detection distance').withUnit('m'),
> > > 		e.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(1500).withValueStep(1).withDescription('Delay time').withUnit('s'),
> > > 		e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
> > > 
> > > 	],
> > > 
> > > 	meta: {
> > > 		tuyaDatapoints: [
> > > 			[104, 'illuminance_lux', tuya.valueConverter.raw],
> > > 			[105, 'presence_state', tuya.valueConverterBasic.lookup({'None': tuya.enum(0), 'Present': tuya.enum(1), 'Moving': tuya.enum(2)})],
> > > 			[106, 'motion_sensitivity', tuya.valueConverter.raw], //2-7
> > > 			[107, 'detection_distance_max', tuya.valueConverter.divideBy100],
> > > 			[109, 'target_distance', tuya.valueConverter.divideBy100],
> > > 			[110, 'fading_time', tuya.valueConverter.raw],
> > > 			[111, 'presence_sensitivity', tuya.valueConverter.raw],// 2-9
> > > 			[112, 'presence', tuya.valueConverter.trueFalse1],
> > > 		],
> > > 	},
> > > };
> > > 
> > > module.exports = definition;
> > > ```
> > 
> > 
> > modify two row ..
> > ```
> > 		[106, 'motion_sensitivity', tuya.valueConverter.divideBy10],
> > 
> > 		[111, 'presence_sensitivity', tuya.valueConverter.divideBy10],
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > now the two value of motion & present sensitivity are corrects
> 
> Confirm works better with this code.
> 
> But question is does anyone have the ceiling version of this sensor? Because the settings not working for 100%. I can set any value for motion and presence sensitivity but is always change to 10: [image]
> 
> I think this might be similar situation with previous 5G generation where the drivers were a bit different for both if I am not mistaken.

Hi, i am using the ceiling version and the setting sticks , try to reset the divece n repair ?

achessoux, 2023-10-26:
> Tested the latest converter with following settings:
>
> * Motion Sensitivity: 2 (minimum I could set)
> * Presence Sensitivity: 3 (same minimum)
> * Detection Distance Max: 1,5m (same minimum)
> * Fading [...]

Hello,

I've the Ceiling version 24G, with the same settings and the custom converter given above, I've the same issue : my toilet light gets on when I pass by the closed door.
No possibility to lower the distance and the sensitivity...
I used to have a movement detector from aqara and it was the opposite: the light turned off after a while when I was sitted (because it is a movement sensor). That's why I bought this presence sensor.

To see this sensor working perfectly, I rely on developer work. So, while waiting for an update, I take the time to gracefully thank @MTokarev for his first steps, @mindtripper for his fine tuning and also @Koenkk for his general and huge implication.

Dataninja126, 2023-10-29:
> modify two row ..
>
> ```
> 		[106, 'motion_sensitivity', tuya.valueConverter.divideBy10],
>
> 		[111, 'presence_sensitivity', tuya.valueConverter.divideBy10],
> ```
>
> now the two value of motion & present [...]

Thank you for this code! The modifications are working  much better; however I'm experiencing the presence  state switch from "None" to Present" and vice versa every few minutes regardless of the sensitivity settings.

Dillton, 2023-10-30:
> Hi, i am using the ceiling version and the setting sticks , try to reset the divece n repair ?

Thanks for the tip but reseting and repairing was the first I tried. So unfortunatelly this does not help.

dinde, 2023-10-30:
Hi guys,

I can't test it because I sent it back but there is the converter provided by seller below:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const legacy = require('zigbee-herdsman-converters/lib/legacy');

const tuya = require('zigbee-herdsman-converters/lib/tuya');

const dataTypes = {
	raw: 0, // [ bytes ]
	bool: 1, // [0/1]
	number: 2, // [ 4 byte value ]
	string: 3, // [ N byte string ]
	enum: 4, // [ 0-255 ]
	bitmap: 5, // [ 1,2,4 bytes ] as bits
};

const dpMap = {
	dpPresenceState: 112, //æ˜¯å¦å­˜åœ¨ï¼Œä»…ä¸ŠæŠ¥
	dpState: 105, //æ„Ÿåº”çŠ¶æ€
	dpMoveSensitivity: 106, //çµæ•åº¦
	dpPresenceSensitivity: 111, //çµæ•åº¦

	dpTimeout: 110, //æ„Ÿåº”å»¶è¿Ÿ

	dpDistance: 109, //ç›®æ ‡è·ç¦»

	dpRange: 107, //æœ€è¿œè·ç¦»èŒƒå›´
	dpIlluminanceLux: 104, //å…‰ç…§åº¦

};
const fzLocal = {
	cluster: 'manuSpecificTuya',
	type: ['commandDataResponse', 'commandDataReport'],
	convert: (model, msg, publish, options, meta) => {
		const dp = msg.data.dpValues[0].dp;
		const data = msg.data;
		const value = legacy.getDataValue(data.dpValues[0]);
		const result = {};

		switch (dp) {
			case dpMap.dpPresenceState:
				result.presence = value ? true : false;
				break;
			case dpMap.dpMoveSensitivity: //çµæ•åº¦
				result.move_sensitivity = (value / 10);
				break;
			case dpMap.dpPresenceSensitivity: //é™æ­¢çµæ•åº¦
				result.presence_sensitivity = (value / 10);
				break;
			case dpMap.dpRange: //é›·è¾¾è·ç¦»15-55
				result.radar_range = (value / 100);
				break;
			case dpMap.dpDistance: //ç›®æ ‡è·ç¦»
				result.distance = (value / 100);
				break;
			case dpMap.dpTimeout: //å»¶è¿Ÿæ—¶é—´0-600
				result.presence_timeout = (value);
				break;
			case dpMap.dpIlluminanceLux: //å…‰ç…§åº¦
				result.illuminance_lux = (value);
				break;

			case dpMap.dpState:
				result.state = {
					0: 'none',
					1: 'presence',
					2: 'move'
				} [value];
				break;

				// meta.logger.debug(
				// 	`æœªè§£æžçš„æ•°æ®DP: ${dp} DATA: ${JSON.stringify(msg.data)}`
				// );

		}
		return result;
	},
}
const tzLocal = {
	key: [
		'move_sensitivity',
		'presence_sensitivity',
		'radar_range',
		'presence_timeout',

	],
	convertSet: async (entity, key, value, meta) => {

		switch (key) {
			case 'move_sensitivity':
				await legacy.sendDataPointValue(entity, dpMap.dpMoveSensitivity, value);
				break;
			case 'presence_sensitivity':
				await legacy.sendDataPointValue(entity, dpMap.dpPresenceSensitivity, value);
				break;
			case 'radar_range':
				await legacy.sendDataPointValue(entity, dpMap.dpRange, value * 100);
				break;
			case 'presence_timeout':
				await legacy.sendDataPointValue(entity, dpMap.dpTimeout, value);
				break;

		}
		return {
			key: value
		};
	},

}

module.exports = [{
	fingerprint: [{
		modelID: 'TS0601',
		manufacturerName: '_TZE204_ijxvkhd0',
	}],
	model: '24G äººä½“å­˜åœ¨ä¼ æ„Ÿå™¨',
	vendor: 'TuYa',
	description: '24G äººä½“å­˜åœ¨ä¼ æ„Ÿå™¨',
	fromZigbee: [fzLocal],
	toZigbee: [tzLocal],
	onEvent: legacy.onEventSetLocalTime,
	exposes: [

		exposes.enum('state', ea.STATE, ['none', 'presence', 'move'])
		.withDescription('æ„Ÿåº”çŠ¶æ€'),

		e.presence().withDescription('æ˜¯å¦åœ¨å®¶'),

		exposes.numeric('distance', ea.STATE)
		.withDescription('ç›®æ ‡è·ç¦»'),

		e.illuminance_lux(),
		exposes.numeric('move_sensitivity', ea.STATE_SET).withValueMin(1)
		.withValueMax(10)
		.withValueStep(1)
		.withDescription('è¿åŠ¨é›·è¾¾çµæ•åº¦'),

		exposes.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(1)
		.withValueMax(10)
		.withValueStep(1)
		.withDescription('é™æ­¢é›·è¾¾çµæ•åº¦'),

		exposes.numeric('radar_range', ea.STATE_SET).withValueMin(1.5)
		.withValueMax(5.5)
		.withValueStep(1)
		.withUnit('ç±³').withDescription('é›·è¾¾æ„Ÿåº”è·ç¦»'),

		exposes.numeric('presence_timeout', ea.STATE_SET).withValueMin(1)
		.withValueMax(1500)
		.withValueStep(1)
		.withUnit('ç§’').withDescription('å»¶è¿Ÿæ—¶é—´'),

	],
	meta: {
		multiEndpoint: true,
		tuyaDatapoints: [

			[112, 'presence', tuya.valueConverter.trueFalse1],
			[106, 'move_sensitivity', tuya.valueConverter.divideBy10],
			[111, 'presence_sensitivity', tuya.valueConverter.divideBy10],

			[107, 'radar_range', tuya.valueConverter.divideBy100],
			[109, 'distance', tuya.valueConverter.divideBy100],
			[110, 'presence_timeout', tuya.valueConverter.raw],
			[104, 'illuminance_lux', tuya.valueConverter.raw],
			[105, 'state', tuya.valueConverterBasic.lookup({
				'none': 0,
				'presence': 1,
				'move': 2
			})],

		],
	},

}, ];

HTH

akumamg, 2023-10-30:
I just tested 

> Hi guys,
>
> I can't test it because I sent it back but there is the converter provided by seller below:
>
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

I just tested and looks like its working :)
But you need translate description.

Dataninja126, 2023-10-30:
@akumamg could you provide the translated  description version :)

dinde, 2023-10-30:
> @akumamg could you provide the translated description version :)

There you go:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const legacy = require('zigbee-herdsman-converters/lib/legacy');

const tuya = require('zigbee-herdsman-converters/lib/tuya');

const dataTypes = {
	raw: 0, // [ bytes ]
	bool: 1, // [0/1]
	number: 2, // [ 4 byte value ]
	string: 3, // [ N byte string ]
	enum: 4, // [ 0-255 ]
	bitmap: 5, // [ 1,2,4 bytes ] as bits
};

const dpMap = {
	dpPresenceState: 112, //æ˜¯å�¦å­˜åœ¨ï¼Œä»…ä¸ŠæŠ¥
	dpState: 105, //æ„Ÿåº”çŠ¶æ€�
	dpMoveSensitivity: 106, //ç�µæ•�åº¦
	dpPresenceSensitivity: 111, //ç�µæ•�åº¦

	dpTimeout: 110, //æ„Ÿåº”å»¶è¿Ÿ

	dpDistance: 109, //ç›®æ ‡è·�ç¦»

	dpRange: 107, //æœ€è¿œè·�ç¦»èŒƒå›´
	dpIlluminanceLux: 104, //å…‰ç…§åº¦

};
const fzLocal = {
	cluster: 'manuSpecificTuya',
	type: ['commandDataResponse', 'commandDataReport'],
	convert: (model, msg, publish, options, meta) => {
		const dp = msg.data.dpValues[0].dp;
		const data = msg.data;
		const value = legacy.getDataValue(data.dpValues[0]);
		const result = {};

		switch (dp) {
			case dpMap.dpPresenceState:
				result.presence = value ? true : false;
				break;
			case dpMap.dpMoveSensitivity: 
				result.move_sensitivity = (value / 10);
				break;
			case dpMap.dpPresenceSensitivity: 
				result.presence_sensitivity = (value / 10);
				break;
			case dpMap.dpRange: 
				result.radar_range = (value / 100);
				break;
			case dpMap.dpDistance: 
				result.distance = (value / 100);
				break;
			case dpMap.dpTimeout: 
				result.presence_timeout = (value);
				break;
			case dpMap.dpIlluminanceLux: 
				result.illuminance_lux = (value);
				break;

			case dpMap.dpState:
				result.state = {
					0: 'none',
					1: 'presence',
					2: 'move'
				} [value];
				break;
		}
		return result;
	},
}
const tzLocal = {
	key: [
		'move_sensitivity',
		'presence_sensitivity',
		'radar_range',
		'presence_timeout',

	],
	convertSet: async (entity, key, value, meta) => {

		switch (key) {
			case 'move_sensitivity':
				await legacy.sendDataPointValue(entity, dpMap.dpMoveSensitivity, value);
				break;
			case 'presence_sensitivity':
				await legacy.sendDataPointValue(entity, dpMap.dpPresenceSensitivity, value);
				break;
			case 'radar_range':
				await legacy.sendDataPointValue(entity, dpMap.dpRange, value * 100);
				break;
			case 'presence_timeout':
				await legacy.sendDataPointValue(entity, dpMap.dpTimeout, value);
				break;

		}
		return {
			key: value
		};
	},

}

module.exports = [{
	fingerprint: [{
		modelID: 'TS0601',
		manufacturerName: '_TZE204_ijxvkhd0',
	}],
	model: 'ZY-M100-24G',
	vendor: 'TuYa',
	description: '24G Micro Motion Sensor',
	fromZigbee: [fzLocal],
	toZigbee: [tzLocal],
	onEvent: legacy.onEventSetLocalTime,
	exposes: [

		exposes.enum('state', ea.STATE, ['none', 'presence', 'move'])
		.withDescription('Presence State'),

		e.presence(),

		exposes.numeric('distance', ea.STATE)
		.withDescription('Distance'),

		e.illuminance_lux(),
		exposes.numeric('move_sensitivity', ea.STATE_SET).withValueMin(1)
		.withValueMax(10)
		.withValueStep(1)
		.withDescription('Motion Sensitivity'),

		exposes.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(1)
		.withValueMax(10)
		.withValueStep(1)
		.withDescription('Presence Sensitivity'),

		exposes.numeric('radar_range', ea.STATE_SET).withValueMin(1.5)
		.withValueMax(5.5)
		.withValueStep(1)
		.withUnit('m').withDescription('Detection Range'),

		exposes.numeric('presence_timeout', ea.STATE_SET).withValueMin(1)
		.withValueMax(1500)
		.withValueStep(1)
		.withUnit('s').withDescription('Presence Time Out'),

	],
	meta: {
		multiEndpoint: true,
		tuyaDatapoints: [

			[112, 'presence', tuya.valueConverter.trueFalse1],
			[106, 'move_sensitivity', tuya.valueConverter.divideBy10],
			[111, 'presence_sensitivity', tuya.valueConverter.divideBy10],

			[107, 'radar_range', tuya.valueConverter.divideBy100],
			[109, 'distance', tuya.valueConverter.divideBy100],
			[110, 'presence_timeout', tuya.valueConverter.raw],
			[104, 'illuminance_lux', tuya.valueConverter.raw],
			[105, 'state', tuya.valueConverterBasic.lookup({
				'none': 0,
				'presence': 1,
				'move': 2
			})],

		],
	},

}, ];`

nickgoh, 2023-10-31:
> There you go:
>
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = [...]

Confirm working , 1 had the higest sensitivity and 10  being the lowest.

Sensitivity value will divided by 10 after set but will return to normal value after a while.

[116 more replies]

closed 2024-11-03 by github-actions[bot]