Koenkk/zigbee2mqtt #18238: Third Reality 3RSNL02043Z occupancy detection not working

darkxst, 2023-07-07:
### What happened?

Just tried testing this night light device, it is not reporting any occupancy data for the motion sensor. I tested both 1.32.0 and latest Edge build.

I see this in the logs
Zigbee2MQTT:debug 2023-07-07 21:24:42: Received Zigbee message from '0xb83dfb0858ca0000', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 2 with groupID null
Zigbee2MQTT:debug 2023-07-07 21:24:42: No converter available for '3RSNL02043Z' with cluster 'ssIasZone' and type 'attributeReport' and data '{"zoneStatus":1}'

I saw this [commit](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5914/commits/5c6c5b7a20fa79004866afdd03bb106cc091f057) but that seems to be using some private cluster and not IasZone.

### What did you expect to happen?

Should detect occupancy when motion sensor is triggered.

### Zigbee2MQTT version

1.32.0

### Adapter firmware version

2103

### Adapter

Thirdreality Zigate dongle

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11981399/log.txt)
[database.txt](https://github.com/Koenkk/zigbee2mqtt/files/11981410/database.txt)

Koenkk, 2023-07-07:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/db78116991cd6b1b6c2b9c21c98b5aa4

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

darkxst, 2023-07-08:
Yes this fixes the occupancy value. It does however add battery and tamper values to the MQTT payload, not sure that this matters though since the entities don't appear in discover data for HA.

The occupancy now also triggers the LED to turn on for a few minutes, I guess this is probably the intended behaviour given the night light style of device. It must be hardcoded in the device firmware?

Finally, not sure if the colour temp range is set correctly? I get the following values when I query the device
```
Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":0,"colorTempPhysicalMax":65279}
```
so not sure where the `[153, 555]` values came from!

darkxst, 2023-07-08:
> The occupancy now also triggers the LED to turn on

This only happens when colour hasnt been set, once I set a colour control of the LED is manual only

Koenkk, 2023-07-08:
> The occupancy now also triggers the LED to turn on for a few minutes, I guess this is probably the intended behaviour given the night light style of device. It must be hardcoded in the device firmware?

Yes, I don't expect the ext converter to have changed this.

> Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":0,"colorTempPhysicalMax":65279}

These are not realistic, 153, 555 is fine.

Can you confirm everything works ok now? Then I will integrate the changes.

darkxst, 2023-07-08:
Yes everything works good now.

Koenkk, 2023-07-09:
Great, integrated the fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-07-09 by Koenkk

jesse-schein, 2023-08-11:
I tried to add this converter to 1.32.2 and its not able to detect the occupancy. It just says "Null" on occupancy.

[image]

[image]

[image]

[image]

jesse-schein, 2023-08-17:
Ahh I see that zigbee2mqtt is merely the UI and MQTT parts. I manually modified zigbee-herdsman-converters and found the right javascript file and manually made changes to it from the other repo. Upon doing that, and deleting my external converter, things work now.

I suspect in the next release of zigbee2mqtt it will be available, since the package.json was using an older version in the most recently released version that I assumed had the fix.

findthelorax, 2023-09-09:
What changes did you make to get this device to work? I have the same issue.