Koenkk/zigbee2mqtt #18241: SP-EUC01 - Move power readings in HA GUI from diagnosis to sensors

MaxDev98, 2023-07-07:
### What happened?

Hi,

I hope this issue fits in here, unfortunately I wasn't sure about the topic I should choose.

I am using a SP-EUC01 smart plug to monitor the power consumption and everything works just fine.
Within the HA GUI for this device:
- total energy consuption is shown below "Sensors"
- current power consuption is shown below "Diagnostics"

### What did you expect to happen?

In my opinion, however, the current power consumption is also a sensor. 
Therefore, it would make more sense to also classify the current power consumption as a "sensor".

This way the current power consumption can also be displayed automatically on the HA dashboard, since all "diagnostic" entities are not displayed.

### How to reproduce it (minimal and precise)

Connect a SP-EUC01 and open the device overview in HA.
You will find "energy" shown below sensors and "power" shown below diagnostics.

### Zigbee2MQTT version

1.32.0

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus (zStack3x0)

mikimause9, 2023-07-23:
I have a very similar situation. I have powerTag A9MEM1570. In my case, the sensor shows three voltages in three phases, but phase A is in the sensor category, while phases B and C are in the diagnostic category. This is very annoying. Hopefully there is some way to move them.

rotilho, 2023-08-04:
It is a sensor but is categorized as diagnostic. There were some discussions around that before. The position from HA is that any sensor not part of the main operation of the device is considered diagnostic.

You can still import whatever you want into your dashboard, it will work as any other sensor.

closed 2023-09-11 by github-actions[bot]

tispokes, 2023-12-21:
@rotilho but you can't insert it into the default dashboard. And why is energy a sensor but power not. And ZHA behaves different.
Is there some hack to change this?

danieldietsch, 2024-06-24:
I have the same issue, but with a https://www.zigbee2mqtt.io/devices/A1Z.html#nous-a1z: 
zigbee2mqtt displays various sensors in its dashboard: 
[image]

For home assistant, these are show as controls and sensors:
[image]
But the current power draw is diagnostic. 
[image]

This seems kind of weird, as I use them mainly to measure exactly this value. It also prevents HA from using that value during autocomplete in the energy configuration. I can use the total data, but this requires me to manually reset that counter when switching devices or locations. 

It would be very nice if there was at least a way to move the value around.

brickronny, 2025-02-05:
Every word @danieldietsch wrote. Don‘t agree to @rothilo‘s argumentation.

DanaGoyette, 2025-02-12:
It's ridiculous to bury a power meter's power meter under Diagnostic sensors, when that feature is the primary reason I'd even buy a power metering smart plug.

In my case, it's happening with a Sengled E1C-NB7.

Every time I've considered switching from ZHA to Z2M, I've run into this issue, and had to give up, because it ruins one of my main uses of the default dashboard.

RolandRosier, 2025-02-16:
I also have a Nouse A1Z Zigbee plug, which puts the instantaneous power meter under Diagnostic sensors - and this doesn't show up in the default dashboard - and this was the main reason that I bought the plug.

However, lifetime power meter is put under Sensors. This makes no sense to me. How can I expose the instantaneous power meter  on the default dashboard (which probably means moving it to Sensors)?

mrfifiak, 2025-02-18:
In my opinion, power is a “part of the main operation of the device”. It’s even more important than energy for me.

hal01, 2025-03-09:
Idem

In addition in ZHA integration, Power is part of sensors
So arguing that HA required it in diagnosis makes no sens at all

why this choice from ZigbeeMQTT ?

hal01, 2025-04-03:
Seems it has been corrected...
when... I don't know, but now for my plugs I have power in "auto" dahsboard and in sensors for MQTT

[image]

[image]

MaxDev98, 2025-04-03:
This issue has been resolved with #26769, zigbee2mqtt v.2.2.0