Koenkk/zigbee2mqtt #18242: [Wrong device]:

uvjim, 2023-07-07:
### Link

https://click-smart.com/products/mode-smart-sockets

### Model

CMA30035

### Description

1 Gang Socket Outlet

### Vendor

ClickSmart+

### Picture (link)

https://www.electrical2go.co.uk/media/catalog/product/c/m/cma30035.png

### Database entry

{"id":17,"type":"Router","ieeeAddr":"0x60a423fffeae3467","nwkAddr":26855,"manufId":4098,"manufName":"_TYZB01_mtunwanm","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,10,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":22,"65534":0,"modelId":"TS011F","manufacturerName":"_TYZB01_mtunwanm","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2123427102},"lastSeen":1688748455311,"defaultSendRequestWhen":"immediate"}

### Notes

It's difficult to find an image on the ClickSmart website now that they've revamped it so hopefully this is OK.

Koenkk, 2023-07-07:
fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-07-07 by Koenkk

uvjim, 2023-07-08:
@Koenkk - thanks for the quick turnaround. I switched to the `edge` add-on in Home Assistant. The text is correct but the image is not (see screenshot).

[image]

Koenkk, 2023-07-08:
@uvjim image will be correct after the next release.