Koenkk/zigbee2mqtt #18243: Aqara Water Sensor SJCGQ11LM reports Unknown (issue in closed case: #17145)

apadua, 2023-07-07:
### What happened?

The listed Aqara water leak sensor (SJCGQ11LM) shows water_leak state as Unknown until it is triggered by water for the first time. After that, they show as Clear when dry.

Although i have no way of testing, it seems like the same thing is happening to the battery_low attribute.

### Device that has never seen Water:
**Regular Update:**
```
Debug 2023-07-07 12:00:50Received Zigbee message from 'Master Bathroom: Water Leak Sensor', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3195,"10":2538,"100":0,"3":30,"4":17320,"5":29,"6":[0,1],"8":518}}' from endpoint 1 with groupID 0
Info 2023-07-07 12:00:50MQTT publish: topic 'zigbee2mqtt/Master Bathroom: Water Leak Sensor', payload '{"battery":100,"battery_low":null,"device_temperature":30,"last_seen":"2023-07-07T12:00:50-04:00","linkquality":76,"power_outage_count":28,"voltage":3195,"water_leak":null}'
```
**Clicking on Button:**
```
Debug 2023-07-07 12:51:02Received Zigbee message from 'Master Bathroom: Water Leak Sensor', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3195,"10":2538,"100":0,"3":30,"4":17320,"5":29,"6":[0,0],"8":518},"modelId":"lumi.sensor_wleak.aq1"}' from endpoint 1 with groupID 0
Info 2023-07-07 12:51:02MQTT publish: topic 'zigbee2mqtt/Master Bathroom: Water Leak Sensor', payload '{"battery":100,"battery_low":null,"device_temperature":30,"last_seen":"2023-07-07T12:51:03-04:00","linkquality":69,"power_outage_count":28,"voltage":3195,"water_leak":null}'
```
**Z2M GUI:**
[image]

### Device that has seen water:
**Regular Update:**
```
Debug 2023-07-07 11:46:45Received Zigbee message from 'Laundry Room: Downstairs AC Water Leak Sensor', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3035,"10":49409,"100":0,"3":29,"4":17320,"5":21,"6":[0,1],"8":518}}' from endpoint 1 with groupID 0
Info 2023-07-07 11:46:45MQTT publish: topic 'zigbee2mqtt/Laundry Room: Downstairs AC Water Leak Sensor', payload '{"battery":100,"battery_low":false,"device_temperature":29,"last_seen":"2023-07-07T11:46:45-04:00","linkquality":109,"power_outage_count":20,"tamper":false,"voltage":3035,"water_leak":false}'
```
**Clicking on Button:**
```
Debug 2023-07-07 12:40:09Received Zigbee message from 'Laundry Room: Downstairs AC Water Leak Sensor', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3035,"10":49409,"100":0,"3":29,"4":17320,"5":21,"6":[0,0],"8":518},"modelId":"lumi.sensor_wleak.aq1"}' from endpoint 1 with groupID 0
Info 2023-07-07 12:40:09MQTT publish: topic 'zigbee2mqtt/Laundry Room: Downstairs AC Water Leak Sensor', payload '{"battery":100,"battery_low":false,"device_temperature":29,"last_seen":"2023-07-07T12:40:09-04:00","linkquality":116,"power_outage_count":20,"tamper":false,"voltage":3035,"water_leak":false}'
```
**Z2M GUI:**
[image]

Parsing for attribute 100 is [linked here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/ee9a7ab3b7a81b55d6a7b4f05b1614958743a6db/src/lib/xiaomi.ts#L231C12-L231C12) and, although there is an "else" for this model, it should only apply when the Model ID is present. This is documented in the code notes and can the logic there be confirmed by the included log messages.

### What did you expect to happen?

New Aqara Leak Sensors would always expose Wet or Dry according to the zigbee message received.

### How to reproduce it (minimal and precise)

Add a new device to the zigbee network. See it as null. Wait as long as you want (in my case, months). Throw it in water. See it as Wet. Remove it from Water. See it as Dry.

### Zigbee2MQTT version

1.32.0 commit: unknown

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

```
Debug 2023-07-07 11:46:45Received Zigbee message from 'Laundry Room: Downstairs AC Water Leak Sensor', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3035,"10":49409,"100":0,"3":29,"4":17320,"5":21,"6":[0,1],"8":518}}' from endpoint 1 with groupID 0
Info 2023-07-07 11:46:45MQTT publish: topic 'zigbee2mqtt/Laundry Room: Downstairs AC Water Leak Sensor', payload '{"battery":100,"battery_low":false,"device_temperature":29,"last_seen":"2023-07-07T11:46:45-04:00","linkquality":109,"power_outage_count":20,"tamper":false,"voltage":3035,"water_leak":false}'
Debug 2023-07-07 12:40:09Received Zigbee message from 'Laundry Room: Downstairs AC Water Leak Sensor', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3035,"10":49409,"100":0,"3":29,"4":17320,"5":21,"6":[0,0],"8":518},"modelId":"lumi.sensor_wleak.aq1"}' from endpoint 1 with groupID 0
Info 2023-07-07 12:40:09MQTT publish: topic 'zigbee2mqtt/Laundry Room: Downstairs AC Water Leak Sensor', payload '{"battery":100,"battery_low":false,"device_temperature":29,"last_seen":"2023-07-07T12:40:09-04:00","linkquality":116,"power_outage_count":20,"tamper":false,"voltage":3035,"water_leak":false}'
Debug 2023-07-07 12:00:50Received Zigbee message from 'Master Bathroom: Water Leak Sensor', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3195,"10":2538,"100":0,"3":30,"4":17320,"5":29,"6":[0,1],"8":518}}' from endpoint 1 with groupID 0
Info 2023-07-07 12:00:50MQTT publish: topic 'zigbee2mqtt/Master Bathroom: Water Leak Sensor', payload '{"battery":100,"battery_low":null,"device_temperature":30,"last_seen":"2023-07-07T12:00:50-04:00","linkquality":76,"power_outage_count":28,"voltage":3195,"water_leak":null}'
Debug 2023-07-07 12:51:02Received Zigbee message from 'Master Bathroom: Water Leak Sensor', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3195,"10":2538,"100":0,"3":30,"4":17320,"5":29,"6":[0,0],"8":518},"modelId":"lumi.sensor_wleak.aq1"}' from endpoint 1 with groupID 0
Info 2023-07-07 12:51:02MQTT publish: topic 'zigbee2mqtt/Master Bathroom: Water Leak Sensor', payload '{"battery":100,"battery_low":null,"device_temperature":30,"last_seen":"2023-07-07T12:51:03-04:00","linkquality":69,"power_outage_count":28,"voltage":3195,"water_leak":null}'
```

TEF2one, 2023-07-13:
Same thing happened to me since last week...
So following you comment I just dipped it in water and teh value updated correctly.

apadua, 2023-07-14:
It won't last. Another week or a server reboot or two  and it will be back to unknown.

It's scary to have 8 and see a few dry and a few unknown.

You want to be sure that your water automations are 100%.

ThrashinVictim, 2023-08-03:
I was having this problem but after a reboot of HA (after upgrade etc.) I get water_leak and battery_settings listed as N/A and never goes back to any other state unless I unpair and re-pair them.

Ky0sHiR00, 2023-09-09:
Got the same problem as you guys

apadua, 2023-09-09:
We can always build a N/A into the policy, but that's horrible since you won't know if the Water sensor is actually alive or not. And it's a water leak sensor, so it's pretty important.

knudsvik, 2023-10-10:
Still an issue

hlyi, 2023-10-22:
I have the same issue. could we report "false" when current status is "N/A"?

djuntgen, 2023-11-10:
Ran into this as well.

Koenkk, 2023-11-10:
This is expected behaviour, you need to trigger the sensor once for z2m to know the state. Assuming the state is dry when pairing is incorrect

johntdyer, 2024-01-10:
@Koenkk  do you have an example of an automation to do trigger them?

apadua, 2024-01-10:
@johntdyer i think he means you need to have it get wet once before z2m knows the state. The problem with that is that z2m forgets the state - so it's harder to make an automation that detects when the device is gone vs when the device is dry for prolonged periods.

The issue i had was that the device got into the weird state some Aqara devices get wehre they're there but really aren't. And i didn't know i had lost leak detection.

I guess the solution then is to find out a way to tell the device is healthy - then assume N/A is dry until it isn't.

vitalybibikov, 2024-02-28:
Thanks for support of such a good component as zigbee2mqtt.

I am making transition from ZHA to zigbee2mqtt and noticed difference in how the latter behaves. 

For me it is a problem as well, as I am using this sensor in external integration (voice assistant).

How it triggers right now:

1) Aqara status is unknown (when sensor is new or mqtt was reloaded)
2) Aqara status changes to Wet (external voice assistant does not care as it is not considering unknown -> wet as a state change)
3) Aqara status changes to Dry
4) Aqara status changes to Wet (external voice assistant notifies me about dry-> wet state change)
5) Go to 1 after restart

I consider this sensor as Bool, not bool?, meaning that it has 2 states: Dry(false)/Wet(true), not null/false/true.

Thanks

cadavre, 2024-04-25:
So... there is a binary_sensor for leakage in Home Assistant, like mine `binary_sensor.ground_boiler_leakage_sensor_water_leak`.

*As long as no water is detected it is in state `unknown`.
* When water is detected it turns `on`.
* After water is no longer detected it turns `off`.

I have no problem with automations, because it triggers an alarm and notification when my `binary_sensor` turns `on`, no matter the previous value.

What is your problem with automations? I would love to see "Clear" instead of `unknown` as well, but it doesn't break any automation rules.

---

There is also a "Trigger count", does anyone of you know what it does?

shake0606, 2024-09-05:
For my part, I had the same problem, I couldn't get it recognized by ZigBee2mqtt, so I thought there was an issue, something wasn't right. The humidity sensor wasn't working (showed N/A). So, I opened it up, removed the two screws, and then used a small flat screwdriver along the edges in the gap to uncap it. I noticed it was a bit oxidized on the motherboard. So, I took out the batteries and unclipped it from its holder, then I rinsed it under water and scrubbed it with a toothbrush. After that, I dried it and reassembled everything, and like magic, it worked perfectly. The only minor downside is that I slightly damaged the protective cover when uncapping it

pour ma part j'avais le même problème impossible de le faire reconnaître par ZigBeetomqtt donc je me suis dit il y a un problème c'est pas normal le capteur d'humidité ne fonctionne pas  N/A donc je l'ai ouvert j'ai enlevé les deux vis et ensuite sur les rebords j'ai passé un petit tournevis plat dans la fente pour le décapsuler et j'ai remarqué qu'il était un peu oxydé au niveau de sa carte mère donc j'ai enlevé les piles et ensuite je les déclipsé de son support et je l'ai passé sous l'eau et frotter avec une brosse à dents je l'ai ensuite séché et remonter tout ça et puis comme par magie tout fonctionne parfaitement le seul inconvénient minime c'est que j'ai un petit peu abîmé le cache de protection lorsque décapsuler

closed 2025-01-13 by github-actions[bot]

Teeseeone, 2025-02-03:
Is this not still an issue? 

[image]

andreadecorte, 2025-02-26:
I also faced this issue after migrating from deconz, which was instead reporting either true or false with this sensor.

We can check if sensor is alive through the other attributes (e.g. device_temperature).
In any case I opened a PR https://github.com/Koenkk/zigbee2mqtt.io/pull/3549 to at least document this behavior.