Koenkk/zigbee2mqtt #18245: Can't Add Devices

negmos, 2023-07-07:
### What happened?

Two raspberry PI servers (PI-1, and PI-2) both have zigbee2mqtt docker containers with exact same configurations (same network info as well), but both are on different locations (different cities).
When connecting the adapter (one adapter only) to PI-1 and trying to add a device (device-1 --> aqara motion sensor) it works great.. i remove it afterwards, and take the adapter and device-1 with me to the other city with PI-2 and try to add the same device using the same exact container and network configs but nothing happens as if i'm not adding anything.
Nothing in the log at all.
Both servers are exactly the same configs, same models, same configuration.yaml file and container settings.
I tried in different location within the home in city-2 to ensure no interreference, i also tried with 1cm away from the antenna, but nothing is helping so far.

### What did you expect to happen?

I expect the same device to be added normally as in the first server.

### How to reproduce it (minimal and precise)

N/A

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20230507

### Adapter

Slaesh's CC2652RB

okastl, 2023-07-09:
The security keys and rx/tx counters are different, so commands requiting "secure" connection (like adding new device) will fail.
Although this sucks (especially the rx/tx counter stuff), this is the way it is.

negmos, 2023-07-09:
Wouldn't the fact that i'm resetting the device overcome this issue?

I tried to also reflash the stick on the 2nd location (city-2) with no luck.. still the same, nothing is showing up.

closed 2023-08-17 by github-actions[bot]