Koenkk/zigbee2mqtt #18252: TypeError: Cannot read properties of undefined (reading 'options')

milobt, 2023-07-08:
### What happened?

Updating to any version after v1.30.4-1 causes error and failure to load

### What did you expect to happen?

Z2M to load after version update.

### How to reproduce it (minimal and precise)

Install any newer version.

### Zigbee2MQTT version

1.30.4-1

### Adapter firmware version

20220219

### Adapter

Electrolama zig-a-zig-ah!

### Debug log

[14:06:15] INFO: Preparing to start...
[14:06:16] INFO: Socat not enabled
[14:06:18] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:79
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

Koenkk, 2023-07-08:
Your ext converter is outdated (they are not meant for long term usage), update the converter or consider making a PR to add support for this device out-of-the-box.

closed 2023-07-08 by Koenkk

milobt, 2023-07-08:
How do I update the external converters?

milobt, 2023-07-08:
I managed to solve the problem.
I'd previously created an external converter for a roller blind, but never committed it, as support has already been commited by someone else, as the following:

Vendor: Zemismart
device: M616EGBZTN
modelID: 'TS0601'
manufacturerName:  '_TZE200_gubdgai2'

So I updated my converter file to reference legacy code and everything loaded again, however, as this device is now supported natively, it should have worked without my converter, which may suggest that my converters need updating?
So, again, how do I go about updating them?
Do they not get updated when z2m is updated? (released, not edge)

My updated converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const legacy= require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'], // The model ID from: Device with modelID
    model: 'M616EGBZTN', // Vendor model number, look on the device for a model number
    vendor: 'EGBZTN', // Vendor of the device (only used for documentation and startup logging)
    manufacturerName: '_TZE200_gubdgai2',
    description: 'Roller Blind Motor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [legacy.fromZigbee.tuya_cover, fz.ignore_basic_report],
    toZigbee: [legacy.toZigbee.tuya_cover_control, legacy.toZigbee.tuya_cover_options],
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.composite('options', 'options')
            .withFeature(exposes.numeric('motor_speed', ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(255)
                .withDescription('Motor speed'))
	    .withFeature(e.binary('reverse_direction', ea.STATE_SET, true, false)
                .withDescription('Reverse the motor direction'))],
};

module.exports = definition;
```

Committed Zemismart converter:

```
{
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_gubdgai2'}],
        model: 'M515EGBZTN',
        vendor: 'Zemismart',
        description: 'Roller shade driver',
        fromZigbee: [legacy.fz.ZMAM02_cover],
        toZigbee: [legacy.tz.ZMAM02_cover],
        exposes: [e.cover_position().setAccess('position', ea.STATE_SET),
            e.enum('motor_direction', ea.STATE_SET, Object.values(legacy.ZMLookups.AM02Direction)),
            e.enum('border', ea.STATE_SET, Object.values(legacy.ZMLookups.AM02Border)),
        ],
    },
```

Thanks.

milobt, 2023-07-08:
I deleted the converter file and removed the reference from my configuration.yaml file and is still working.
So hopefully this is fixed now.