Koenkk/zigbee2mqtt #18259: Publish 'set' 'state' to 'xxx' failed: 'Error: 'transition' is not a number, got string ()'

karleyson, 2023-07-08:
### What happened?

I didn't change anything and it worked fine. But since an update last month (cannot exactly say which, i cannot control a simple bulp. Not possible to turn on or of - or anything. I get always this message:

Publish 'set' 'state' to 'xxx' failed: 'Error: 'transition' is not a number, got string ()'

As i get it only on this device. Others work fine. I even reconnected it, but every time i try to send an command, i get this error.

Zigbee Modell   Retro Bulb Gold XXL white+ambiance
Zigbee Manufacturer: MLI

tint LED-Globe Retro Gold XXL E27
IEEE Addresse
0xa4c13800aa440a52
Netzwerk Adresse
0xAD39
Firmware-Version
MDFIL201
Hersteller
[Müller Licht](https://www.zigbee2mqtt.io/supported-devices/#v=M%C3%BCller%20Licht)
Modell
[404065](https://www.zigbee2mqtt.io/devices/404065.html#m%25C3%25BCller%2520licht-404065)
Spannungsversorgung

### What did you expect to happen?

Lamp switches on / off

### How to reproduce it (minimal and precise)

Maybe it's only on this type of light and a config error in your files, that wasn't updatet? I have a remote, that is directly connected to the bulb, and this works great. It's only on the software.

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

20220219

### Adapter

 zStack3x0

### Debug log

Debug 2023-07-08 23:09:22Received MQTT message on 'zigbee2mqtt/Schlafzimmer Lampe /set' with data '{"state":"OFF"}'
Debug 2023-07-08 23:09:22Publishing 'set' 'state' to 'Schlafzimmer Lampe '
Error 2023-07-08 23:09:22Publish 'set' 'state' to 'Schlafzimmer Lampe ' failed: 'Error: 'transition' is not a number, got string ()'
Debug 2023-07-08 23:09:22Error: 'transition' is not a number, got string () at toNumber (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:509:15) at Object.getTransition (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:391:28) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.js:870:38) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:248:52) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:126:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1184:20)
Info 2023-07-08 23:09:22MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Schlafzimmer Lampe ' failed: 'Error: 'transition' is not a number, got string ()'","meta":{"friendly_name":"Schlafzimmer Lampe "},"type":"zigbee_publish_error"}'

Koenkk, 2023-07-09:
You probably have an invalid default transition set for this device in your configuration.yaml (`transition: ''`)

blubkens, 2023-07-09:
I got the same error on my TS0201 _TZ3000_dowj6gyi Temperature & humidity sensor. 
It is no longer reporting values because of the string error. 
So far i have never set any custom transitions for any of the devices i use.

karleyson, 2023-07-09:
Hello Koenk,
no .. my configuration yaml is clean  (- never change a running system ;-) .. so i didn't 
The only custom thing is "serial:   port: /dev/ttyUSB0"

But anyway.... as you say, there must be something with the configuration. So i removed the device and added it .. rebooted HA in between ... not it works. Thanks for supporting

closed 2023-07-09 by karleyson

RaunoVV, 2023-11-09:
> Hello, Had the same problem after an update of Zigbee2MQTT. Instead of removing the device, I manually set '0' to transition in Settings (specific) of the device. Seem to work right after that change.

Same happened to me with one of the dimmer lights, Setting transition  manually worked.