Koenkk/zigbee2mqtt #18271: [Wrong device]: VINDSTYRKA (instead of vendor "IKEA" it appears as "IKEA of Sweden")

ctorrao, 2023-07-09:
### Link

https://www.ikea.com/pt/en/p/vindstyrka-air-quality-sensor-smart-00498231/

### Model

VINDSTYRKA

### Description

Vindstyrka air quality and humidity sensor

### Vendor

IKEA of Sweden

### Picture (link)

https://www.ikea.com/pt/en/images/products/vindstyrka-air-quality-sensor-smart__1137096_pe880020_s5.jpg

### Database entry

{"id":22,"type":"Router","ieeeAddr":"0x287681fffea4f5f0","nwkAddr":8376,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"VINDSTYRKA","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,4,1026,1029,64599,64636,1066,64638],"outClusterList":[3,25,32,514],"clusters":{"genBasic":{"attributes":{"modelId":"VINDSTYRKA","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":8,"appVersion":16,"stackVersion":106,"hwVersion":1,"dateCode":"20220719","swBuildId":"1.0.010"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":16,"stackVersion":106,"hwVersion":1,"dateCode":"20220719","swBuildId":"1.0.010","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1688942868350}

### Notes

I noticed that the same device now appears with a new vendor name "IKEA of Sweden". And therefore it logs the following message after pairing:
`Device '0x287681fffea4f5f0' with Zigbee model 'VINDSTYRKA' and manufacturer name 'IKEA of Sweden' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`

Koenkk, 2023-07-10:
`VINDSTYRKA` is supported, please make sure you are using the latest z2m version.

ctorrao, 2023-07-11:
Maybe you are right, I don't why but the version of this addon in my HA got stuck in v1.18.1-1. But it's odd not having an update anywhere to do. Am I missing something? Should I do again the install procedure. I need to investigate what happen.

ctorrao, 2023-07-11:
I notice that I'm still in the deprecated repo of the zigbee2mqtt addon. I will try with the new repo, but I expect that device it's working correctly with that one.

Sorry about this, if some people with the same problem reach this issue, then should check with repo and version of the addon they have installed.

Closing the issue, thanks for the help.

closed 2023-07-11 by ctorrao