Koenkk/zigbee2mqtt #18274: Install Code over QR Code.

tech2mar, 2023-07-10:
### Is your feature request related to a problem? Please describe

Hi there,

Would it be possible to add and commission devices via QR code scanning, I have some new devices and it would be awesome if I could just scan the devices via the install code, I know there is a way to add this manually or just copy paste from my documents... But I would like to see this in the GUI, I think this could be done. Would be nice. 

### Describe the solution you'd like

Would it not be nice to have some Android App there for the GUI? This would be a dream. 😳 

### Describe alternatives you've considered

That's all. 

### Additional context

Thanks for the great work

Koenkk, 2023-07-10:
This should be done via the frontend, please create an issue in that repo.

tech2mar, 2023-07-10:
Thanks 😃 I will.

closed 2023-08-17 by github-actions[bot]