Koenkk/zigbee2mqtt #18280: [New device support]: Adaprox Fingerbot Sense

phloryan, 2023-07-10:
### Link

https://www.kickstarter.com/projects/adaprox/fingerbot-sense-more-than-just-a-button-pusher

### Database entry

{"id":10,"type":"EndDevice","ieeeAddr":"","nwkAddr":52541,"manufId":4417,"manufName":"_TZ3210_dse8ogfy","powerSource":"Battery","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[6,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0011\u0000\u0000\u0000\u0000\u0011\u0014\u0000\u0000\u0000\u0013","65506":54,"65508":1,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3210_dse8ogfy","zclVersion":3,"appVersion":82,"modelId":"TS0001","powerSource":3}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e949d","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":153093895},"lastSeen":1689005697938,"defaultSendRequestWhen":"immediate"}

### Comments

Looks like it is mostly identical to https://www.zigbee2mqtt.io/devices/TS0001_fingerbot_1.html but is detected as https://www.zigbee2mqtt.io/devices/TS0001.html#tuya-ts0001, triggering works but everything else doesn't.

Trying to set e.g. delay via `{"delay": 10}` fails with `{"level":"error","message":"No converter available for 'delay' (10)"}`.

phloryan, 2023-07-11:
nvm, I wasn't on the latest zigbee2mqtt. After updating, the device is now discovered as fingerbot and is fully functional.

closed 2023-07-11 by phloryan

Sheng2216, 2024-03-05:
I am using the latest docker image ([1.36.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.0) commit: [86ed71c](https://github.com/Koenkk/zigbee2mqtt/commit/86ed71c)), but I cannot detect the fingerbot plus at all from the Zigbee2mqtt UI. I have tried pressing and holding the reset button on the fingerbot until the LED flashes rapidly, but the device is not displayed there. Interestingly, I can see it on the Xiaomi Smart Home app on my cellphone...