Koenkk/zigbee2mqtt #18283: Danalock doesn't work via Router

Nazarbayevv, 2023-07-10:
### What happened?

I have Danalock, which works correct when connect to Coordinator directly.
But when it connect to Router (this is tuya/Matsee plus current clamp meter _TZE204_cjbofhxw in my case) I receive notifications from lock when lock/unlock manually or via bluetooth from danalock App, but when I send commands to lock from zigbee2mqtt it doesn't work.

FW of my lock is 0.17.15

FW from status:
  "installed_version": 285212692,
  "latest_version": 285474816

### How to reproduce it (minimal and precise)

Connect Danalock via Router to network

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

20210708

### Adapter

SONOFF ZB Dongle-P, zStack3x0

Nazarbayevv, 2023-07-12:
Strange situation. I deleted device and joined to my network via Coordinator directly, it reconnected to Router after a while and it works correctly.

closed 2023-07-12 by Nazarbayevv

dhadanalock, 2023-10-02:
Hy there...
Please update the firmware of your lock to 0.21.7 (current latest release) or higher.
And also take a look at https://danalock.com/zigbee for information on what hardware versions can be updated to a later Zigbee firmware and how to do it.

Short answer - scan the QR code on the inside of your lock battery cover, if the production date is 1st of March 2021 or later then your Zigbee module can be updated and we strongly recommend you do so.

If you have any questions then please reach out to our support team by writing an e-mail support@danalock.com.

BR. Dan
Support Tech @ Danalock