Koenkk/zigbee2mqtt #18287: NWK_TABLE_FULL  second outage after upgrade to 1.32.1

Claudio1L, 2023-07-11:
### What happened?

most devices offline, second outage after upgrade to 1.32.1

2023-07-11 10:02:43Failed to configure 'Condizionatore Marco energia ', attempt 1 (Error: Read 0xa4c138b9b11ecff7/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":30749,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)')) at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32))
2023-07-11 10:02:48Failed to configure 'startap', attempt 1 (Error: Read 0xa4c1385e4353de79/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":30882,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)')) at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32))

[image]

### What did you expect to happen?

to be stable

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

20230507

### Adapter

Sonoff usb zStack3x0

Claudio1L, 2023-07-11:
I'm trying to reinstall 1.32.0. but I'm forced to go to 1.32.1
Could you please share the procedure to downgrade?

thanks

joschaschultze, 2023-07-14:
I have massive problems with 1.32.1, too. Devices are offline or reachable with a massive delay. Getting spammed with error messages. Also rolled back to 20221226

closed 2023-08-21 by github-actions[bot]