Koenkk/zigbee2mqtt #18296: Xiaomi WXKG15LM (WRS-R02) - cannot change click_mode

shalak, 2023-07-12:
### What happened?

Changing the `click_mode` from Z2M returns following error:

```
error 2023-07-12 13:48:43Publish 'get' 'click_mode' to 'Schody Pietro' failed: 'Error: Read 0x54ef44100038bd20/1 aqaraOpple([293], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (95))'
info 2023-07-12 13:48:43MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'click_mode' to 'Schody Pietro' failed: 'Error: Read 0x54ef44100038bd20/1 aqaraOpple([293], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (95))'","meta":{"friendly_name":"Schody Pietro"},"type":"zigbee_publish_error"}'
```

The [documentation mentions](https://www.zigbee2mqtt.io/devices/WXKG15LM.html):
> If you have issues changing the click_mode with Zigbee2MQTT you can switch between click mode physically. This is done by clicking either of the two rockers quickly five times. The click mode will then toggle between fast mode and multi mode.

Unfortunately, this doesn't work.

The device has Production Date of 11/2021

### What did you expect to happen?

`click_mode` is changed

### How to reproduce it (minimal and precise)

1. Pair WRS-R02 with Z2M
2. In device's setting click: `Exposes` -> `click_mode` -> `fast` or `multi`
3. Observe error

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Debug log

```
debug 2023-07-12 14:08:09 Publishing 'set' 'click_mode' to 'Schody Pietro'
error 2023-07-12 14:08:19 Publish 'set' 'click_mode' to 'Schody Pietro' failed: 'Error: Write 0x54ef44100038bd20/1 aqaraOpple({"293":{"value":2,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (217))'
debug 2023-07-12 14:08:19 Error: Write 0x54ef44100038bd20/1 aqaraOpple({"293":{"value":2,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (217)) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:656:23) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:81:20) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:486:28) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.js:5120:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:248:36)
info 2023-07-12 14:08:19 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'click_mode' to 'Schody Pietro' failed: 'Error: Write 0x54ef44100038bd20/1 aqaraOpple({\"293\":{\"value\":2,\"type\":32}}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (217))'","meta":{"friendly_name":"Schody Pietro"},"type":"zigbee_publish_error"}'
```

(this is after simply setting `debug` log level in the log viewer, I was unable to change Z2M log level to debug, (the documented way of doing it](https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging) results in broken configuration (`Integration error: advanced - Integration 'advanced' not found. `)

shalak, 2023-08-12:
I'm unable to remove the `stale` label, but the bug is still there.

shalak, 2023-09-13:
Is there a way to keep the issue report not become closed automatically?

simposiummm, 2023-09-28:
Hi,

Same problem for me

Koenkk, 2023-09-28:
What may help is to re-pair the device close to the coordinator and try to set it again.

RakotVT, 2023-11-02:
Hi, same problem for me. Re-pair doesn't help

easterism, 2024-03-09:
same issue

```
Error: Write 0x54ef441000637ccf/1 manuSpecificLumi({"293":{"value":1,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 58797 - 1 - 186 - 64704 - 4 after 10000ms)

closed 2024-10-06 by github-actions[bot]