Koenkk/zigbee2mqtt #18297: Home Assistant Errors with Danfoss Icon through Zigbee2Mqtt

daansan, 2023-07-12:
### What happened?

I've installed zigbee2mqtt with a danfoss icon system. The integration works well and I can access the climate settings in Home Assistant.

However I am running into the following Home Assistant errors that I cannot find the solution for. Would this be related to Zigbee2Mqtt?

Below is a subset of my home assistant log that shows the errors about missing attributes. As far as I can see with an MQTT client the published messages do not always contain those attributes.

2023-07-12 14:14:22.601 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: 
2023-07-12 14:14:22.601 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'local_temperature_l13' when rendering '{{ value_json.local_temperature_l13 }}'
2023-07-12 14:14:22.602 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'occupied_heating_setpoint_l13' when rendering '{{ value_json.occupied_heating_setpoint_l13 }}'
2023-07-12 14:14:22.603 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'system_mode_l14' when rendering '{{ value_json.system_mode_l14 }}'
2023-07-12 14:14:22.604 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: 
2023-07-12 14:14:22.605 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'local_temperature_l14' when rendering '{{ value_json.local_temperature_l14 }}'
2023-07-12 14:14:22.606 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'occupied_heating_setpoint_l14' when rendering '{{ value_json.occupied_heating_setpoint_l14 }}'
2023-07-12 14:14:22.607 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'system_mode_l15' when rendering '{{ value_json.system_mode_l15 }}'
2023-07-12 14:14:22.608 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: 
2023-07-12 14:14:22.609 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'local_temperature_l15' when rendering '{{ value_json.local_temperature_l15 }}'
2023-07-12 14:14:22.610 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'occupied_heating_setpoint_l15' when rendering '{{ value_json.occupied_heating_setpoint_l15 }}'

### Zigbee2MQTT version

1.3.2.1

### Adapter firmware version

20220219

### Adapter

CC2652RB Zigbee Controller - Coördinator

daansan, 2023-07-12:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12027817/log.txt)

daansan, 2023-07-13:
Might be related to https://github.com/Koenkk/zigbee2mqtt/issues/18180

daansan, 2023-08-13:
This issue can be walked around by not using MQTT Discovery in Zigbee2Mqtt and manually creating the MQTT sensors in Home Assistant for those attributes that are always available in the MQTT payload.

closed 2023-09-20 by github-actions[bot]

Symbiot78, 2023-10-19:
@daansan  - I've got the same issue. Can you give me a walkthrough of your solution. I am not sure what you've done to walk around the issue ...

nbarrientos, 2024-02-14:
Same over here, example:

```
Feb 14 09:57:48 tuc home-assistant[288466]: 2024-02-14 09:57:48.382 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: 
Feb 14 09:57:48 tuc home-assistant[288466]: 2024-02-14 09:57:48.382 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'local_temperature_l10' when rendering '{{ value_json.local_temperature_l10 }}'
Feb 14 09:57:48 tuc home-assistant[288466]: 2024-02-14 09:57:48.382 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'occupied_heating_setpoint_l10' when rendering '{{ value_json.occupied_heating_setpoint_l10 }}'
Feb 14 09:57:48 tuc home-assistant[288466]: 2024-02-14 09:57:48.382 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'system_mode_l11' when rendering '{{ value_json.system_mode_l11 }}'
Feb 14 09:57:48 tuc home-assistant[288466]: 2024-02-14 09:57:48.382 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: 
Feb 14 09:57:48 tuc home-assistant[288466]: 2024-02-14 09:57:48.383 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'local_temperature_l11' when rendering '{{ value_json.local_temperature_l11 }}'
Feb 14 09:57:48 tuc home-assistant[288466]: 2024-02-14 09:57:48.383 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'occupied_heating_setpoint_l11' when rendering '{{ value_json.occupied_heating_setpoint_l11 }}'
Feb 14 09:57:48 tuc home-assistant[288466]: 2024-02-14 09:57:48.383 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'system_mode_l12' when rendering '{{ value_json.system_mode_l12 }}'
Feb 14 09:57:48 tuc home-assistant[288466]: 2024-02-14 09:57:48.383 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: 
Feb 14 09:57:48 tuc home-assistant[288466]: 2024-02-14 09:57:48.383 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'local_temperature_l12' when rendering '{{ value_json.local_temperature_l12 }}'
Feb 14 09:57:48 tuc home-assistant[288466]: 2024-02-14 09:57:48.384 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'occupied_heating_setpoint_l12' when rendering '{{ value_json.occupied_heating_setpoint_l12 }}'
Feb 14 09:57:48 tuc home-assistant[288466]: 2024-02-14 09:57:48.384 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'system_mode_l13' when rendering '{{ value_json.system_mode_l13 }}'
```