Koenkk/zigbee2mqtt #18299: [Wrong device]: BR TuYa (Gleco) 16A Zigbee Smart Plug (BR variant picture - GP100-BRZ) detected as BSD29

pabloalcantara, 2023-07-12:
### Link

https://www.aliexpress.com/item/1005005073066732.html?spm=a2g0o.order_list.order_list_main.68.21ef18026oEOGw

### Model

GP100-BRZ

### Description

Smart Plug Tuya (Gleco) GP100-BRZ

### Vendor

Tuya (Gleco)

### Picture (link)

https://ae01.alicdn.com/kf/S042dd2a41e634e6185301bf8ff1c99bbH/Tuya-WiFi-Smart-Brasil-Plug-Switch-Zigbee-Smart-Brasil-Socket-Smart-Life-Brasil-Outlet-com-Power.png

### Database entry

{     "device": {         "applicationVersion": 100,         "dateCode": "20210625",         "friendlyName": "0x70b3d52b60027498",         "hardwareVersion": 2,         "ieeeAddr": "0x70b3d52b60027498",         "manufacturerID": 4660,         "manufacturerName": "_TZ3000_okaz9tjs",         "model": "BSD29",         "networkAddress": 36433,         "powerSource": "Mains (single phase)",         "softwareBuildID": "500+TZSKT31BS102",         "stackVersion": 1,         "type": "Router",         "zclVersion": 2     },     "last_seen": "2023-07-12T10:54:58.631Z",     "linkquality": 48,     "child_lock": null,     "current": 0.01,     "energy": 0,     "indicator_mode": null,     "power": 0,     "power_outage_memory": null,     "state": "ON",     "update": {         "installed_version": -1,         "latest_version": -1,         "state": null     },     "update_available": null,     "voltage": 130,     "elapsed": 288 }

### Notes

Every function os working as BSD29, except OTA

Koenkk, 2023-07-13:
We don't have an OTA for this device, to grab it a TuYa gateway is required. see https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html

closed 2023-08-20 by github-actions[bot]

matoxbr, 2023-09-14:
Pablo how did you add on HA? Because in my case don't work. Showed unsupported.

pabloalcantara, 2023-09-15:
I didn´t have problems adding the component, only the OTA function didn´t work.

matoxbr, 2023-09-18:
Do you used Hub or usb dongle at home assistant?

pabloalcantara, 2023-09-18:
I'm using the sonoff P dongle

matoxbr, 2023-09-18:
the same me, but when i search for new device at zigbee2mqtt don't work. I think is necessary add for external converters.