Koenkk/zigbee2mqtt #18305: No converter available for 'TS0202_1'

Amir-hy, 2023-07-12:
### What happened?

I have 5 motion sensors model:  _TZ3000_mg4dy6z6 and it's working good
I'm trying to add another 2 devices with the same model but getting:
No converter available for 'TS0202_1' with cluster 'ssIasZone' and type 'attributeReport' and data '{"currentZoneSensitivityLevel":1}'

Tried to to rejoin the devices several time but I'm still getting the error again.

Debug log:
Received Zigbee message from 'floor2_livingroom_motion', type 'attributeReport', cluster 'ssIasZone', data '{"currentZoneSensitivityLevel":1}' from endpoint 1 with groupID 0

No converter available for 'TS0202_1' with cluster 'ssIasZone' and type 'attributeReport' and data '{"currentZoneSensitivityLevel":1}'

MQTT publish: topic 'zigbee2mqtt/floor2_livingroom_motion', payload '{"battery":100,"battery_low":false,"last_seen":"2023-07-12T23:17:02+03:00","linkquality":116,"occupancy":false,"tamper":false,"voltage":null}'

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

20220219

### Adapter

Sonoff ZBDongle-E cp210x

closed 2023-07-16 by Amir-hy

yundin2000, 2023-07-17:
Same problem here

Amir-hy, 2023-07-23:
> Same problem here

Try to force remove the device then pair it again.