Koenkk/zigbee2mqtt #18309: Hue wall switch module (929003017102) battery depletion in 5 days (not 5 years as advertised)

bjornarfjelldal, 2023-07-13:
### What happened?

H, @Koenkk! As mentioned in an issue for the same device regarding binding, we are many who experience an insane battery depletion in Z2M, compared to the Hue bridge. Since you have aquired one of these yourself, it would be great if you could have a look at this one :)

As per #12545 and numerous forum posts on HA/reddit.

Personally I am on the skyconnect/EZSP 12, HAOS/Zigbee2MQTT addon, 80+ devices whereas half of them are wired.

When paired, they pair to the nearest node, being only 30cm away from it.

Battery depletes in days, as shown in graph. 

[image]

I have tried to delete all reporting from it and deactive availability, but to no avail. 

Others report of different coordinators and setups, leaving me to think its not my setup, but something either hardware related on the wall switch itself, or something in Z2M. But I may of course be wrong.

Thanks for your great contribution! :)

### What did you expect to happen?

Battery to last longer

### How to reproduce it (minimal and precise)

Pair device in a mature zigbee environment, observe battery loss.

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

7.3.0.0 build 0

### Adapter

Skyconnect

Koenkk, 2023-07-13:
Note: some discussion about this can already be found here: https://github.com/Koenkk/zigbee2mqtt/discussions/12545

I will pair the device to my production network and report back.

Koenkk, 2023-07-22:
Mine still has 100% battery, so I expect this issue is caused by your hardware setup (e.g. the EZSP adapter). Do you have any sniffing hardware? If yes, follow https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html and provide a sniff of 10 minutes + the network address of your device (can be found in the z2m frontend)

bjornarfjelldal, 2023-08-22:
Not stale, I have sniffing hardware, but haven't had time to sniff yet.. :)

rendyhd, 2023-08-29:
I think I have the same issue, just went through the second battery in about 3 weeks

timdonovanuk, 2023-10-06:
I'm finding the hue wall modules completely unreliable. I'm going to open a separate issue for it, but mine frequently go unresponsive, despite LQI of 70+ (still reported as online in z2m, as I guess being edge devices they take a few hours to be treated as offline). The ones that do work sometimes take multiple attempts to work, like they're sleeping. I expected this from cheaper zigbee devices like Aqara (in fact, I replaced Aqara for this reason with the hue wall modules), but these are pretty expensive.

closed 2023-11-15 by github-actions[bot]

bikerglen, 2024-01-26:
> Mine still has 100% battery, so I expect this issue is caused by your hardware setup (e.g. the EZSP adapter). Do you have any sniffing hardware? If yes, follow [...]

@Koenkk, what adapter are you using? I see 802.15.4 data requests from the hue wall switch module every 500 ms with my Sonoff Dongle P running the latest firmware.

Koenkk, 2024-01-28:
@bikerglen If I understood from another issue, you also observed this behaviour with the Hue bridge right?

bikerglen, 2024-01-28:
@Koenkk Yes, I did see this behavior with the Hue wall switch module connected to the Hue bridge. I only left them connected for a few minutes. The period was the same, 500 ms.

I've since learned a bit more about Zigbee polling so I'd like to repeat the experiment but this time leave the wall switch module connected to the bridge for a few hours instead of just a few minutes. There's a Keepalive and Poll Control Cluster in the Zigbee spec. I don't think the wall switch module exposes this cluster but maybe the Hue hub eventually writes something to the wall switch module that is similar and that reduces the poll rate. Anyway, that's another experiment I'm going to run.

robertteir, 2024-03-19:
[image]

I had a similar issue with rapid depletion on one of my switches and not the other one. The difference that I observed was that the firmware on the lights the switch was controlling was different. So, I upgraded the lights and then the depletion stopped. Not sure if this is just a coincidence butt adding it to the conversation.